verilator/verilator #4486: verilator_coverage executable docs: request to clarify

mczyz-antmicro, 2023-09-12:
Hey,
I have a few remarks after reading the [verilator_coverage executable part of the docs](https://verilator.org/guide/latest/exe_verilator_coverage.html).

1. https://github.com/verilator/verilator/blob/master/docs/guide/exe_verilator_coverage.rst?plain=1#L67-L71
My suggestion here is to explicitly state that number denotes the number of coverage hits and that lines starting with the ' ' space symbol have sufficient coverage. Also, I think that an example would be great here (correct me if I misunderstood the report):
```
   153724    input  logic en; // OK, because starts with space, there were 153724 coverage hits
  %000000    input  logic scan_mode; // Not OK, because starts with %, signal never toggled
```
2. https://github.com/verilator/verilator/blob/master/docs/guide/exe_verilator_coverage.rst?plain=1#L90
This line seems like a copy-paste artifact, can you clarify?

3. It seems that options `annotate-all` `annotate-min` and `annotate-points` only make sense when used together with `annotate`. Could you put suggestions into the option descriptions about which options are meant to be used together?

4. https://github.com/verilator/verilator/blob/master/docs/guide/exe_verilator_coverage.rst?plain=1#L80-L83
This sentence is a bit confusing to me, mostly because of the first part "Specify if the coverage point does not include the count number of coverage hits". Could you rephrase it?

Very much obliged, if you find the time to address my concerns.

wsnyder, 2023-09-12:
1. Correct.
2. Correct, remove it.
3. Correct, just for annotate.
4. That is ugly.  Basically it sets the coverage amount needed to either print "%" or " ", as in your #1.

Perhaps you'd consider making a pull request to fix these in the docs?

mczyz-antmicro, 2023-09-13:
OK

mczyz-antmicro, 2023-09-18:
Open PR#4504 @wsnyder can you take a look?

closed 2023-09-19 by wsnyder

ETOgaosion, 2023-10-21:
Hello guys, the official [document](https://veripool.org/guide/latest/exe_verilator_coverage.html) has been outdated with current rst, quite misleading, and `-` option shall all be `--`, and `--read` not gonna work.

wsnyder, 2023-10-21:
@ETOgaosion this us not related to this issue,  please file a new issue. Perhaps you could instead send a pull request to fix the documentation? BTW - or -- work the same.