verilator/verilator #4491: no verilog.h

hyfgouzi, 2023-09-15:
making verilator in src
make -C src 
make[1]: Entering directory '/home/hyf/verilator/src'
/usr/bin/perl ./config_rev.pl . >config_rev.h
make -C obj_dbg -j 1  TGT=../../bin/verilator_bin_dbg VL_DEBUG=1 -f ../Makefile_obj serial
make[2]: Entering directory '/home/hyf/verilator/src/obj_dbg'
/usr/bin/perl ../astgen -I..  --classes
If you get errors from verilog.y below, try upgrading bison to version 1.875 or newer.
/usr/bin/perl ../bisonpre --yacc /usr/bin/bison -d -v -o V3ParseBison.c ../verilog.y
  edit ../verilog.y V3ParseBison_pretmp.y
  /usr/bin/bison -d -v --report=itemset --report=lookahead -b V3ParseBison_pretmp -o V3ParseBison_pretmp.c V3ParseBison_pretmp.y
  edit V3ParseBison_pretmp.output V3ParseBison.output
  edit V3ParseBison_pretmp.c V3ParseBison.c
  edit V3ParseBison_pretmp.h V3ParseBison.h
make[2]: Leaving directory '/home/hyf/verilator/src/obj_dbg'
make -C obj_dbg       TGT=../../bin/verilator_bin_dbg VL_DEBUG=1 -f ../Makefile_obj
make[2]: Entering directory '/home/hyf/verilator/src/obj_dbg'
      Compile flags:  g++ -MMD -I. -I.. -I.. -I../../include -I../../include -DYYDEBUG -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC="" -DDEFENV_SYSTEMC_ARCH="" -DDEFENV_SYSTEMC_INCLUDE="" -DDEFENV_SYSTEMC_LIBDIR="" -DDEFENV_VERILATOR_ROOT="/usr/local/share/verilator"
ccache g++  -MMD -I. -I.. -I.. -I../../include -I../../include -DYYDEBUG         -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -c ../Verilator.cpp -o Verilator.o
ccache g++  -MMD -I. -I.. -I.. -I../../include -I../../include -DYYDEBUG         -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -c ../V3Active.cpp -o V3Active.o
ccache g++  -MMD -I. -I.. -I.. -I../../include -I../../include -DYYDEBUG         -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -c ../V3ActiveTop.cpp -o V3ActiveTop.o
ccache g++  -MMD -I. -I.. -I.. -I../../include -I../../include -DYYDEBUG         -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -c ../V3Assert.cpp -o V3Assert.o
ccache g++  -MMD -I. -I.. -I.. -I../../include -I../../include -DYYDEBUG         -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -c ../V3AssertPre.cpp -o V3AssertPre.o
ccache g++  -MMD -I. -I.. -I.. -I../../include -I../../include -DYYDEBUG         -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -c ../V3Ast.cpp -o V3Ast.o
ccache g++  -MMD -I. -I.. -I.. -I../../include -I../../include -DYYDEBUG         -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -c ../V3AstNodes.cpp -o V3AstNodes.o
ccache g++  -MMD -I. -I.. -I.. -I../../include -I../../include -DYYDEBUG         -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -c ../V3Begin.cpp -o V3Begin.o
ccache g++  -MMD -I. -I.. -I.. -I../../include -I../../include -DYYDEBUG         -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -c ../V3Branch.cpp -o V3Branch.o
ccache g++  -MMD -I. -I.. -I.. -I../../include -I../../include -DYYDEBUG         -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -c ../V3Broken.cpp -o V3Broken.o
ccache g++  -MMD -I. -I.. -I.. -I../../include -I../../include -DYYDEBUG         -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -c ../V3CCtors.cpp -o V3CCtors.o
ccache g++  -MMD -I. -I.. -I.. -I../../include -I../../include -DYYDEBUG         -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -c ../V3CUse.cpp -o V3CUse.o
ccache g++  -MMD -I. -I.. -I.. -I../../include -I../../include -DYYDEBUG         -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -c ../V3Case.cpp -o V3Case.o
ccache g++  -MMD -I. -I.. -I.. -I../../include -I../../include -DYYDEBUG         -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -c ../V3Cast.cpp -o V3Cast.o
ccache g++  -MMD -I. -I.. -I.. -I../../include -I../../include -DYYDEBUG         -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -c ../V3Cdc.cpp -o V3Cdc.o
ccache g++  -MMD -I. -I.. -I.. -I../../include -I../../include -DYYDEBUG         -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -c ../V3Changed.cpp -o V3C [... cut ...] Wno-char-subscripts -Wno-unused -c ../V3ParseGrammar.cpp -o V3ParseGrammar.o
In file included from ../V3ParseGrammar.cpp:22:
verilog.c:387:10: fatal error: verilog.h: No such file or directory
compilation terminated.
make[2]: *** [../Makefile_obj:300: V3ParseGrammar.o] Error 1
make[2]: Leaving directory '/home/hyf/verilator/src/obj_dbg'
make[1]: *** [Makefile:66: ../bin/verilator_bin_dbg] Error 2
make[1]: Leaving directory '/home/hyf/verilator/src'
make: *** [Makefile:221: verilator_exe] Error 2

wsnyder, 2023-09-15:
What OS are you using?

Please post the results of "./configure" also

wsnyder, 2023-09-15:
What OS and distribution and distro version are you using?

Please post the results of "./configure" also

This very likely is a problem with "bison" make sure it's installed.

wsnyder, 2023-09-26:
I'm assuming you figured this out, if not post more information.

closed 2023-09-26 by wsnyder

mattielee, 2024-02-13:
hi! I have this same issue. I made sure to install bison and this is my 01.configure log: 
[image]
[image]

hyfgouzi, 2024-02-13:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。

wsnyder, 2024-02-13:
Please use a the most recent version instead of one from 2020.

mattielee, 2024-02-14:
Hi! unfortunately, I am trying to use verilator through [pyMTL3 ](https://pymtl3.readthedocs.io/en/latest/index.html) which requires verilator 4.036 specifically for some reason...

wsnyder, 2024-02-14:
Not sure we will be able to help with older versions, but there is no problem shown in what you posted.

mattielee, 2024-02-14:
I have the same problem as the issue. Here is my specific log: 

Last 15 lines from /home/mattiellee/.cache/Homebrew/Logs/verilator/02.make:
ccache g++-11 -I/home/linuxbrew/.linuxbrew/Cellar/verilator/4.036/include -I/home/linuxbrew/.linuxbrew/Cellar/verilator/4.036/include -MMD -I. -I.. -I.. -I../../include -I../../include -DYYDEBUG   -O2 -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/home/linuxbrew/.linuxbrew/Cellar/verilator/4.036/share/verilator\" -c ../V3WidthSel.cpp -o V3WidthSel.o
ccache g++-11 -I/home/linuxbrew/.linuxbrew/Cellar/verilator/4.036/include -I/home/linuxbrew/.linuxbrew/Cellar/verilator/4.036/include -MMD -I. -I.. -I.. -I../../include -I../../include -DYYDEBUG   -O2 -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/home/linuxbrew/.linuxbrew/Cellar/verilator/4.036/share/verilator\" -Wno-char-subscripts -Wno-unused -c ../V3ParseImp.cpp -o V3ParseImp.o
ccache g++-11 -I/home/linuxbrew/.linuxbrew/Cellar/verilator/4.036/include -I/home/linuxbrew/.linuxbrew/Cellar/verilator/4.036/include -MMD -I. -I.. -I.. -I../../include -I../../include -DYYDEBUG   -O2 -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/home/linuxbrew/.linuxbrew/Cellar/verilator/4.036/share/verilator\" -Wno-char-subscripts -Wno-unused -c ../V3ParseGrammar.cpp -o V3ParseGrammar.o
/usr/bin/flex --version
In file included from ../V3ParseGrammar.cpp:22:
verilog.c:313:10: fatal error: verilog.h: No such file or directory
compilation terminated.
make[2]: *** [../Makefile_obj:301: V3ParseGrammar.o] Error 1
make[2]: *** Waiting for unfinished jobs....
flex 2.6.4
/usr/bin/flex -d -oV3Lexer_pregen.yy.cpp ../verilog.l
make[2]: Leaving directory '/tmp/verilator-20240213-453209-9kdzr8/verilator-4.036/src/obj_opt'
make[1]: *** [Makefile:60: ../bin/verilator_bin] Error 2
make[1]: Leaving directory '/tmp/verilator-20240213-453209-9kdzr8/verilator-4.036/src'
make: *** [Makefile:221: verilator_exe] Error 2

Do not report this issue to Homebrew/brew or Homebrew/homebrew-core!

mattielee, 2024-02-14:
Here is the full log: 

2024-02-13 05:38:17 +0000

make

pod2man bin/verilator verilator.1
------------------------------------------------------------
pod2man bin/verilator_coverage verilator_coverage.1
pod2man bin/verilator_gantt verilator_gantt.1
pod2man bin/verilator_profcfunc verilator_profcfunc.1
making verilator in src
/usr/bin/make -C src
make[1]: Entering directory '/tmp/verilator-20240213-457797-wptgi5/verilator-4.036/src'
mkdir -p obj_dbg
mkdir -p obj_opt
/usr/bin/make -C obj_dbg -j 1  TGT=../../bin/verilator_bin_dbg VL_DEBUG=1 -f ../Makefile_obj serial
/usr/bin/make -C obj_dbg       TGT=../../bin/verilator_coverage_bin_dbg VL_DEBUG=1 VL_VLCOV=1 -f ../Makefile_obj serial_vlcov
/usr/bin/make -C obj_opt -j 1  TGT=../../bin/verilator_bin -f ../Makefile_obj serial
make[2]: Entering directory '/tmp/verilator-20240213-457797-wptgi5/verilator-4.036/src'
make[2]: warning: -j1 forced in submake: resetting jobserver mode.
make[2]: Entering directory '/tmp/verilator-20240213-457797-wptgi5/verilator-4.036/src'
make[2]: warning: -j1 forced in submake: resetting jobserver mode.
/usr/bin/perl ../astgen -I..  --classes
make[2]: Entering directory '/tmp/verilator-20240213-457797-wptgi5/verilator-4.036/src/obj_dbg'
/usr/bin/perl ../vlcovgen --srcdir ..
/usr/bin/perl ../astgen -I..  --classes
touch vlcovgen.d
make[2]: Leaving directory '/tmp/verilator-20240213-457797-wptgi5/verilator-4.036/src/obj_dbg'
/usr/bin/make -C obj_dbg       TGT=../../bin/verilator_coverage_bin_dbg VL_DEBUG=1 VL_VLCOV=1 -f ../Makefile_obj
make[2]: Entering directory '/tmp/verilator-20240213-457797-wptgi5/verilator-4.036/src/obj_dbg'
ccache g++-11 -I/home/linuxbrew/.linuxbrew/Cellar/verilator/4.036/include -I/home/linuxbrew/.linuxbrew/Cellar/verilator/4.036/include -MMD -I. -I.. -I.. -I../../include -I../../include -DYYDEBUG   -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/home/linuxbrew/.linuxbrew/Cellar/verilator/4.036/share/verilator\" -c ../VlcMain.cpp -o VlcMain.o
      Compile flags:  g++-11 -I/home/linuxbrew/.linuxbrew/Cellar/verilator/4.036/include -I/home/linuxbrew/.linuxbrew/Cellar/verilator/4.036/include -I/home/linuxbrew/.linuxbrew/Cellar/verilator/4.036/include -MMD -I. -I.. -I.. -I../../include -I../../include -DYYDEBUG -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC="" -DDEFENV_SYSTEMC_ARCH="" -DDEFENV_SYSTEMC_INCLUDE="" -DDEFENV_SYSTEMC_LIBDIR="" -DDEFENV_VERILATOR_ROOT="/home/linuxbrew/.linuxbrew/Cellar/verilator/4.036/share/verilator"
If you get errors from verilog.y below, try upgrading bison to version 1.875 or newer.
/usr/bin/perl ../bisonpre --yacc /usr/bin/bison -d -v -o V3ParseBison.c ../verilog.y
If you get errors from verilog.y below, try upgrading bison to version 1.875 or newer.
/usr/bin/perl ../bisonpre --yacc /usr/bin/bison -d -v -o V3ParseBison.c ../verilog.y
  edit ../verilog.y V3ParseBison_pretmp.y
  edit ../verilog.y V3ParseBison_pretmp.y
  /usr/bin/bison -d -v --report=itemset --report=lookahead -b V3ParseBison_pretmp -o V3ParseBison_pretmp.c V3ParseBison_pretmp.y
  /usr/bin/bison -d -v --report=itemset --report=lookahead -b V3ParseBison_pretmp -o V3ParseBison_pretmp.c V3ParseBison_pretmp.y
  edit V3ParseBison_pretmp.output V3ParseBison.output
  edit V3ParseBison_pretmp.output V3ParseBison.output
  edit V3ParseBison_pretmp.c V3ParseBison.c
  edit V3ParseBison_pretmp.c V3ParseBison.c
  edit V3ParseBison_pretmp.h V3ParseBison.h
make[2]: Leaving directory '/tmp/verilator-20240213-457797-wptgi5/verilator-4.036/src/obj_dbg'
/usr/bin/make -C obj_dbg       TGT=../../bin/verilator_bin_dbg VL_DEBUG=1 -f ../Makefile_obj
make[2]: Entering directory '/tmp/verilator-20240213-457797-wptgi5/verilator-4.036/src/obj_dbg'
ccache g++-11 -I/home/linuxbrew/.linuxbrew/Cellar/verilator/4.036/include -I/home/linuxbrew/.linuxbrew/Cellar/verilator/4.036/include -MMD -I. -I.. -I.. -I../../include -I../../include -DYYDEBUG   -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/home/linuxbrew/.linuxbrew/Cellar/verilator/4.036/share/verilator\" -c ../Verilator.cpp -o Verilator.o
      Compile flags:  g++-11 -I/home/linuxbrew/.linuxbrew/Cellar/verilator/4.036/include -I/home/linuxbrew/.linuxbrew/Cellar/verilator/4.036/include -I/home/linuxbrew/.linuxbrew/Cellar/verilator/4.036/include -MMD -I. -I.. -I.. -I../../include -I../../include -DYYDEBUG -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC="" -DDEFENV_SYSTEMC_ARCH="" -DDEFENV_SYSTEMC_INCLUDE="" -DDEFENV_SYSTEMC_LIBDIR="" -DDEFENV_VERILATOR_ROOT="/home/linuxbrew/.linuxbrew/Cellar/verilator/4.036/share/verilator"
ccache g++-11 -I/home/linuxbrew/.linuxbrew/Cellar/verilator/4.036/include -I/home/linuxbrew/.linuxbrew/Cellar/verilator/4.036/include -MMD -I. -I.. -I.. -I../../include -I../../include -DYYDEBUG   -ggdb -DVL_DEBUG [... cut ...]
Kernel: Linux 5.15.133.1-microsoft-standard-WSL2 x86_64 GNU/Linux
OS: Ubuntu 22.04.3 LTS (jammy)
WSL: 2 (Microsoft Store)
Host glibc: 2.35
/usr/bin/gcc: 11.4.0
/usr/bin/ruby: N/A
glibc: N/A
gcc@11: N/A
gcc: 13.2.0
xorg: N/A

HOMEBREW_CC: gcc-11
HOMEBREW_CXX: g++-11
MAKEFLAGS: -j12
CMAKE_PREFIX_PATH: /home/linuxbrew/.linuxbrew
HOMEBREW_GIT: git
ACLOCAL_PATH: /home/linuxbrew/.linuxbrew/share/aclocal
PATH: /home/linuxbrew/.linuxbrew/Homebrew/Library/Homebrew/shims/linux/super:/home/linuxbrew/.linuxbrew/opt/binutils/bin:/usr/bin:/bin:/usr/sbin:/sbin

mattielee, 2024-02-14:
This is me trying to install using homebrew, but I ran into the same exact error message when trying to install directly from the repo using the instructions [here](https://verilator.org/guide/latest/install.html)

wsnyder, 2024-02-14:
Similar to as suggested in the original report, bison isn't creating the correct output, and/or something in homebrew is breaking it.  What "bison --version" is being used? 3.8.2 should work.

mattielee, 2024-02-14:
I am using bison 3.8.2.

wsnyder, 2024-02-14:
Specifically bison should be making and referring to V3ParseBison.h, not verilog.y.  Should work but try 3.8.4.  Also I don't know what homebrew does differently; so you'll need to debug on your end.  I double confirmed that 3.8.4 works fine on Ubuntu 22.04 for what it's worth.

mattielee, 2024-02-14:
Okay thank you! I will try to install from source again.