verilator/verilator #4492: vlSymsp

hyfgouzi, 2023-09-15:
在 main.cpp:2 包含的文件中:
testbench.h: 在成员函数'virtual uint8 t testbench :: read (uint32 t) 中:testbench.h: 127: 30: 错误: struct Vriscv tcm top'没有名为 VISymsp'的成员; 您的意思是Vriscv tcm top Symsconst Vriscv tcm top::vlSymsp”吗? (无法从此上下文访问)127|返回 m dut->m rtl-> VISymsp->TOP v u tcm.read(addr);
111100000
在 testbench.h:12 包含的文件中，
来自 main.cpp:2:
./verilated/Vriscv tcm top.h:26:33: 注意: 此处声明为私有
26  Vriscv tcm top Syms* const wlSymsp;
在 main.cpp:2 包含的文件中:
testbench.h: 在成员函数"virtual uint32 t testbench.get opcode0"中:testbench.h:158:37: 警告: 返回 void [-Wreturn-type]158 的函数中没有 return 语句 uint32 t get opcode(void){]
make[1]: *** [makefile.build sysc tb:56: obi/main.o] 错误1make[1]: 离开目录/home/alan/riscv/top tcm axi/tb'make: *** [makefile :30: build] 错误 2
------------------简自来 -------- ---------
主要错误消息是:testbench.h:111:23: error: 'struct Vriscv tcm top没有名为 VSymsp"的成员;您的意思是"Vriscv tm top Syms* constVriscv tcm top:vlSymsp”吗? (无法从此上下文访问)
111| m dut->m rtl-> VISymsp->TOP v u tcm.write(addr, 数据):
我尝试使用 vlSymsp 替换 VISymsp 但收到错误消息”vlSymsp 是私有的我想知道这是否是由 verilator 版本不同引起的?
有人能帮我解决这个错误吗?

wsnyder, 2023-09-15:
I'm sorry, but I and many here are English only, it would help to post after a Google translate, here it is:

```
In the file main.cpp:2 contains:
testbench.h: in member function 'virtual uint8 t testbench::read (uint32 t): testbench.h: 127:30: error: struct Vriscv tcm top' has no member named VISymsp'; you meant Vriscv tcm top Symsconst Vriscv tcm top::vlSymsp"? (cannot be accessed from this context) 127 | return m dut->m rtl-> VISymsp->TOP v u tcm.read(addr);
111100000
In the file included by testbench.h:12,
From main.cpp:2:
./verilated/Vriscv tcm top.h:26:33: Note: This is declared private
26 Vriscv tcm top Syms* const wlSymsp;
In the file main.cpp:2 contains:
testbench.h: In member function "virtual uint32 t testbench.get opcode0": testbench.h:158:37: Warning: There is no return statement uint32 t get opcode(void) in the function that returns void [-Wreturn-type]158 {]
make[1]: *** [makefile.build sysc tb:56: obi/main.o] Error 1make[1]: Leaving directory /home/alan/riscv/top tcm axi/tb'make: *** [ makefile :30: build] Error 2
------------------Jian Zilai-------- ---------
The main error message is: testbench.h:111:23: error: 'struct Vriscv tcm top has no member named VSymsp'; did you mean "Vriscv tm top Syms* constVriscv tcm top:vlSymsp"? (Can't get this context from this access)
111| m dut->m rtl-> VISymsp->TOP v u tcm.write(addr, data):
I tried replacing VISymsp with vlSymsp but got the error message "vlSymsp is private" I was wondering if this is caused by a different verilator version?
Can anyone help me resolve this error?
```

hyfgouzi, 2023-09-15:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。

wsnyder, 2023-09-15:
Your testbench shouldn't generally be using vlSyms, that is internal, please see the examples directory for how to interface a model.  If you understand what vlSyms is, or it's Verilator generated code that is using vlSyms please post the Verilog or C++ code that is making this error, ideally as a complete self contained test case.

wsnyder, 2023-09-26:
I'm assuming you figured this out, if not post more information.

closed 2023-09-26 by wsnyder