verilator/verilator #4496: Support UVM delayed assignment inside public function/task

wsnyder, 2023-09-16:
UVM has this:

```
task uvm_wait_for_nba_region;
  int nba;
  int next_nba;
  next_nba++;
  //vvvv -- %Error-UNSUPPORTED: t/t_uvm_pkg_todo.vh:875:7: Unsupported: Delayed assignment inside public function/task
  nba <= next_nba;
  @(nba);
endtask
```

This blocks UVM post-elaboration.  Despite the message the function isn't public, it's because it (probably correctly) couldn't inline.

kbieganski, 2023-09-19:
Is there a reason we don't allow NBAs in non-inlined functions? Or is it okay to just limit the check to actual public functions?

wsnyder, 2023-09-19:
>Is there a reason we don't allow NBAs in non-inlined functions? Or is it okay to just limit the check to actual public functions?

Before --timing it would not be able to properly schedule the old/new version of the variable.

I think that's still true, but feel free to write a test to check it works correctly with --timing.

PR #4572 (2023-10-21): Support NBAs in non-inlined functions/tasks (#4496)
Closes #4496

Before this patch, non-blocking assignments in non-inlined functions/tasks
resulted in an error like:

```
%Error-UNSUPPORTED: ...: Unsupported: Delayed assignment inside public function/task
```

That's because `V3Delayed` requires access to the process that contains the
assignment. That's not really possible if it's in a non-inlined function,
especially a virtual one.

A subtler problem is the lifetime of variables on the LHS of the assignment. In
an inlined function, these variables are static, so there is no issue. However,
 [... cut]
closed 2023-10-21 by wsnyder