verilator/verilator #4497: Fix uvm_sequencer wrong reference type

wsnyder, 2023-09-16:
Towards UVM elaboration #1538 fix class parameterization of the following:

```
function uvm_sequencer::new (string name, uvm_component parent=null);
  super.new(name, parent);
  //vvv %Error: t/t_uvm_pkg_todo.vh:19869:21: Function Argument expects a CLASSREFDTYPE 'uvm_sequencer__Tz97_TBz97', got CLASSREFDTYPE 'uvm_sequencer__Tz97'
  seq_item_export = new ("seq_item_export", this);
endfunction
```

donlon, 2023-09-16:
V3Param doesn't generate correct new name for some parameterized classes. I'm trying to fix it...

franAyachi, 2024-10-08:
Hi!, Is there any update about : Function Argument expects a CLASSREFDTYPE 'uvm_sequencer__Tz97_TBz97', got CLASSREFDTYPE 'uvm_sequencer__Tz97'  ? Actually I am trying to compile some tests with riscv-dv, I guess it doesnt use any sequencer, so is there any way to solve this issue somehow?