verilator/verilator #4498: Bit selection with add operation generates wrong result

Tang-Haojin, 2023-09-17:
Thanks for taking the time to report this.

**Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)**
``` verilog
module our(clock, in);
  input clock;
  input [4:0] in;
  wire [31:0][38:0] testvar = {32{39'd114514}};
  always @(posedge clock) begin
    $display("in=%d, result=%d", in, testvar[in + 5'b1]);
    if (in == 5'd31) $finish;
  end
endmodule
```
When `in` traverse from 0 to 31, the output would be like:
```
in= 0, result=      114514
in= 1, result=      114514
in= 2, result=      114514
in= 3, result=      114514
in= 4, result=      114514
in= 5, result=      114514
in= 6, result=      114514
in= 7, result=      114514
in= 8, result=      114514
in= 9, result=      114514
in=10, result=      114514
in=11, result=      114514
in=12, result=      114514
in=13, result=      114514
in=14, result=      114514
in=15, result=      114514
in=16, result=      114514
in=17, result=      114514
in=18, result=      114514
in=19, result=      114514
in=20, result=      114514
in=21, result=      114514
in=22, result=      114514
in=23, result=      114514
in=24, result=      114514
in=25, result=      114514
in=26, result=      114514
in=27, result=      114514
in=28, result=      114514
in=29, result=      114514
in=30, result=      114514
in=31, result=           0
- our.v:7: Verilog $finish
```
When `in` is 31, the result should be 114514 rather than 0, because the length of `in + 5'b1` should be `max(L(in), L(5'b1))` in vector bit-select context, namely 5, accroding to IEEE 1800.

**What 'verilator --version' are you using?  Did you try it with the git master version?**
Verilator 5.017 devel rev v5.014-113-gcd1fe64e6
And this is the master branch until now.

**What OS and distribution are you using?**
```
> lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.2 LTS
Release:        20.04
Codename:       focal
> uname -a
Linux <hostname> 5.15.0-69-generic #76~20.04.1-Ubuntu SMP Mon Mar 20 15:54:19 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
```

May we assist you in trying to fix this in Verilator yourself?

Tang-Haojin, 2023-09-17:
We tested in vcs, and it outputs properly.
```
...
in=24, result=      114514
in=25, result=      114514
in=26, result=      114514
in=27, result=      114514
in=28, result=      114514
in=29, result=      114514
in=30, result=      114514
in=31, result=      114514
$finish called from file "our.v", line 7.
```

wsnyder, 2023-09-17:
Thanks for the good example.

IEEE does not state that the index to an element of an array is sized to the power-of-two of that array.  It does in 7.4.2 say that the ranges are integral. Try this:

```
module our;
   wire [31:0][38:0] testvar = {32{39'd114514}};
   initial begin
      int in = 31;
      bit [4:0] in5 = 5'31;
      $display("in=%d, result=%d", in, testvar[in + 5'b1]);
      $display("in=%d, result=%d", in5, testvar[in5 + 5'b1]);
      $finish;
   end
endmodule
```

Verilator agrees with 3 of the major simulators (after converting X to 0).  VCS disagrees and you noted also is what you are comparing against.  From a preventing-a-design-bug standpoint I think Verilator and the majority are doing the safer thing.  So I consider this a VCS bug (arguably - I doubt they would change), not a Verilator bug.

closed 2023-09-17 by wsnyder

Tang-Haojin, 2023-09-17:
Thank you for your kindly explanation. May I have a further question?

I noticed that in IEEE 1800-2017: _11.5.1 Vector bit-select and part-select addressing_, it says:

> Bit-selects extract a particular bit from a vector, packed array, packed structure, parameter, or concatenation. The bit can be addressed using an expression that shall be evaluated in a self-determined context.

Do you mean that, here, "bit-select" only refers to one-dimension array select, like

``` verilog
wire [3:0] one_d = 4'b1111;
one_d[2'd3 + 2'd1]; // should equal to one_d[2'd0]
```

but not refers to the element selection of packed array that have multi-dimension, like

``` verilog
wire [3:0][1:0] two_d;
two_d[2'd3+2'd1]; // do not have to equal to two_d[2'd0]
```

wsnyder, 2023-09-18:
Bit selects are just bits within a word, not elements of a packed array - note it says "packed" in the sentence.

Tang-Haojin, 2023-09-18:
I see. Thanks a lot.