verilator/verilator #4499: configure fails to find libsystemc.so

yurivict, 2023-09-17:
systemc-2.3.4 is installed, but configure fails:
```
configure:9503: checking whether SystemC is found (in system path)
configure:9520: c++ -o conftest -O2 -pipe -fstack-protector-strong -isystem /usr/local/include -fno-strict-aliasing  -isystem /usr/local/include   -isystem /usr/local/include   -fstack-protector-strong -L/usr/local/lib   conftest.cpp  -lsystemc >&5
conftest.cpp:22:70: warning: non-void function does not return a value [-Wreturn-type]
                     extern "C" int sc_main(int argc, char* argv[]) {}
                                                                     ^
1 warning generated.
ld: error: undefined symbol: sc_core::sc_api_version_2_3_4_cxx201703L<&sc_core::SC_DISABLE_VIRTUAL_BIND_UNDEFINED_>::sc_api_version_2_3_4_cxx201703L(sc_core::sc_writer_policy)
>>> referenced by conftest.cpp
>>>               /tmp/conftest-1963fa.o:(_GLOBAL__sub_I_conftest.cpp)
c++: error: linker command failed with exit code 1 (use -v to see invocation)
```

Version: 5.016
FreeBSD 13.2

wsnyder, 2023-09-17:
I committed a fix for the warning, but that's unrelated to your larger "ld" problem.

There's something broken in your systemc installation if the ld error occurs, it's not installed properly or otherwise.

This is unrelated to Verilator. As you've found in the past I and most of us don't use FreeBSD and you'll need to debug it.  I would start by making your own SC program and getting it working outside Verilator (the code it prints for example).

Closing as not much we can help further, but feel free to follow up with what the problem was to help others.

closed 2023-09-17 by wsnyder

yurivict, 2023-11-01:
The C++ level mismatch caused this problem.