verilator/verilator #4500: how to use program block for testbench

muzafferkal, 2023-09-17:
I'm working on a simple systemverilog testbench for a module and I decided to add clocking & program blocks for easier integration. The problem is I am getting the following error when I instantiate my "program" inside the top level testbench module:

%Error-UNSUPPORTED: cim_tb.sv:96:1: Unsupported: program decls within module decls

This is my first attempt at using programs in my testbench but this structure looks like it should work. Any ideas about a work-around?

My setup is similar to following and I'm getting the error at endmodule line:

module cim_tb;
DUT u0;
program foo;
endprogram
endmodule

donlon, 2023-09-17:
SystemVerilog's program is very similar to a module. It should be declared in the top level. So just

``` SystemVerilog
program foo;
endprogram

module cim_tb;
  DUT u0();
  foo myfoo();
endmodule
```

closed 2023-09-17 by wsnyder

muzafferkal, 2023-09-17:
This is what I ended up trying after I posted the question and it worked. Thanks.