verilator/verilator #4501: Add standard waivers?

wsnyder, 2023-09-18:
I'm seeking opinions on UVM waivers.

We're getting closer to having UVM.  There's many lint waivers currently reported inside UVM.

When a lint failure seems it shouldn't be reported I've been fixing Verilator. Some others I've fed upstream to Accelera, but it will be a while before those propagate to an official release.

I'm thinking it will be a better user experience if we by default suppress warnings inside the UVM code.  One path would be to suppress them inside uvm-verilator code - but that presumes a user is using that version, and also the suppressions may not match what a specific version of verilator needs, after e.g. a new warning is added.

So I'm thinking we should have a default waiver file, e.g. `include/verilated_waivers.vlt`.  This would be used unless `--no-std-waiver` is used (and `--no-std` would change to `--no-std-lib`, and `--no-std` would then mean `--no-std-lib --no-std-waiver`.

For example here's some needed:

```
`verilator_config
lint_off -rule CONSTRAINTIGN -file "*/uvm/src/reg/uvm_mem_mam.svh"
lint_off -rule CONSTRAINTIGN -file "*/uvm/src/reg/uvm_reg_field.svh"
lint_off -rule CONSTRAINTIGN -file "*/uvm/src/reg/uvm_reg_fifo.svh"
lint_off -rule CONSTRAINTIGN -file "*/uvm/src/reg/uvm_reg_item.svh"
lint_off -rule CONSTRAINTIGN -file "*/uvm/src/seq/uvm_sequence_library.svh"
lint_off -rule MISINDENT -file "*/uvm/src/base/uvm_cmdline_processor.svh"
lint_off -rule MISINDENT -file "*/uvm/src/seq/uvm_sequencer_base.svh"
lint_off -rule MISINDENT -file "*/uvm/src/reg/uvm_reg_map.svh"
```

Also I'm wondering if we should be even more specific and also as a new feature match against a module (/package/class) name?

```
lint_off -rule MISINDENT -module uvm_heartbeat -file "*/uvm/src/base/uvm_cmdline_processor.svh"
lint_off -rule MISINDENT -module uvm_push_sequencer -file "*/uvm/src/seq/uvm_sequence_base.svh"
lint_off -rule MISINDENT -module uvm_reg_map  -file "*/uvm/src/reg/uvm_reg_map.svh"
lint_off -rule MISINDENT -module uvm_reg_mem_built_in_seq -file "*/uvm/src/reg/sequences/uvm_reg_mem_hdl_paths_seq.svh"
lint_off -rule MISINDENT -module uvm_sequencer_base -file "*/uvm/src/seq/uvm_sequencer_base.svh"

```

Thoughts, especially @gezalore and @wallento?

PeterMonsson, 2023-09-18:
Thank you for thinking about this. In general people seem to use the version that their simulator vendor gave them. For many that means they are stuck on UVM 1.2 indefinitely. 

I would suggest that you consider a version specific waiver file per UVM release so that management of waiver files is simpler. 

That means there would be
accellera_uvm_2017-1.1.vlt 
accellera_uvm_2020-2.0.vlt
verilator_uvm_2020-2.0.vlt
etc. 

Users would have to pass the correct waiver file to verilator. 

In general I think that verilator should choose a strategy that is easy to maintain. Old UVM releases seem live on for decades.

gezalore, 2023-10-13:
I think a default waiver corresponding to input SV source code that is shipped with Verilator is appropriate and sufficient. 

Conceptually all this is just a question on how to handle "library code". If it'a a library we maintain and distribute (and sneakily add to the compilation under some options), we should ship (and also sneakily include) a default waiver for it, so it compiles without warnings (the onus is on us to make sure it works, the user needs to be able to assume it's correct).

At the same time, we should not be responsible for providing waivers for libraries we do not bundle with the tool. (It should be easy enough for the user to do it themselves if they don't care about warnings from some code they imported from someone else, but there is no conceptual difference between UVM or your favourite collection of FIFOs here.)

I don't see the benefit of waiving only inside specific packages or modules. At least not for library files.

FWIW C++ has the same problem: I have to use some garbage non-standard third party library full of warnings, but I want -Wall -Werror. You can use `-isystem` instead of `-I` and the compilers will not report warnings in headers found through a `-isystem` path. If you want to make it super easy for the user, you could consider something like a `-group-warn-off`/`-group-warn-on` to bracket input files on the command line, but that of course does not cover files found with `-y` or `+incdir`, so `lint_off -rule * -file /my/grotty/lib/*` is probably the most reasonable way to make this super easy (maybe that works already?)

wsnyder, 2023-10-18:
This proposal is superseded by #4562.

closed 2023-10-18 by wsnyder