verilator/verilator #4502: Executing a `$fatal()` should stop the simulation with an exit code

tudortimi, 2023-09-18:
Executing a `$fatal()` should stop the simulation with an exit code. This is specified in the IEEE 1800 standard:

[image]

I'm using an older version, 5.011 devel rev v5.010-34-g444020f7c, but I didn't notice any activity related to this since then.

Is this something that would be reasonable for a beginner to implement?

wsnyder, 2023-09-18:
Fatal calls stop which unless you override it does give bad status.

```
module t;
   initial $fatal("bye");
endmodule
```

```
verilator --binary t.v
obj_dir/Va && echo ok || echo bad
[0] %Fatal: a.v:2: Assertion failed in TOP.t
%Error: /svaha/wsnyder/d/a.v:2: Verilog $stop
Aborting...
Aborted (core dumped)
bad
```

tudortimi, 2023-09-18:
I'll have to have a look in more detail what's going on in my setup. I'll have to admit I haven't tested it in isolation before opening the case.

tudortimi, 2023-09-18:
I had a look and the problem was that I was piping the executable produced by Verilator to `tee` to produce a log file. I learned the hard way that this doesn't propagate the exit code.

closed 2023-09-18 by tudortimi