verilator/verilator #4505: Write output of executable to log file

tudortimi, 2023-09-18:
Would it be possible to write the output of the executable produced by Verilator to a log file? I know that it's possible to pipe to `tee`, but this has a couple of drawbacks:

- it requires the user to take care to propagate the exit code of the executable, which can go wrong (see #4502)
- it merges stdout and stderr (though I'm not sure whether stderr is used)
- is more difficult to script, requiring the user to either call the executable from a shell and combine with `tee` or fiddle with forwarding stdout/stderr of the subprocess call

All other simulators have a way of specifying a log file. This can be optional unless the user activates, maybe with a plusarg similar to `+verilator+prof+exec+file+<filename>`.

wsnyder, 2023-09-25:
Getting logfiles is surprisingly complicated due to error handling and such; the only reliable method is to have a wrapper program that forks a child, which is a lot of work to maintain.  Also many/most test frameworks already have their own way of capturing output so this would be redundant for those.

But I see how this is useful, let me think about it for a while.

Meantime, see e.g. https://stackoverflow.com/questions/1221833/pipe-output-and-capture-exit-status-in-bash

tudortimi, 2023-09-25:
Asking just out of convenience, since you're very familiar with Perl: do you know how to do it using `system()` without the pipe? I'd like to avoid using shell in scripts, because it causes all sorts of problems with quoting and args with meta-characters. Can Perl redirect output streams as well?

wsnyder, 2023-09-25:
See Perl's exec().  It can take an array so no quoting is needed.