verilator/verilator #4507: Getting error on installation

Rosith111, 2023-09-19:
i was getting below error when ever i give command of sudo make install

making verilator in src
make -C src 
make[1]: Entering directory '/home/roshit/Tools/verilator/src'
make -C obj_dbg -j 1  TGT=../../bin/verilator_bin_dbg VL_DEBUG=1 -f ../Makefile_obj serial
make[2]: Entering directory '/home/roshit/Tools/verilator/src/obj_dbg'
make[2]: Nothing to be done for 'serial'.
make[2]: Leaving directory '/home/roshit/Tools/verilator/src/obj_dbg'
make -C obj_dbg       TGT=../../bin/verilator_bin_dbg VL_DEBUG=1 -f ../Makefile_obj
make[2]: Entering directory '/home/roshit/Tools/verilator/src/obj_dbg'
      Compile flags:  g++ -Og -ggdb -gz -DVL_DEBUG -D_GLIBCXX_DEBUG -DVERILATOR_INTERNAL_ -MMD -I. -I.. -I.. -I../../include -I../../include -MP -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC="" -DDEFENV_SYSTEMC_ARCH="" -DDEFENV_SYSTEMC_INCLUDE="" -DDEFENV_SYSTEMC_LIBDIR="" -DDEFENV_VERILATOR_ROOT="/home/roshit/Tools/verilator/build/share/verilator"
make[2]: Leaving directory '/home/roshit/Tools/verilator/src/obj_dbg'
make -C obj_dbg       TGT=../../bin/verilator_coverage_bin_dbg VL_DEBUG=1 VL_VLCOV=1 -f ../Makefile_obj serial_vlcov
make[2]: Entering directory '/home/roshit/Tools/verilator/src/obj_dbg'
make[2]: Nothing to be done for 'serial_vlcov'.
make[2]: Leaving directory '/home/roshit/Tools/verilator/src/obj_dbg'
make -C obj_dbg       TGT=../../bin/verilator_coverage_bin_dbg VL_DEBUG=1 VL_VLCOV=1 -f ../Makefile_obj
make[2]: Entering directory '/home/roshit/Tools/verilator/src/obj_dbg'
      Compile flags:  g++ -Og -ggdb -gz -DVL_DEBUG -D_GLIBCXX_DEBUG -DVERILATOR_INTERNAL_ -MMD -I. -I.. -I.. -I../../include -I../../include -MP -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC="" -DDEFENV_SYSTEMC_ARCH="" -DDEFENV_SYSTEMC_INCLUDE="" -DDEFENV_SYSTEMC_LIBDIR="" -DDEFENV_VERILATOR_ROOT="/home/roshit/Tools/verilator/build/share/verilator"
make[2]: Leaving directory '/home/roshit/Tools/verilator/src/obj_dbg'
make -C obj_opt -j 1  TGT=../../bin/verilator_bin -f ../Makefile_obj serial
make[2]: Entering directory '/home/roshit/Tools/verilator/src/obj_opt'
make[2]: Nothing to be done for 'serial'.
make[2]: Leaving directory '/home/roshit/Tools/verilator/src/obj_opt'
make -C obj_opt       TGT=../../bin/verilator_bin -f ../Makefile_obj
make[2]: Entering directory '/home/roshit/Tools/verilator/src/obj_opt'
      Compile flags:  g++ -O3 -DVERILATOR_INTERNAL_ -MMD -I. -I.. -I.. -I../../include -I../../include -MP -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC="" -DDEFENV_SYSTEMC_ARCH="" -DDEFENV_SYSTEMC_INCLUDE="" -DDEFENV_SYSTEMC_LIBDIR="" -DDEFENV_VERILATOR_ROOT="/home/roshit/Tools/verilator/build/share/verilator"
make[2]: Leaving directory '/home/roshit/Tools/verilator/src/obj_opt'
make[1]: Leaving directory '/home/roshit/Tools/verilator/src'
help2man --no-info --no-discard-stderr --version-string=- bin/verilator_gantt -o verilator_gantt.1
make: help2man: Command not found
Makefile:191: recipe for target 'verilator_gantt.1' failed
make: *** [verilator_gantt.1] Error 127

can anyone help to resolve this problem

wsnyder, 2023-09-19:
>make: help2man: Command not found

Install help2man.

See the installs needed in the documentation https://veripool.org/guide/latest/install.html

```
   # Prerequisites:
   #sudo apt-get install git help2man perl python3 make autoconf g++ flex bison ccache
   #sudo apt-get install libgoogle-perftools-dev numactl perl-doc
   #sudo apt-get install libfl2  # Ubuntu only (ignore if gives error)
   #sudo apt-get install libfl-dev  # Ubuntu only (ignore if gives error)
   #sudo apt-get install zlibc zlib1g zlib1g-dev  # Ubuntu only (ignore if gives error)
```

closed 2023-09-19 by wsnyder

Rosith111, 2023-09-22:
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python3 is already the newest version (3.6.7-1~ 18.04).
python3 set to manually installed.
The following packages were automatically installed and are no longer required:
  fonts-liberation2 fonts-opensymbol gir1.2-gst-plugins-base-1.0
  gir1.2-gstreamer-1.0 gir1.2-gudev-1.0 gir1.2-udisks-2.0
  grilo-plugins-0.3-base gstreamer1.0-gtk3 libboost-date-time1.65.1
  libboost-filesystem1.65.1 libboost-iostreams1.65.1 libboost-locale1.65.1
  libcdr-0.1-1 libclucene-contribs1v5 libclucene-core1v5 libcmis-0.5-5v5
  libcolamd2 libdazzle-1.0-0 libe-book-0.1-1 libedataserverui-1.2-2 libeot0
  libepubgen-0.1-1 libetonyek-0.1-1 libevent-2.1-6 libexiv2-14
  libfreerdp-client2-2 libfreerdp2-2 libgc1c2 libgee-0.8-2 libgexiv2-2
  libgom-1.0-0 libgpgmepp6 libgpod-common libgpod4 liblangtag-common
  liblangtag1 liblirc-client0 liblua5.3-0 libmediaart-2.0-0 libmspub-0.1-1
  libodfgen-0.1-1 libqqwing2v5 libraw16 librevenge-0.0-0 libsgutils2-2
  libssh-4 libsuitesparseconfig5 libvncclient1 libwinpr2-2 libxapian30
  libxmlsec1-nss lp-solve media-player-info python3-mako python3-markupsafe
  syslinux syslinux-common syslinux-legacy usb-creator-common
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  automake autotools-dev g++-7 gcc gcc-7 git-man libasan4 libatomic1
  libbison-dev libc-dev-bin libc6 libc6-dbg libc6-dev libcilkrts5
  liberror-perl libgcc-7-dev libgoogle-perftools4 libitm1 liblsan0 liblzma-dev
  liblzma5 libmpx2 libperl5.26 libquadmath0 libsigsegv2 libstdc++-7-dev
  libtcmalloc-minimal4 libtsan0 libubsan0 libunwind-dev libunwind8
  linux-libc-dev m4 manpages-dev perl-base perl-modules-5.26
Suggested packages:
  autoconf-archive gnu-standards autoconf-doc libtool bison-doc distcc
  build-essential flex-doc g++-multilib g++-7-multilib gcc-7-doc
  libstdc++6-7-dbg gcc-multilib gcc-doc gcc-7-multilib gcc-7-locales
  libgcc1-dbg libgomp1-dbg libitm1-dbg libatomic1-dbg libasan4-dbg
  liblsan0-dbg libtsan0-dbg libubsan0-dbg libcilkrts5-dbg libmpx2-dbg
  libquadmath0-dbg git-daemon-run | git-daemon-sysvinit git-doc git-el
  git-email git-gui gitk gitweb git-cvs git-mediawiki git-svn glibc-doc
  liblzma-doc libstdc++-7-doc m4-doc make-doc libterm-readline-gnu-perl
  | libterm-readline-perl-perl
The following NEW packages will be installed:
  autoconf automake autotools-dev bison ccache flex g++ g++-7 gcc gcc-7 git
  git-man help2man libasan4 libatomic1 libbison-dev libc-dev-bin libc6-dev
  libcilkrts5 liberror-perl libfl-dev libfl2 libgcc-7-dev
  libgoogle-perftools-dev libgoogle-perftools4 libitm1 liblsan0 liblzma-dev
  libmpx2 libquadmath0 libsigsegv2 libstdc++-7-dev libtcmalloc-minimal4
  libtsan0 libubsan0 libunwind-dev linux-libc-dev m4 make manpages-dev numactl
  perl-doc zlib1g-dev zlibc
The following packages will be upgraded:
  libc6 libc6-dbg liblzma5 libperl5.26 libunwind8 perl perl-base
  perl-modules-5.26 zlib1g
9 upgraded, 44 newly installed, 0 to remove and 276 not upgraded.
Need to get 62.0 MB of archives.
After this operation, 184 MB of additional disk space will be used.
Get:1 http://us.archive.ubuntu.com/ubuntu bionic-updates/main amd64 libperl5.26 amd64 5.26.1-6ubuntu0.7 [3,532 kB]
Get:2 http://us.archive.ubuntu.com/ubuntu bionic-updates/main amd64 perl amd64 5.26.1-6ubuntu0.7 [201 kB]
Get:3 http://us.archive.ubuntu.com/ubuntu bionic-updates/main amd64 perl-base amd64 5.26.1-6ubuntu0.7 [1,392 kB]
Get:4 http://us.archive.ubuntu.com/ubuntu bionic-updates/main amd64 perl-modules-5.26 all 5.26.1-6ubuntu0.7 [2,764 kB]
Get:5 http://us.archive.ubuntu.com/ubuntu bionic-updates/main amd64 libc6-dbg amd64 2.27-3ubuntu1.6 [5,162 kB]
Get:6 http://us.archive.ubuntu.com/ubuntu bionic-updates/main amd64 libc6 amd64 2.27-3ubuntu1.6 [2,831 kB]
Get:7 http://us.archive.ubuntu.com/ubuntu bionic-updates/main amd64 zlib1g amd64 1:1.2.11.dfsg-0ubuntu2.2 [56.9 kB]
Get:8 http://us.archive.ubuntu.com/ubuntu bionic/main amd64 libsigsegv2 amd64 2.12-1 [14.7 kB]
Get:9 http://us.archive.ubuntu.com/ubuntu bionic/main amd64 m4 amd64 1.4.18-1 [197 kB]
Get:10 http://us.archive.ubuntu.com/ubuntu bionic/main amd64 flex amd64 2.6.4-6 [316 kB]
Get:11 http://us.archive.ubuntu.com/ubuntu bionic-updates/main amd64 liblzma5 amd64 5.2.2-1.3ubuntu0.1 [91.1 kB]
Get:12 http://us.archive.ubuntu.com/ubuntu bionic/main amd64 autoconf all 2.69-11 [322 kB]
Get:13 http://us.archive.ubuntu.com/ubuntu bionic/main amd64 autotools-dev all 20180224.1 [39.6 kB]
Get:14 http://us.archive.ubuntu.com/ubuntu bionic/main amd64 automake all 1:1.15.1-3ubuntu2 [509 kB]
Get:15 http://us.archive.ubuntu.com/ubuntu bionic/main amd64 libbison-dev amd64 2:3.0.4.dfsg-1build1 [339 kB]
Get:16 http://us.archive.ubuntu.com/ubuntu bionic/main amd64 bison amd64 2:3.0.4.dfsg-1build1 [266 kB]
Get:17 http://us.archive.ubuntu.com/ubuntu bionic/main amd64 ccache amd64 3.4.1-1 [110 kB]
Get:18 http://us.archive.ubuntu.com/ubuntu bionic-updates/main amd64 libitm1 amd64 8.4.0-1ubuntu1~ 18.04 [27.9 kB]
[... 796 lines cut ...]
mkdir /home/roshit/Tools/verilator/build/share/verilator/examples/cmake_tracing_sc
/bin/sh ./src/mkinstalldirs /home/roshit/Tools/verilator/build/share/verilator/examples/cmake_protect_lib
mkdir /home/roshit/Tools/verilator/build/share/verilator/examples/cmake_protect_lib
/bin/sh ./src/mkinstalldirs /home/roshit/Tools/verilator/build/share/verilator/examples/xml_py
[... cut ...]
/usr/bin/install -c -m 644 verilator-config-version.cmake /home/roshit/Tools/verilator/build/share/verilator

Installed binaries to /home/roshit/Tools/verilator/build/bin/verilator
Installed man to /home/roshit/Tools/verilator/build/share/man/man1
Installed examples to /home/roshit/Tools/verilator/build/share/verilator/examples

For documentation see 'man verilator' or 'verilator --help'
For forums and to report bugs see https://verilator.org

install.sh: line 20: verilator: command not found

after adding those files also it is not installing no errors shown can u help me

wsnyder, 2023-09-23:
>install.sh: line 20: verilator: command not found

That isn't coming from the verilator installation, but presumably something you wrote.

Maybe verilator isn't in your PATH?

Rosith111, 2023-09-25:
yeah it is installed but in the root terminal it was not shown

is there any changes to do for getting

Rosith111, 2023-09-25:
it is shown in user terminal