verilator/verilator #4510: Update compiler flags via `--compiler`

poemonsense, 2023-09-21:
Thanks for taking the time to report this.

**What would you like added/supported?**

Currently Verilator supports the `--compiler` flag to indicate a compiler preference.
However, this option affects only the C++ code style as in V3Options.cpp without changing the compiler flags which seem to be configured at Verilator compile time.
This would cause compiling errors because some flags of clang++ are not accepted by g++, and the vice versa.
For example, I'm using g++ to build a design with `--timing`, with the Verilator configured by default using clang++.
If I override the make scripts with `CC=gcc CXX=g++ LINK=g++`, it returns errors as follows.
```
g++-10: error: unrecognized command-line option ‘-Qunused-arguments’
g++-10: error: unrecognized command-line option ‘-fbracket-depth=4096’
g++-10: error: unrecognized command-line option ‘-fbracket-depth=4096’
g++-10: error: unrecognized command-line option ‘-fcoroutines-ts’; did you mean ‘-fcoroutines’?
```

Actually I don't know the workaround except for configuring the Verilator with g++.

Can you attach an example that runs on other simulators?  (Must be openly licensed, ideally in test_regress format.)
N/A

**May we assist you in trying to fix this in Verilator yourself?**
Sure. If you could provide some help on how this could be fixed, I'm glad to implement it.

wsnyder, 2023-09-23:
The ways to fix this are to:
1. Compute all flags at runtime. This is slow and an ugly rework.  One alternative is to use the cmake framework instead of make.
2. Compute flags for e.g. gcc and clang *both* at configure time.  This won't work if changing compiler version.  Also configure has no realistic clean way to do this.

So I don't see a effort/maintainance-realistic way to support this, and it's likely something done by only sophisticated users that can hopefully handle having two installations with what is needed.  Not fixing, sorry.

wsnyder, 2023-09-23:
P.S. Might have not been clear from above: using the cmake framework is supported by Verilator at runtime, so might do what you want.

closed 2023-09-23 by wsnyder

poemonsense, 2023-09-24:
It’s great to see cmake could address this issue. Didn’t expect different functionalities between make and cmake. I agree it’s not practically realistic to generate the flags if this is the case. It’s just a minor issue. 

Hopefully we have another project using cmake as the verilator caller script. So I’ll have a try on this and considering switch our makefiles to cmakelists. Thanks.