verilator/verilator #4514: forceable signal can not be forced when --coverage-toggle option is activated

fredrequin, 2023-09-21:
Hello,

in our project, we use Verilator v4.219. We do not plan to move to v5 for the moment.
Internaly, we use a COCOTB-like stimuli language that got translated into C++.
We do have a"signal_force" keyword that uses Verilator's forceable feature.
We have noticed that when the --coverage-toggle option is activated, the signal does not get forced anymore.
That seems quite logical that these two features generate a conflict.
For the moment, we plan to manually exclude the forceable signal from the coverage.
It might be useful to take into account this corner case inside Verilator and to generate a warning.

Frédéric Requin

wsnyder, 2023-09-25:
I don't see why coverage is breaking forcable, I worry there's a larger problem there (unrelated to mixing the two), so think it needs more debug vs just erroring out.

fredrequin, 2023-09-29:
Hello,
after carefully looking at how we use __VforceEn, __VforceVal and __VForceRd, I think the mistake is on our side.
In our stimuli language, we mimic ModelSim commands "signal_freeze" and "signal_deposit" thanks to Verilator's forceable feature.
We were wrongly using __VForceRd for "signal_deposit" which seemed to work due to some scheduling luck.
Now, we use __VforceEn and __VforceVal then wait until __VForceRd gets the correct value.
First tests seem OK.
Sorry for the false alarm.
Frédéric REQUIN

closed 2023-09-29 by wsnyder