verilator/verilator #4517: Support changeable Python path or add explanation in manual

PaperL, 2023-09-23:
I want to run a simulation with Cocotb and Verilator.
Firstly, I followed the instructions in [the manual](https://verilator.org/guide/latest/install.html#install-into-a-specific-prefix) and installed Verilator. Secondly, I created a new Conda environment and installed Cocotb. Then I tried to run the [demo of cocotb](https://github.com/cocotb/cocotb#usage). When I `make SIM=verilator`, I got:
```bash
......
/usr/bin/python3 /tools/verilator/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include Vtop.cpp Vtop___024root__DepSet_h84412442__0.cpp Vtop___024root__DepSet_heccd7ead__0.cpp Vtop__Dpi.cpp Vtop___024root__Slow.cpp Vtop___024root__DepSet_h84412442__0__Slow.cpp Vtop___024root__DepSet_heccd7ead__0__Slow.cpp Vtop__Syms.cpp > Vtop__ALL.cpp
Fatal Python error: Py_Initialize: Unable to get the locale encoding
ModuleNotFoundError: No module named 'encodings'
......
```
It was really confusing and I found that I cannot use any Conda environment to run the code. So I tracked the build process and found that a Python3 path is hardcoded in `$(VERILATOR_ROOT)/include/verilated.mk` like `PYTHON3 = /usr/bin/python3`.

So I'd like to ask why the Python3 path needs to be hardcoded when installing Verilator and whether Verilator will support Conda virtual environment in the future.
And I think the current manual should remind users that changing the Python executable asks for a rebuild of Verilator.

wsnyder, 2023-09-23:
You can override this at runtime with an environment variable, just e.g. 

    export PYTHON3=/path/to/python3 
    
note that's true of any well-written makefile, not just Verilator's.

It's a reasonable ask this be documented in https://veripool.org/guide/latest/environment.html I'll add it.

gatk555, 2023-10-30:
Nice resonse, but unfortunately Verilator is not using a "well-written makefile".  Looking in today's source:
./include/verilated.mk.in:PYTHON3 = @PYTHON3@

In a Debian package becomes: that becomes:
PYTHON3 = /usr/bin/python3

or in MSYS:
PYTHON3 = /mingw64/bin/python3

So the environment variable is overidden and there is no obvious way to have Verilator run the makefile with "-e" to prevent that.  It is easily fixed by changing the assignment to the "?=" form.  Do I need to fork and create a PR for that to happen?

My thanks go to all involved for a great piece of software.

Giles

wsnyder, 2023-10-30:
The ?= existed for a while but were removed, see #4549 

The specific issue there was with CC/LD not PYTHON, but we could likewise get poisoned from upstream environment.  Perhaps we need a VERILATOR_PYTHON?

gatk555, 2023-10-30:
I found the reversion a little odd, in that the only real reason seems to have been that CXX does not work because it is pre-defined in gmake.  To me that just points to a need for an additional change: use a different variable for the compiler in verilate.mk and override the default .cpp.o rule to make use of it.

Unexpected effects from an accidentally poisonous environment seem a separate issue.  There could be a command option to pass "-e" to make, so that you need to ask explicitly for overrides.

In any case my enthusiasm for any change on these lines has waned, as I found that using a Windows-native python3 does not work.  It seems to treat the MSYS-style path's root as "c:\Program Files\git".  Beats me.

But it seems that some change should be made to Verilator, as the documentation currently describes something that does not work.

Thanks,

Giles

wsnyder, 2023-11-01:
I updated the documentation, given environment vars don't work, you should still be able to pass these to make to override (`make PYTHON3=foo`).

closed 2023-11-01 by wsnyder