verilator/verilator #4519: Can't generate correct tex name

cheungxi, 2023-09-26:
When i run `make pdf` in docs directory,  output text tell me **can't find _build/latex/verlateor.pdf** .

So, I started looking for the reason.

I tried
```bash
sphinx-build -M latex "guide" "_build" -c guide
```
This is based on the docs/Makefile line73. 
```
        $(SPHINXBUILD) -M $@ "$(SOURCEDIR)" "$(BUILDDIR)" $(SPHINXOPTS)
```

after run this command, this will generate Verilator.tex (not verilator.tex) in **docs/build/latex**

so can't correct generate pdf name.

wsnyder, 2023-09-26:
What version of sphynx are you using?  This works for me.

```
sphinx-build -M html "guide" "_build" -c guide
Running Sphinx v5.2.2
```

cheungxi, 2023-09-27:
> What version of sphynx are you using? This works for me.
> 
> ```
> sphinx-build -M html "guide" "_build" -c guide
> Running Sphinx v5.2.2
> ```

Ok, I retested. 

when use sphinx v3.4.3 and v.5.3.0 can correctly generate tex file name (verilator.tex).

when use sphinx v1.8.4 generated tex file name is **V**erilator.tex (uppercase V).

wsnyder, 2023-09-29:
I think the answer will be use the recent version.

closed 2023-10-04 by wsnyder