verilator/verilator #4521: $bits supported in creation of union?

polyee13, 2023-09-26:
Thanks for taking the time to report this.
I see $bits is supported in certain contexts per https://verilator.org/guide/latest/languages.html#systemverilog-2005-ieee-1800-2005-support

I assume $bits isn't supported in the context below?

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

Seems like there is an issue in interpreting a struct that part of a union. 
I took the example_c and inserted a packed union, which contains a single packed struct. 

`module top
  ( input logic clk
  , input logic rst
  , output logic out1
  );

  localparam WORD_WIDTH = 16;

   initial begin
      $display("Hello World!");
      $finish;
   end

   typedef union packed {
     struct packed {
     logic [15:0]  temp_1 ;
     logic [15:0]  temp_2 ;
     logic [15:0]  temp_3 ;
   } regs ;
   logic [($bits(regs)/WORD_WIDTH)-1:0] [WORD_WIDTH-1:0] words ;
   } my_union_t ;

   my_union_t my_union;
endmodule`

This is the error. I'm only verilating as you can see.
-- Verilator hello-world simple example
-- VERILATE --------
verilator -cc -j top.sv sim_main.cpp
%Error: top.sv:28:18: Can't find definition of variable: 'regs'
   28 |    logic [($bits(regs)/WORD_WIDTH)-1:0] [WORD_WIDTH-1:0] words ;
      |                  ^~~~
%Error: Exiting due to 1 error(s)
Makefile:36: recipe for target 'default' failed
make: *** [default] Error 1

This is a snippet from a known working version of a design that's synthesized with Vivado 2022, and simulated with VCS 2021.
If I exclude $bits and hard-code the number of bits contained in the struct, 48, it compiles.
i.e. logic [2:0] [15:0] words ;

What 'verilator --version' are you using?  Did you try it with the git master version? I checkout out stable branch
Verilator 5.016 2023-09-16 rev v5.014-149-g57c816f90

What OS and distribution are you using?
Ubuntu 18.04 Kernel 4.15

May we assist you in trying to fix this in Verilator yourself?
Sure.

wsnyder, 2023-09-26:
A typedef is legal in $bits but as you noted this seems to fail. It also fails on one of the 4 other simulators I tried.

Perhaps you'd be willing to attempt the fix in a pull request?  First, convert your test to test_regress format, and see it fail - also please assert that the returned $bits is the correct value.  V3LinkDot resolves these, if you use "--debug --debugi-V3LinkDot 9" you'll see it finds a typedef when expecting a variable.  Change the code to also accept a typedef.  Then see the test pass.

polyee13, 2023-09-29:
If this is an issue with Verilator itself, I am not familiar with debugging this type of application, since I mainly work on FPGA's and code SystemVerilog.

polyee13, 2023-12-04:
@wsnyder anyone else willing to take this on?

wsnyder, 2023-12-05:
@polyee13 not sure the question you're asking. Note contributors select themselves what they work on.

polyee13, 2023-12-19:
@wsnyder 
Ok, please help clarify a few things and I'll try and tackle it. 
You mentioned
"convert your test to test_regress format, and see it fail" --> What is the test_regress format?
"assert that the returned $bits is the correct value" --> Using V3LinkDot as you mentioned, but what is that?
Is there a tutorial available or another PR I can reference that illustrates the debugging methodology. 
Again, I come from purely an FPGA background, so I only debug SV code, and sometimes C.

wsnyder, 2023-12-19:
>Ok, please help clarify a few things and I'll try and tackle it.

Great

>"convert your test to test_regress format, and see it fail" --> What is the test_regress format?

See docs/internals.rst, also for how to debug

>"assert that the returned $bits is the correct value"

This was referring to making the Verilog test self-checking like most of the other tests.  You can then make sure passes on other simulators.

>V3LinkDot as you mentioned, but what is that?

src/V3LinkDot.cpp is likely what needs fixing.