verilator/verilator #4524: Support for names structure fields in FST dumps

margej, 2023-09-29:
I don't know if this is possible or not within the constraints of the FST format, or whether this is a limitation of Verilator's FST generation, but where RTL contains a variable which is an instance of a user defined structure, it would be very helpful if the FST dump contained individual names signals for each structure element/field instead of just a flat vector.

Further complications include:
- Nested structures
- Unions
- Arrays of structures
- Arrays of structures within other other structures
etc. etc.

So this would possibly be a significant amount of effort?
I was assuming if not directly supported, a struct variable could be represented in the dump as a pseudo module in the hierarchy, perhaps allowing even VCD format to support this?

Appologies if this has been raised before, a quick search didn't show any similar issues.

wsnyder, 2023-09-29:
For VCD there is no way to say an e.g. bit is a piece of a larger bus, so the only alternative would be to expand every single one, including expanding all of the cycle by cycle data.  E.g. a packed struct of 20 elements would take about 20x the space and be 20x slower.  Probably undesirable.

For FST I don't know if struct fields can be efficiently represented or not (e.g. have FST know that certain bits are part of a struct and see the field or entire struct at view time without storing data inefficiently).  Please check with [gtkwave](https://github.com/gtkwave/gtkwave).   If not, ask or contribute having it added there.  Once in GTKwave come back here and please consider making a pull request to add support to Verilator.

sifferman, 2024-02-24:
Is this different from what is provided by `--trace-structs`

wsnyder, 2024-02-24:
trace-structs splits the members into separate variables, the FST created looses all concept of the structure container, as there is no way to represent it (this issue).

jjones-github, 2024-02-29:
Some small amount of feedback [here](https://github.com/gtkwave/gtkwave/issues/312). I am working on a comparison with `vcs` and `fsdb` + `verdi` vs `verilator` and `fst` + `gtkwave.

margej, 2024-07-16:
Looking at [this](https://blog.timhutt.co.uk/fst_spec/) unofficial FST spec, it does appear to support structures (and records in VHDL) via specific scope types FST_ST_VCD_STRUCT and FST_ST_VHDL_RECORD

If I understand the format correctly, it may be possible to represent a var of type struct by entering it in to the "heirarchy block" as a new FST_ST_VCD_SCOPE record, with type = FST_ST_VCD_STRUCT, and populating it with the individual fields (this sounds like it could used realtively easily in conjustion with --trace-structs to povide some sort of container information)

I'm also wondering if so called "structural aliases" can be used to associate those individual fields back to a single bit-vector type variable. This would be ideal since it wouldn't present additional run-time dumping overhead of dumping each field individually (which is presumably what --trace-structs has to do)

When I find some time I'll have a look into whether I can get the GTKWave's FST writer module (see $VERILATOR_ROOT/include/gtkwave/fstapi.h) to produce something usable.

This basic example might be a good starting point for anyone interested in doing the same:
https://github.com/bluecmd/fst-example/blob/master/example.c