verilator/verilator #4526: Verilator seemingly using more than one thread even with `--threads 1`

gussmith23, 2023-09-29:
I suspect I'm invoking Verilator incorrectly here. I'm Verilating two designs, and then compiling the results together with a testbench and then running.

Here's my Makefile:
```Makefile
.PHONY: build

VERILATOR=verilator

simulate: /home/gus/lakeroad-evaluation/out/robustness_experiments/mult_0_stage_signed_8_bit/lakeroad_xilinx_ultrascale_plus/verilator/testbench /home/gus/lakeroad-evaluation/out/robustness_experiments/mult_0_stage_signed_8_bit/lakeroad_xilinx_ultrascale_plus/verilator/testbench_inputs.txt
	/home/gus/lakeroad-evaluation/out/robustness_experiments/mult_0_stage_signed_8_bit/lakeroad_xilinx_ultrascale_plus/verilator/testbench < /home/gus/lakeroad-evaluation/out/robustness_experiments/mult_0_stage_signed_8_bit/lakeroad_xilinx_ultrascale_plus/verilator/testbench_inputs.txt

/home/gus/lakeroad-evaluation/out/robustness_experiments/mult_0_stage_signed_8_bit/lakeroad_xilinx_ultrascale_plus/verilator/verilator_obj_dir/ground_truth_module/Vground_truth_module__ALL.a: /home/gus/lakeroad-evaluation/out/robustness_experiments/mult_0_stage_signed_8_bit/lakeroad_xilinx_ultrascale_plus/verilator/verilator_obj_dir/ground_truth_module/Vground_truth_module.mk

/home/gus/lakeroad-evaluation/out/robustness_experiments/mult_0_stage_signed_8_bit/lakeroad_xilinx_ultrascale_plus/verilator/verilator_obj_dir/test_module/Vtest_module__ALL.a: /home/gus/lakeroad-evaluation/out/robustness_experiments/mult_0_stage_signed_8_bit/lakeroad_xilinx_ultrascale_plus/verilator/verilator_obj_dir/test_module/Vtest_module.mk

/home/gus/lakeroad-evaluation/out/robustness_experiments/mult_0_stage_signed_8_bit/lakeroad_xilinx_ultrascale_plus/verilator/testbench: /home/gus/lakeroad-evaluation/out/robustness_experiments/mult_0_stage_signed_8_bit/lakeroad_xilinx_ultrascale_plus/lakeroad_result.sv /home/gus/lakeroad-evaluation/robustness-testing-verilog-files/generated/mult_0_stage_signed_8_bit.sv /home/gus/lakeroad-evaluation/out/robustness_experiments/mult_0_stage_signed_8_bit/lakeroad_xilinx_ultrascale_plus/verilator/testbench.cc
	mkdir -p /home/gus/lakeroad-evaluation/out/robustness_experiments/mult_0_stage_signed_8_bit/lakeroad_xilinx_ultrascale_plus/verilator/verilator_obj_dir/test_module
	mkdir -p /home/gus/lakeroad-evaluation/out/robustness_experiments/mult_0_stage_signed_8_bit/lakeroad_xilinx_ultrascale_plus/verilator/verilator_obj_dir/ground_truth_module
	$(VERILATOR) --threads 1 --no-timing --Mdir /home/gus/lakeroad-evaluation/out/robustness_experiments/mult_0_stage_signed_8_bit/lakeroad_xilinx_ultrascale_plus/verilator/verilator_obj_dir/test_module --cc /home/gus/lakeroad-evaluation/out/robustness_experiments/mult_0_stage_signed_8_bit/lakeroad_xilinx_ultrascale_plus/lakeroad_result.sv --prefix Vtest_module -I/home/gus/lakeroad-evaluation/lakeroad-private/DSP48E2 -DXIL_XECLIB -Wno-UNOPTFLAT -Wno-LATCH -Wno-WIDTH -Wno-STMTDLY -Wno-CASEX -Wno-TIMESCALEMOD -Wno-PINMISSING
	$(VERILATOR) --threads 1 --no-timing --Mdir /home/gus/lakeroad-evaluation/out/robustness_experiments/mult_0_stage_signed_8_bit/lakeroad_xilinx_ultrascale_plus/verilator/verilator_obj_dir/ground_truth_module --cc /home/gus/lakeroad-evaluation/robustness-testing-verilog-files/generated/mult_0_stage_signed_8_bit.sv --prefix Vground_truth_module -I/home/gus/lakeroad-evaluation/lakeroad-private/DSP48E2 -DXIL_XECLIB -Wno-UNOPTFLAT -Wno-LATCH -Wno-WIDTH -Wno-STMTDLY -Wno-CASEX -Wno-TIMESCALEMOD -Wno-PINMISSING
	make -C /home/gus/lakeroad-evaluation/out/robustness_experiments/mult_0_stage_signed_8_bit/lakeroad_xilinx_ultrascale_plus/verilator/verilator_obj_dir/test_module -f Vtest_module.mk
	make -C /home/gus/lakeroad-evaluation/out/robustness_experiments/mult_0_stage_signed_8_bit/lakeroad_xilinx_ultrascale_plus/verilator/verilator_obj_dir/ground_truth_module -f Vground_truth_module.mk
	$(CXX) -o /home/gus/lakeroad-evaluation/out/robustness_experiments/mult_0_stage_signed_8_bit/lakeroad_xilinx_ultrascale_plus/verilator/testbench -I/home/gus/verilator/include /home/gus/lakeroad-evaluation/out/robustness_experiments/mult_0_stage_signed_8_bit/lakeroad_xilinx_ultrascale_plus/verilator/testbench.cc /home/gus/lakeroad-evaluation/out/robustness_experiments/mult_0_stage_signed_8_bit/lakeroad_xilinx_ultrascale_plus/verilator/verilator_obj_dir/ground_truth_module/Vground_truth_module__ALL.a /home/gus/lakeroad-evaluation/out/robustness_experiments/mult_0_stage_signed_8_bit/lakeroad_xilinx_ultrascale_plus/verilator/verilator_obj_dir/test_module/Vtest_module__ALL.a /home/gus/verilator/include/verilated.cpp /home/gus/verilator/include/verilated_threads.cpp -lpthread
```

Sorry it's messy; it's automatically generated.

When I run the resulting testbench, I see it hovers at about 120% CPU, implying it's using more than one thread (despite `--thread 1`).

The testbench also runs MUCH slower than it should overall -- it's doing about 250 simulations of a very small design, but taking seconds (up to a minute) to run.

Furthermore, I looked at the flamegraph of what was happening:
[image]

We can see most of the time is spent in threading constructs.

Am I calling Verilator incorrectly here? Why is it using threads?

Thanks so much!

wsnyder, 2023-09-29:
See https://veripool.org/guide/latest/verilating.html#multithreading

With --threads 1, each model can use one thread.  You have two models though, so two threads, and there must be some parallelism happening as you are getting 120%.  The verilated common libraries are still thread safe, so locks are needed in some places.  Use numactl as described in the documentation and you'll get a massive speedup.

Alternatively using separate VerilatedContext's will remove most all the common locking as your two models are no longer sharing anything.

Closing as answered but feel free to follow up.

closed 2023-09-29 by wsnyder

gussmith23, 2023-09-29:
Thanks Wilson. I missed the note about numactl! I didn't even know I could use separate VerilatedContexts -- that seems like the easiest change! As always appreciate the quick response, and love the tool!!

gussmith23, 2023-09-30:
hm, I just checked -- I'm already using separate contexts. Will try the numactl thing later. 

Is there any way to **guarantee** single-threaded mode?

gussmith23, 2023-10-02:
Bumping my above question @wsnyder!

gussmith23, 2023-10-02:
I'm not quite sure how best to use numactl -- all of my cores are on the same socket, so it's not clear to me that choosing specific cores will make things better! Furthermore, I'm trying to run a lot of simulations all at once (in different processes), and I'd rather not have to manually orchestrate all the processes with numactl.

Is there any way to just run things in a purely single-threaded way? It seems like this was more possible with old Verilator -- is that no longer an option? 

To further convey the state I'm in, here's how long my simulation is taking:
```
6.10s user 32.53s system 118% cpu 32.630 total
```
If I'm reading this right, the`system` category is probably capturing all the synchronization time...that's a lot of time spent synchronizing!

Maybe it would also help to post my testbench:
```c++
// Testbench file which tests two Verilated designs against each other: a
// ground-truth design and a design under test.

#include <stdio.h>
#include <cassert>
#include "verilated.h"
#include "/home/gus/lakeroad-evaluation/out/robustness_experiments/mult_0_stage_signed_8_bit/lakeroad_xilinx_ultrascale_plus/verilator/verilator_obj_dir/test_module/Vtest_module.h"
#include "/home/gus/lakeroad-evaluation/out/robustness_experiments/mult_0_stage_signed_8_bit/lakeroad_xilinx_ultrascale_plus/verilator/verilator_obj_dir/ground_truth_module/Vground_truth_module.h"
#include <inttypes.h>

// User must provide the implementations of these macros. In Python, it likely
// looks something like:
// ```python
// " ".join(
//        [f"module->{name}=input[{i}];" for i, (name, _) in enumerate(module_inputs)]
//    )
// ```
// (and similarly for clock signal)
//
// This currently assumes that the input signals and clock signals have the same
// name. This isn't necessarily the case.
#define SET_MODULE_INPUTS(module, inputs) module->a=inputs[0]; module->b=inputs[1];
#define SET_MODULE_CLOCK(module, clock) module->None=clock;

#define NUM_INPUTS 2
#define INITIATION_INTERVAL 0

int main(int argc, char** argv) {

  srand(0);

  int num_module_args, num_test_cases;
  assert(scanf("%d %d\n", &num_module_args, &num_test_cases) == 2);

  uint64_t inputs[NUM_INPUTS];

  for (int test_case_i = 0; test_case_i < num_test_cases; test_case_i++) {
    
    VerilatedContext* contextp_test = new VerilatedContext;
    VerilatedContext* contextp_ground_truth = new VerilatedContext;
    Vtest_module* test_module = new Vtest_module{contextp_test};
    Vground_truth_module* ground_truth_module = new Vground_truth_module{contextp_ground_truth};
    //contextp->commandArgs(argc, argv);

    // Parse inputs from command line.
    printf("inputs:");
    for (int i = 0; i < NUM_INPUTS; i++) {
      assert(scanf("%" SCNu64, &inputs[i]) == 1);
      printf(" %" PRIu64, inputs[i]);
    }
    printf("\n");

#if INITIATION_INTERVAL == 0
    SET_MODULE_INPUTS(test_module, inputs)
    test_module->eval();
    SET_MODULE_INPUTS(ground_truth_module, inputs)
    ground_truth_module->eval();
#else
    // CLOCK CYCLE 1

    // First half of clock cycle 1: clk=0, inputs=desired inputs.
    SET_MODULE_INPUTS(test_module, inputs)
    SET_MODULE_CLOCK(test_module, 0)
    SET_MODULE_INPUTS(ground_truth_module, inputs)
    SET_MODULE_CLOCK(ground_truth_module, 0)

    test_module->eval();
    ground_truth_module->eval();

    // Second half of clock cycle 1: clk=1, inputs=random values/0.
    if (true == true) {
      for (int i = 0; i < NUM_INPUTS; i++) {
        inputs[i] = rand();
      }
    } else {
      for (int i = 0; i < NUM_INPUTS; i++) {
        inputs[i] = 0;
      }
    }

    SET_MODULE_CLOCK(test_module, 1)
    SET_MODULE_CLOCK(ground_truth_module, 1)
    
    test_module->eval();
    ground_truth_module->eval();
    
    // CLOCK CYCLE N>1

    for (int clock_cycle_i = 1; clock_cycle_i < INITIATION_INTERVAL; clock_cycle_i++) {

        // Either zero out inputs or use random inputs.
      if (true == true) {
        for (int i = 0; i < NUM_INPUTS; i++) {
          inputs[i] = rand();
        }
      } else {
        for (int i = 0; i < NUM_INPUTS; i++) {
          inputs[i] = 0;
        }
      }

      SET_MODULE_INPUTS(test_module, inputs)
      SET_MODULE_INPUTS(ground_truth_module, inputs)
      SET_MODULE_CLOCK(test_module, 0)
      SET_MODULE_CLOCK(ground_truth_module, 0)

      test_module->eval();
      ground_truth_module->eval();
      
      SET_MODULE_INPUTS(test_module, inputs)
      SET_MODULE_INPUTS(ground_truth_module, inputs)
      SET_MODULE_CLOCK(test_module, 1)
      SET_MODULE_CLOCK(ground_truth_module, 1)
      
      test_module->eval();
      ground_truth_module->eval();
      
    }
#endif

  printf("actual, expected = %lu, %lu\n", test_module->out, ground_truth_module->out);
  assert(test_module->out == ground_truth_module->out);

  delete ground_truth_module;
  delete test_module;
  delete contextp_test;
  delete contextp_ground_truth;

  }

}
```

gussmith23, 2023-10-02:
BTW, numactl does speed things up:

```
 3.95s user 17.68s system 109% cpu 19.720 total
```

but again, numactl is really non-ideal when I'm trying to run a bunch of simulations on a large machine. Having to assign each one to specific processors requires me to track which simulations I've put on which processors.

Thanks again for whatever help you can give me!

wsnyder, 2023-10-03:
>Is there any way to guarantee single-threaded mode? 

No, unfortunately we removed that a while ago to simplify the code.

In theory the completely separate Contexts were expected to completely fix that issue.  We'd certainly accept fixes to clean up any locks to make it possible to have your separate simulations completely decoupled.  (Of course it will still run faster on separate cores, but at least you won't get killed by lock contention.)

>but again, numactl is really non-ideal when I'm trying to run a bunch of simulations on a large machine. Having to assign each one to specific processors requires me to track which simulations I've put on which processors.

I completely agree, unfortunately the Linux kernel doesn't seem to give us any alternative that works.

gussmith23, 2023-10-03:
@wsnyder thanks for the info. Any suggestions on how to debug where the locks are getting held? I would certainly look into it if you had any leads!

wsnyder, 2023-10-03:
Use "gdb" and put a breakpoint where include/* calls std::this_thread::yield(), then once those are cleaned up VL_CPU_RELAX().