verilator/verilator #4527: Further clarification of Multi-Core support

jordankrim, 2023-09-29:
Since nothing changed (i.e. my test still fails), I want to be able to verify that Verilator is doing what I asked it to do. My HW (model)/DUT) has 4 cores and my VM (Linux) I am running it on has 8 cores.
My Verilator compile line is this:
verilator --cc --exe --trace --threads 4 --trace-threads 4 --prof-exec
I invoked my verilated simulation with:
numactl -m 0 -C 0,1,2,3 -- <verilated simulation which is the same as before without the numactl ahead of it>
Am I really running on 4 cores and how do I verify that?

wsnyder, 2023-09-30:
I should have put in other thread you can still message to followup on closed issues.

Anyhow "cat /proc/cpuinfo" will show you your CPU numbers, ideally you want different physical cores.

Note you need at least 5 threads because tracing is a thread.  There's no harm in extra -C's.

    numactl -m 0 -C 0,1,2,3,4,5,6,7 -- 

Verilator will complain if two threads land on same CPU within a single simulation (in your case with two it doesn't know between them).

closed 2023-09-30 by wsnyder

jordankrim, 2023-10-02:
I want to clarify the problem because I think while I was able to get working your answer using numactl, it isn't exactly where the problem is.   I think the numactl addresses the problem of Verilator running on the Host number of CPUs.  What I mean by this is I am running on an Oracle Box VM which I have set up to have 8 Processors (CPUs) (or threads ?).   Verilator is running on these Host CPUs so that is where the numactl is supposed to help.

My question/problem is not on the host.  I am saying that we want to launch multiple threads on the target design (i.e. DUT/SOC/FPGA etc).  This is where I am trying to determine whether Verilator is working correctly.  As I stated before I am not sure why my simulation is failing.  I am trying to determine whether it is Verilator or some problem in my testbench (harness) that is around my DUT.