verilator/verilator #4528: very simple test generating error

muzafferkal, 2023-10-01:
I'm using the following command line:
----------------------------------
verilator -sv  --debug --cc --timing --binary --trace-fst -Wno-INITIALDLY --trace-depth 0 --top add_tb --timescale 1ns/100ps add_tb.sv
Starting Verilator 5.017 devel rev v5.014-145-g44e7d2ebe
----------------------------------

I get the following error:
----------------------------------
%Error: ulimit -s unlimited 2>/dev/null; exec setarch --addr-no-randomize /usr/local/bin/verilator_bin_dbg -sv --debug --cc --timing --binary --trace-fst -Wno-INITIALDLY --trace-depth 0 --top add_tb --timescale 1ns/100ps add_tb.sv
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec setarch --addr-no-randomize /usr/local/bin/verilator_bin_dbg -sv --debug --cc --timing --binary --trace-fst -Wno-INITIALDLY --trace-depth 0 --top add_tb --timescale 1ns/100ps add_tb.sv
make: *** [Makefile:9: add] Error 255
----------------------------------

Also the line which says // this should also work breaks compile, not sure why. 
Could you please let me know what I'm doing wrong?

Here is my code:
----------------------------------
// adder trial testbench

module FA(input  logic a, b, ci,
          output logic s, co);

    logic [1:0] sum = a + b + ci;
    assign s = sum[0];
    assign co = sum[1];
endmodule

module rca #(parameter W = 4) 
            (input [W-1:0] a, [W-1:0] b,
             output logic [W:0] s);

    logic [W:0] c;
    assign c[0] = 0;
    genvar i;
    generate
            for (i=0; i < W-1; i = i+1) begin
                FA uadd(.a(a[i]), .b(b[i]), .s(s[i]), .ci(c[i]), .co(c[i+1]));
            end
    endgenerate
    assign s[W] = c[W];
endmodule

program test_add #(W=4) (
    input clk,
    output logic arst_n,
    output [W-1:0]          a,
    output [W-1:0]          b,
    input  [W:0]            s
    );
    clocking cb_add @(posedge clk);
        default input #100ps output #100ps;
        // input act_rdy, wgt_rdy, out_vld, out_dat;
        //output act_dat, act_vld, wgt_vld, wgt_adr, wgt_dat, out_rdy;
    endclocking

    logic [W-1:0] ai, bi;
    logic [W:0] sum;
    int err = 0;
    logic [W:0] errb;
    initial begin
        arst_n = 0;
        for (ai = 0; ai < 7; ai = ai +1) begin
            for (bi = 0; bi < 7; bi = bi + 1) begin
                sum = ai + bi;
                // err = int'(s - sum);
                errb = s - sum;
                $display("error is %e", err);            
            end
        end
        
        #1000 $finish();
    end
endprogram

module add_tb;

parameter W = 4;

initial $dumpfile("add.fst");
initial $dumpvars(0, add_tb);

logic arst_n;
logic clk;

    always begin
        #0.5 clk = 1;
        #0.5 clk = 0;
    end

logic [W-1:0] a, b;
logic [W:0] s;

// the following should also work?
// rca #(W=4) urca(.a, .b, .s);
rca urca(.a, .b, .s);

test_add test(
    .clk,
    .arst_n,
    .a,
    .b,
    .s
);

endmodule

wsnyder, 2023-10-01:
First issue is Verilator shouldn't have crashed without an error. This is fixed it, then prints:

`%Error: Internal Error: t/t_0.v:9:14: ../V3Number.cpp:917: Real operation on wrong sized/non-real number
`

Second issue was your code is using an int for a real display of "%e". I suspect this is a mistake in your code and you intended "%d", but anyhow this is fixed.

```
int err = 0;
...
$display("error is %e", err);  // %e is for a real
```

closed 2023-10-01 by wsnyder

muzafferkal, 2023-10-02:
thanks for quite turn around. Is there any chance this error can be reported over the verilog line numbers as opposed to generated cpp?

wsnyder, 2023-10-02:
With the fix there is no error it just works, but anyhow the error does show the verilog line "t/t_0.v:9"  for testing I was using a different filename t_0.v