verilator/verilator #4529: recursive module

jalcim, 2023-10-02:
Like this old issue #659

for this code

```
`ifndef __ADDX__
 `define __ADDX__

 `include "src/alu/arithm/add.v"

module addX (a, b, cin, out, cout);
   parameter WIRE = 8;

   input [WIRE-1:0] a, b;
   input            cin;

   output [WIRE-1:0] out;
   output            cout;

   wire              ret;

   if (WIRE > 1)
     begin
        add add0 (a[0], b[0], cin, out[0], ret);
        addX #(.WIRE(WIRE-1)) recall(a[WIRE-1:1],
                                     b[WIRE-1:1],
                                     ret,
                                     out[WIRE-1:1],
                                     cout);
     end
   else
     add add0 (a[0], b[0], cin, out[0], cout);

endmodule

`endif
```

you can see here, verilator define WIRE=3, and instanciate only 1 add cell
[image]

jalcim, 2023-10-02:
for which one wanna test

here the icarus code of test:

```
`include "src/alu/arithm/addX.v"

module test_addX;
   parameter WIRE = 8;
   reg [WIRE-1:0] a, b;
   reg		  cin;
   wire [WIRE-1:0] out;
   wire		   cout;

   addX #(.WIRE(WIRE)) addX_inst(a, b, cin, out, cout);

   initial
     begin
	$dumpfile("signal_addX.vcd");
	$dumpvars;
	$display("\t\ttime,\ta,\tb, \tcin, \tout,\tcout");
	$monitor("%d \t%d \t%d \t%b \t%d \t%b", $time, a, b, cin, out, cout);

	a <= 0;
	b <= 0;
	cin <= 0;
	#100;
	a <= 24;
	b <= 0;
	cin <= 0;
	#100;
	a <= 0;
	b <= 54;
	cin <= 0;
	#100;
	a <= 23;
	b <= 54;
	cin <= 0;
	#100;
	a <= 23;
	b <= 54;
	cin <= 1;
	#100;
     end
endmodule
```

jalcim, 2023-10-02:
addX module after yosys synthesis

```
/* Generated by Yosys 0.33 (git sha1 2584903a0, gcc 12.2.0-14 -fPIC -Os) */

(* dynports =  1  *)
(* hdlname = "\\addX" *)
(* src = "src/alu/arithm/addX.v:6.1-33.10" *)
module \$paramod\addX\WIRE=s32'00000000000000000000000000000001 (a, b, cin, out, cout);
  (* src = "src/alu/arithm/addX.v:9.21-9.22" *)
  input a;
  wire a;
  (* src = "src/alu/arithm/addX.v:9.24-9.25" *)
  input b;
  wire b;
  (* src = "src/alu/arithm/addX.v:10.15-10.18" *)
  input cin;
  wire cin;
  (* src = "src/alu/arithm/addX.v:13.17-13.21" *)
  output cout;
  wire cout;
  (* src = "src/alu/arithm/addX.v:12.22-12.25" *)
  output out;
  wire out;
  (* module_not_derived = 32'd1 *)
  (* src = "src/alu/arithm/addX.v:27.10-27.46" *)
  add \genblk1.add0  (
    .a(a),
    .b(b),
    .cin(cin),
    .cout(cout),
    .out(out)
  );
endmodule

(* dynports =  1  *)
(* hdlname = "\\addX" *)
(* src = "src/alu/arithm/addX.v:6.1-33.10" *)
module \$paramod\addX\WIRE=s32'00000000000000000000000000000010 (a, b, cin, out, cout);
  (* src = "src/alu/arithm/addX.v:9.21-9.22" *)
  input [1:0] a;
  wire [1:0] a;
  (* src = "src/alu/arithm/addX.v:9.24-9.25" *)
  input [1:0] b;
  wire [1:0] b;
  (* src = "src/alu/arithm/addX.v:10.15-10.18" *)
  input cin;
  wire cin;
  (* src = "src/alu/arithm/addX.v:13.17-13.21" *)
  output cout;
  wire cout;
  (* src = "src/alu/arithm/addX.v:12.22-12.25" *)
  output [1:0] out;
  wire [1:0] out;
  (* src = "src/alu/arithm/addX.v:15.15-15.18" *)
  wire ret;
  (* module_not_derived = 32'd1 *)
  (* src = "src/alu/arithm/addX.v:19.6-19.41" *)
  add \genblk1.add0  (
    .a(a[0]),
    .b(b[0]),
    .cin(cin),
    .cout(ret),
    .out(out[0])
  );
  (* src = "src/alu/arithm/addX.v:20.24-24.15" *)
  \$paramod\addX\WIRE=s32'00000000000000000000000000000001  \genblk1.recall  (
    .a(a[1]),
    .b(b[1]),
    .cin(ret),
[... 249 lines cut ...]
  wire cout;
  (* src = "src/alu/arithm/addX.v:12.22-12.25" *)
  output [7:0] out;
  wire [7:0] out;
  (* src = "src/alu/arithm/addX.v:15.15-15.18" *)
  wire ret;
  (* module_not_derived = 32'd1 *)
  (* src = "src/alu/arithm/addX.v:19.6-19.41" *)
  add \genblk1.add0  (
    .a(a[0]),
    .b(b[0]),
    .cin(cin),
    .cout(ret),
    .out(out[0])
  );
  (* src = "src/alu/arithm/addX.v:20.24-24.15" *)
  \$paramod\addX\WIRE=s32'00000000000000000000000000000111  \genblk1.recall  (
    .a(a[7:1]),
    .b(b[7:1]),
    .cin(ret),
    .cout(cout),
    .out(out[7:1])
  );
endmodule
```

jalcim, 2023-10-02:
gtkwave with verilator and addX (synthesized with yosys)
[image]

jalcim, 2023-10-02:
for curious you can find the CPP code here
https://gitlab.com/jalcim/logilib/-/tree/test-alu/cosim/alu/arithm?ref_type=heads

wsnyder, 2023-10-02:
Can you please post complete code (there's a missing include)?  Also, ideally, can you make the verilog self-checking that is it looks at parameters or otherwise so it passes on Icarus and fails with Verilator, using "--binary" with Verilator.  Thanks.

jalcim, 2023-10-02:
add.v
```
`ifndef __ADD__
 `define __ADD__

module add (a, b, cin, out, cout);
   input a, b, cin;
   output out, cout;
   /* verilator lint_off UNOPTFLAT */
   wire [2:0] line;

   nand nand1 (line[0], a, cin);
   xor xor1 (line[1], a, cin);
   nand nand2 (line[2], line[1], b);
   nand nand3 (cout, line[0], line[2]);
   assign out = (a ^ b ^ cin);
endmodule // full_add

`endif
```

wsnyder, 2023-10-03:
Are you sure you're using the most recent version of Verilator?  This test case (put into test_regress/t_EXAMPLE.v in the kit) works fine for me.  If you are, please modify code shown below to show the problem.

```
module t(/*AUTOARG*/
   // Inputs
   clk
   );
   input clk;

   integer cyc = 0;
   reg [63:0] crc;
   reg [63:0] sum;

   localparam WIDTH = 8;

   wire [7:0] a = crc[7:0];
   wire [7:0] b = crc[15:8];
   wire cin = crc[16];

   /*AUTOWIRE*/
   // Beginning of automatic wires (for undeclared instantiated-module outputs)
   wire                 cout;                   // From test of addX.v
   wire [WIDTH-1:0]     out;                    // From test of addX.v
   // End of automatics

   addX test(/*AUTOINST*/
             // Outputs
             .out                       (out[WIDTH-1:0]),
             .cout                      (cout),
             // Inputs
             .a                         (a[WIDTH-1:0]),
             .b                         (b[WIDTH-1:0]),
             .cin                       (cin));

   wire [63:0] result = {55'h0, cout, out};

   // Test loop
   always @ (posedge clk) begin
`ifdef TEST_VERBOSE
      $write("[%0t] cyc==%0d crc=%x result=%x\n", $time, cyc, crc, result);
`endif
      cyc <= cyc + 1;
      crc <= {crc[62:0], crc[63] ^ crc[2] ^ crc[0]};
      sum <= result ^ {sum[62:0], sum[63] ^ sum[2] ^ sum[0]};
      if (cyc == 0) begin
         // Setup
         crc <= 64'h5aef0c8d_d70a4497;
         sum <= '0;
      end
      else if (cyc < 10) begin
         sum <= '0;
      end
      else if (cyc == 99) begin
         $write("[%0t] cyc==%0d crc=%x sum=%x\n", $time, cyc, crc, sum);
         if (crc !== 64'hc77bb9b3784ea091) $stop;
         // What checksum will we end up with (above print should match)
`define EXPECTED_SUM 64'h95fcff51cc8bbe75
         if (sum !== `EXPECTED_SUM) $stop;
         $write("*-* All Finished *-*\n");
         $finish;
      end
   end

endmodule
module add (a, b, cin, out, cout);
   input a, b, cin;
   output out, cout;
   // verilator lint_off UNOPTFLAT
   wire [2:0] line;

   nand nand1 (line[0], a, cin);
   xor xor1 (line[1], a, cin);
   nand nand2 (line[2], line[1], b);
   nand nand3 (cout, line[0], line[2]);
   assign out = (a ^ b ^ cin);
endmodule

module addX (a, b, cin, out, cout);
   parameter WIDTH = 8;

   input [WIDTH-1:0] a, b;
   input            cin;

   output [WIDTH-1:0] out;
   output            cout;

   wire              ret;

   if (WIDTH > 1) begin
      add add0 (a[0], b[0], cin, out[0], ret);
      addX #(.WIDTH(WIDTH-1)) recall(a[WIDTH-1:1],
                                     b[WIDTH-1:1],
                                     ret,
                                     out[WIDTH-1:1],
                                     cout);
   end
   else begin
      add add0 (a[0], b[0], cin, out[0], cout);
   end

endmodule
```

jalcim, 2023-10-03:
```
TESTS DONE, FAILED: Passed 3346  Failed 2  Skipped 241  Time 216:20
```

where can i find log file ?

wsnyder, 2023-10-03:
test_regress/obj_dist/*.log has the logs, though it also goes to the screen.  You can run a single test by running it directly test_regress/t/t_EXAMPLE.pl; see docs/internals.rst for more details.

wsnyder, 2023-10-03:
So... there's no issue and this can be closed?

jalcim, 2023-10-05:
I just provide you my output, i didn't understand whats work inside these test.
What i see its we can't use recursivity in verilog code for using with verilator (first message),

wsnyder, 2023-10-05:
Recursive modules are intended to work.

When I run the example you provided it worked fine.

So, either you aren't using the latest Verilator and it has been fixed, or modify the example I posted to show it failing on Verilator.

There's no way the C++ wrapper could affect this behavior, so it's not relevant.

jalcim, 2023-10-08:
sorry i have take time for read the internal test documentation, then i take time for respond.

my verilator version
```
verilator --version
Verilator 5.014 2023-08-06 rev v5.014-35-ge6b0bdd4d
```

Your exemple is on verilog not on C++,
in your exemple all the simulation is on verilog, and you didn't use the wrapper.

I dont say verilator can't build it, i say verilator synthesize bad things like that.

https://user-images.githubusercontent.com/6641392/271872678-9fe071a7-32f6-47c1-8e26-0e7ba64dd9dc.png

Then its normal if your t_test dont show any problem.

wsnyder, 2023-10-08:
Maybe you are calling in C++ e.g.

 top->trace(tfp.get(), 1);

rather than ", 99" so only getting one level in the trace file?

wsnyder, 2023-10-18:
I'm assuming you figured this out, otherwise please address the comments to make a reproducing case and reopen. Thanks.

closed 2023-10-18 by wsnyder