verilator/verilator #4531: Issue with streaming operator on packed structs since v5.016

wltr, 2023-10-02:
Hi, while upgrading to v5.016, I get the internal error below when trying to use the streaming operator on a packed struct. This worked in prior versions.
```
%Error: Internal Error: verilator_test_tb.sv:11:43: ../V3AstNodeDType.h:1133: call isCompound on subdata type, not reference
   11 |   function automatic logic [63:0] reverse(data_t d);
      |                                           ^~~~~~
```
Example to reproduce:
```
`default_nettype none
module verilator_test_tb (
  input wire [63:0] data_i,
  output logic [63:0] data_o
);

  typedef struct packed {
    logic [63:0] dummy;
  } data_t;

  function automatic logic [63:0] reverse(data_t d);
    return {<<{d}};
  endfunction

  assign data_o = reverse(data_i);

endmodule
`default_nettype wire
```
I found a work-around by adding a cast:
```
  function automatic logic [63:0] reverse(data_t d);
    return {<<{$bits(d)'(d)}};
  endfunction
```
Verilator version:
```
% verilator --version
Verilator 5.016 2023-09-16 rev UNKNOWN.REV
```

wsnyder, 2023-10-03:
@RRozak perhaps you could fix or offer advice if @wltr can attempt a fix?

Here's a self-testing test.

```
// DESCRIPTION: Verilator: Verilog Test module
//
// This file ONLY is placed into the Public Domain, for any use,
// without warranty, 2023 by Wilson Snyder.
// SPDX-License-Identifier: CC0-1.0

module t(/*AUTOARG*/
   // Inputs
   clk
   );
   input clk;

   integer cyc = 0;
   reg [63:0] crc;
   reg [63:0] sum;

   // Take CRC data and apply to testblock inputs
   wire [63:0] in = crc[63:0];

   /*AUTOWIRE*/
   // Beginning of automatic wires (for undeclared instantiated-module outputs)
   logic [63:0]         out;                    // From test of Test.v
   // End of automatics

   Test test(/*AUTOINST*/
             // Outputs
             .out                       (out[63:0]),
             // Inputs
             .in                        (in[63:0]));

   // Test loop
   always @ (posedge clk) begin
`ifdef TEST_VERBOSE
      $write("[%0t] cyc==%0d crc=%x out=%x\n", $time, cyc, crc, out);
`endif
      cyc <= cyc + 1;
      crc <= {crc[62:0], crc[63] ^ crc[2] ^ crc[0]};
      sum <= out ^ {sum[62:0], sum[63] ^ sum[2] ^ sum[0]};
      if (cyc == 0) begin
         // Setup
         crc <= 64'h5aef0c8d_d70a4497;
         sum <= '0;
      end
      else if (cyc < 10) begin
         sum <= '0;
      end
      else if (cyc == 99) begin
         $write("[%0t] cyc==%0d crc=%x sum=%x\n", $time, cyc, crc, sum);
         if (crc !== 64'hc77bb9b3784ea091) $stop;
         // What checksum will we end up with (above print should match)
`define EXPECTED_SUM 64'h29271cf844d6f90c
         if (sum !== `EXPECTED_SUM) $stop;
         $write("*-* All Finished *-*\n");
         $finish;
      end
   end

endmodule

module Test (
             input wire [63:0]   in,
             output logic [63:0] out);

   typedef struct  packed {
      logic [63:0] dummy;
   } data_t;

   function automatic logic [63:0] reverse(data_t d);
      return {<<{d}};
   endfunction

   assign out = reverse(in);

endmodule
```

RRozak, 2023-10-03:
The problem is not really with stream operations, but with `isCompound` method. It doesn't accept reference types.

wsnyder, 2023-10-03:
@RRozak  thanks for the fast fix!

closed 2023-10-03 by wsnyder