verilator/verilator #4534: Extent to which UVM is supported

arshadrohan5, 2023-10-03:
Is support for UVM enabled currently? I faced a PKGNODECL error while Verilating a simple UVM testbench. Screenshots are attached.
[image]

wsnyder, 2023-10-03:
Not yet, see #1538

closed 2023-10-03 by wsnyder

arshadrohan5, 2023-10-03:
How can I get started on adding support for UVM? What are the technical prerequisites that I should be aware of to initiate my contributions to Verilator effectively?

wsnyder, 2023-10-03:
We'd love help!  If you look through the sub issues described in #1538, and see which you'd like to contribute to, then post in that sub issue your intent, then we'll give some pointers.   (Many also have good starting points, just need more work to complete.)