verilator/verilator #4535: Launching multiple threads on multiple cores on the DUT

jordankrim, 2023-10-03:
I do not know how to append to an already closed issue and since I didn't receive the normal speedy reply, I thought it might have gotten lost, so I am starting a new issue again with a title that may clarify more of what my problem is.

I want to clarify the problem because I think while I was able to get working your answer using numactl, it isn't exactly where the problem is. I think the numactl addresses the problem of Verilator running on the Host number of CPUs. What I mean by this is I am running on an Oracle Box VM which I have set up to have 8 Processors (CPUs) (i.e. capable of 8 parallel threads ?). Verilator is running on these Host CPUs so that is where the numactl is supposed to help.

My question/problem is not on the host. I am saying that we want to launch multiple threads on the target design (i.e. DUT/SOC/FPGA etc). This is where I am trying to determine whether Verilator is working correctly. As I stated before I am not sure why my simulation is failing. I am trying to determine whether it is Verilator or some problem in my testbench (harness) that is around my DUT.

wsnyder, 2023-10-03:
Are you saying some simulation is giving wrong results?  If so, I'm confused why you are titled this issue involving multiple threads and mentioning that here.

jordankrim, 2023-10-04:
Yes.  I have a simulation that passes on an EDA simulator but fails on Verilator.  I am just trying to determine why.  It could have something to do with the fact that the simulation attempts to launch on multiple cores (there are 4) on the DUT.  But, it may be a problem with the DUT (testbench) harness vs.  Verilator.  Just trying to get some insight as to whether Verilator should be able to accomplish this as is (natively) or needing some special incantation or not at all.

wsnyder, 2023-10-04:
Well try it not using multiple threads.  Run verilator and another simulator using waves, see where they differ, trace down to exact signal that differs, then make a small code example that shows the difference.

wsnyder, 2023-10-21:
I'm assuming there's nothing left to answer here, but feel free to follow up.

closed 2023-10-21 by wsnyder