verilator/verilator #4538: Address Sanitizer use-after-free when multithread verilating using CMake

nuudlman, 2023-10-04:
Verilator 5.016, Arch Linux package.
gcc 13.2.1
mold 2.2.0
cmake 3.27.6

ASan Output:

```
=================================================================
==60214==ERROR: AddressSanitizer: heap-use-after-free on address 0x60e0003e42f8 at pc 0x5609c7636c05 bp 0x7f3ad39ef320 sp 0x7f3ad39ef310
READ of size 1 at 0x60e0003e42f8 thread T5
    #0 0x5609c7636c04 in VerilatedTrace<VerilatedVcd, VerilatedVcdBuffer>::parallelWorkerTask(void*, bool) /usr/share/verilator/include/verilated_trace_imp.h:472
    #1 0x5609c76ab457 in VlWorkerThread::workerLoop() /usr/share/verilator/include/verilated_threads.cpp:88
    #2 0x5609c76ab457 in VlWorkerThread::startWorker(VlWorkerThread*, VerilatedContext*) /usr/share/verilator/include/verilated_threads.cpp:96
    #3 0x7f3ad9ae1942 in execute_native_thread_routine /usr/src/debug/gcc/gcc/libstdc++-v3/src/c++11/thread.cc:104
    #4 0x7f3ad98aa9ea  (/usr/lib/libc.so.6+0x8c9ea) (BuildId: 8bfe03f6bf9b6a6e2591babd0bbc266837d8f658)
    #5 0x7f3ad992e7cb  (/usr/lib/libc.so.6+0x1107cb) (BuildId: 8bfe03f6bf9b6a6e2591babd0bbc266837d8f658)

0x60e0003e42f8 is located 152 bytes inside of 160-byte region [0x60e0003e4260,0x60e0003e4300)
freed by thread T0 here:
    #0 0x7f3ad9ee311a in operator delete(void*, unsigned long) /usr/src/debug/gcc/gcc/libsanitizer/asan/asan_new_delete.cpp:164
    #1 0x5609c7660f6a in std::__new_allocator<std::_List_node<VerilatedTrace<VerilatedVcd, VerilatedVcdBuffer>::ParallelWorkerData> >::deallocate(std::_List_node<VerilatedTrace<VerilatedVcd, VerilatedVcdBuffer>::ParallelWorkerData>*, unsigned long) /usr/include/c++/13.2.1/bits/new_allocator.h:168
    #2 0x5609c7660f6a in std::allocator<std::_List_node<VerilatedTrace<VerilatedVcd, VerilatedVcdBuffer>::ParallelWorkerData> >::deallocate(std::_List_node<VerilatedTrace<VerilatedVcd, VerilatedVcdBuffer>::ParallelWorkerData>*, unsigned long) /usr/include/c++/13.2.1/bits/allocator.h:210
    #3 0x5609c7660f6a in std::allocator_traits<std::allocator<std::_List_node<VerilatedTrace<VerilatedVcd, VerilatedVcdBuffer>::ParallelWorkerData> > >::deallocate(std::allocator<std::_List_node<VerilatedTrace<VerilatedVcd, VerilatedVcdBuffer>::ParallelWorkerData> >&, std::_List_node<VerilatedTrace<VerilatedVcd, VerilatedVcdBuffer>::ParallelWorkerData>*, unsigned long) /usr/include/c++/13.2.1/bits/alloc_traits.h:516
    #4 0x5609c7660f6a in std::__cxx11::_List_base<VerilatedTrace<VerilatedVcd, VerilatedVcdBuffer>::ParallelWorkerData, std::allocator<VerilatedTrace<VerilatedVcd, VerilatedVcdBuffer>::ParallelWorkerData> >::_M_put_node(std::_List_node<VerilatedTrace<VerilatedVcd, VerilatedVcdBuffer>::ParallelWorkerData>*) /usr/include/c++/13.2.1/bits/stl_list.h:522
    #5 0x5609c7660f6a in std::__cxx11::_List_base<VerilatedTrace<VerilatedVcd, VerilatedVcdBuffer>::ParallelWorkerData, std::allocator<VerilatedTrace<VerilatedVcd, VerilatedVcdBuffer>::ParallelWorkerData> >::_M_clear() /usr/include/c++/13.2.1/bits/list.tcc:81

previously allocated by thread T0 here:
    #0 0x7f3ad9ee2002 in operator new(unsigned long) /usr/src/debug/gcc/gcc/libsanitizer/asan/asan_new_delete.cpp:95
    #1 0x5609c7637cd6 in std::__new_allocator<std::_List_node<VerilatedTrace<VerilatedVcd, VerilatedVcdBuffer>::ParallelWorkerData> >::allocate(unsigned long, void const*) /usr/include/c++/13.2.1/bits/new_allocator.h:147
    #2 0x5609c7637cd6 in std::allocator<std::_List_node<VerilatedTrace<VerilatedVcd, VerilatedVcdBuffer>::ParallelWorkerData> >::allocate(unsigned long) /usr/include/c++/13.2.1/bits/allocator.h:198
    #3 0x5609c7637cd6 in std::allocator_traits<std::allocator<std::_List_node<VerilatedTrace<VerilatedVcd, VerilatedVcdBuffer>::ParallelWorkerData> > >::allocate(std::allocator<std::_List_node<VerilatedTrace<VerilatedVcd, VerilatedVcdBuffer>::ParallelWorkerData> >&, unsigned long) /usr/include/c++/13.2.1/bits/alloc_traits.h:482
    #4 0x5609c7637cd6 in std::__cxx11::_List_base<VerilatedTrace<VerilatedVcd, VerilatedVcdBuffer>::ParallelWorkerData, std::allocator<VerilatedTrace<VerilatedVcd, VerilatedVcdBuffer>::ParallelWorkerData> >::_M_get_node() /usr/include/c++/13.2.1/bits/stl_list.h:518
    #5 0x5609c7637cd6 in std::_List_node<VerilatedTrace<VerilatedVcd, VerilatedVcdBuffer>::ParallelWorkerData>* std::__cxx11::list<VerilatedTrace<VerilatedVcd, VerilatedVcdBuffer>::ParallelWorkerData, std::allocator<VerilatedTrace<VerilatedVcd, VerilatedVcdBuffer>::ParallelWorkerData> >::_M_create_node<void (* const&)(void*, VerilatedTraceBuffer<VerilatedVcdBuffer>*), void* const&, VerilatedTraceBuffer<VerilatedVcdBuffer>* const&>(void (* const&)(void*, VerilatedTraceBuffer<VerilatedVcdBuffer>*), void* const&, VerilatedTraceBuffer<VerilatedVcdBuffer>* const&) /usr/include/c++/13.2.1/bits/stl_list.h:710
    #6 0x5609c7637cd6 in void std::__cxx11::list<VerilatedTrace<VerilatedVcd, VerilatedVcdBuffer>::ParallelWorkerData, std::allocator<VerilatedTrace<VerilatedVcd, VerilatedVcdBuffer>::ParallelWorkerData> >::_M_insert<void (* const&)(void*, VerilatedTraceBuffer<VerilatedVcdBuffer>*), void* const&, VerilatedTraceBuffer<VerilatedVcdBuffer>* const&>(std::_List_iterator<VerilatedTrace<VerilatedVcd, VerilatedVcdBuffer>::ParallelWorkerData>, void (* const&)(void*, VerilatedTraceBuffer<VerilatedVcdBuffer>*), void* const&, VerilatedTraceBuffer<VerilatedVcdBuffer>* const&) /usr/include/c++/13.2.1/bits/stl_list.h:2005
    #7 0x5609c7637cd6 in VerilatedTrace<VerilatedVcd, VerilatedVcdBuffer>::ParallelWorkerData& std::__cxx11::list<VerilatedTrace<VerilatedVcd, VerilatedVcdBuffer>::ParallelWorkerData, std::allocator<VerilatedTrace<VerilatedVcd, VerilatedVcdBuffer>::ParallelWorkerData> >::emplace_back<void (* const&)(void*, VerilatedTraceBuffer<VerilatedVcdBuffer>*), void* const&, VerilatedTraceBuffer<VerilatedVcdBuffer>* const&>(void (* const&)(void*, VerilatedTraceBuffer<VerilatedVcdBuffer>*), void* const&, VerilatedTraceBuffer<VerilatedVcdBuffer>* const&) /usr/include/c++/13.2.1/bits/stl_list.h:1321
    #8 0x5609c7637cd6 in VerilatedTrace<VerilatedVcd, VerilatedVcdBuffer>::runCallbacks(std::vector<VerilatedTrace<VerilatedVcd, VerilatedVcdBuffer>::CallbackRecord, std::allocator<VerilatedTrace<VerilatedVcd, VerilatedVcdBuffer>::CallbackRecord> > const&) /usr/share/verilator/include/verilated_trace_imp.h:506
    #9 0x5609c763d26f in VerilatedTrace<VerilatedVcd, VerilatedVcdBuffer>::dump(unsigned long) /usr/share/verilator/include/verilated_trace_imp.h:609

Thread T5 created by T0 here:
    #0 0x7f3ad9e4a497 in __interceptor_pthread_create /usr/src/debug/gcc/gcc/libsanitizer/asan/asan_interceptors.cpp:208
    #1 0x7f3ad9ae1a29 in __gthread_create /usr/src/debug/gcc/gcc-build/x86_64-pc-linux-gnu/libstdc++-v3/include/x86_64-pc-linux-gnu/bits/gthr-default.h:663
    #2 0x7f3ad9ae1a29 in std::thread::_M_start_thread(std::unique_ptr<std::thread::_State, std::default_delete<std::thread::_State> >, void (*)()) /usr/src/debug/gcc/gcc/libstdc++-v3/src/c++11/thread.cc:172
    #3 0x5609c75e326f in VerilatedContext::threadPoolp() /usr/share/verilator/include/verilated.cpp:2607

SUMMARY: AddressSanitizer: heap-use-after-free /usr/share/verilator/include/verilated_trace_imp.h:472 in VerilatedTrace<VerilatedVcd, VerilatedVcdBuffer>::parallelWorkerTask(void*, bool)
Shadow bytes around the buggy address:
  0x60e0003e4000: fd fd fd fd fd fd fd fd fd fd fd fd fa fa fa fa
  0x60e0003e4080: fa fa fa fa fd fd fd fd fd fd fd fd fd fd fd fd
  0x60e0003e4100: fd fd fd fd fd fd fd fd fa fa fa fa fa fa fa fa
  0x60e0003e4180: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
  0x60e0003e4200: fd fd fd fd fa fa fa fa fa fa fa fa fd fd fd fd
=>0x60e0003e4280: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd[fd]
  0x60e0003e4300: fa fa fa fa fa fa fa fa 00 00 00 00 00 00 00 00
  0x60e0003e4380: 00 00 00 00 00 00 00 00 00 00 00 00 fa fa fa fa
  0x60e0003e4400: fa fa fa fa 00 00 00 00 00 00 00 00 00 00 00 00
  0x60e0003e4480: 00 00 00 00 00 00 00 00 fa fa fa fa fa fa fa fa
  0x60e0003e4500: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==60214==ABORTING
```
See https://github.com/nuudlman/verilator-use-after-free-repro for code.

wsnyder, 2023-10-05:
Perhaps you'd be willing to look into what's causing this and propose a pull to fix it?

nuudlman, 2023-10-05:
I don't really have the time unfortunately.

wsnyder, 2024-01-18:
I ended up having to make several changes to get it to compile, then I didn't see the error.  If you'd like please change the example to use latest verilator and provide a shell script to call cmake and show the issue, thanks.

closed 2024-01-24 by wsnyder