verilator/verilator #4542: Scheduling of impure recursive functions

kbieganski, 2023-10-05:
Verilator allows recursive functions since commit aa60847.

However, the scheduler is still unable to peek into function calls. This means impure recursive functions are not being scheduled properly.

Consider this example:

```systemverilog
module t;
    logic bar, qux;

    task foo(logic recurse);
        if (recurse) foo(0);
        else bar = ~bar;
    endtask

    assign qux = $c("1") & bar;

    initial begin
        for (int i = 0; i < 10; i++) #1 foo(1);
        #1 $finish;
    end

    always @(edge bar) $display("bar=%b", bar);
    always @(edge qux) $display("qux=%b", qux);
endmodule
```

The output is:

```
bar=1
bar=0
bar=1
bar=0
bar=1
bar=0
bar=1
bar=0
bar=1
bar=0
- example.sv:13: Verilog $finish
```

So `qux` is never written to. If you remove this part:

```systemverilog
        if (recurse) foo(0);
        else /* keep the assignment */
```

Then the result is correct.

I see two options:

* Make the scheduler look into function calls to account for side effects. Doesn't seem too hard, but non-obvious issues may crop up.
* Disallow recursive functions in synthesizable code, but keep them in verification/class code, as static scheduling doesn't affect it anyway. That way we can still have the recursion in UVM.

wsnyder, 2023-10-05:
I believe this is seen because the function inliner gets disabled with recursion. That is, recursion isn't really related to this issue.

>Disallow recursive functions in synthesizable code, but keep them in verification/class code, as static scheduling doesn't affect it anyway. That way we can still have the recursion in UVM.

1. I believe recursion isn't related.
2. There's no way to identify synthesizable code.
3. Some models like PLL and memory models while not synthesizable will need proper scheduling.

So I don't see this one as an option.

kbieganski, 2023-10-05:
1. Like you said, the inlining is disabled with recursion, so it is related. From the perspective of the user impure recursive functions is the thing that doesn't work correctly. I think there should be a warning or error about that, and the user doesn't care about inlining that they have no control over.
2. In this case I meant something like "called from a class" as opposed to "called from a module."
3. I'm all for having more features. My proposal was for the likely case that this won't be worked on for a while.

wsnyder, 2023-10-05:
There's many causes of not-inlining.  Anyhow we agree.

I think the action here as you indicate is to add a warning for now.  I see your point - this can be when it's called from a AstNodeProcedure, and if we check before/at scheduling only non-inlined functions will exist at that point.

kbieganski, 2023-10-05:
Do the other causes of not-inlining break scheduling? If so, then I suppose the message should be more general. I admit I haven't looked into those, it's just this recent change that piqued my interest.

wsnyder, 2023-10-05:
Anything that keeps the function may break it.  Recursive, too large, member function, no inline verilator pragma, DPI export, probably others.

ArturBieniek4, 2026-09-02:
Fixed by #7924.

closed 2026-09-02 by ArturBieniek4