verilator/verilator #4544: Optimize empty expression statements

wsnyder, 2023-10-07:
Debug of #4543 showed we have expression statements like this

```
    1:2:3:1:1:2:1: EXPRSTMT 0x555556e67d90 <e2261#> {e14cg} @dt=0x555556eaaea0@(G/d)
    1:2:3:1:1:2:1:1: COMMENT 0x555556e85340 <e1240> {e14cg}  Function: logWrapper
    1:2:3:1:1:2:1:2: CONST 0x555556f43300 <e1437> {e10ao} @dt=0x555556eaaea0@(G/d)  2.3025850929940459
```

The EXPRSTMT was inserted by inlining a function, but later V3Const should be taught to compress this down to just a constant (by removing outer exprstmt since it has only comment).

I tried a fix for this, it fixes this case but t_interface_array.pl fails because the exprstatement was not pure, and is pure after removal, causing the new pure casting to correctly fail in V3Broken, since the caching really is no longer correct.  So blocked on fixing that caching somehow.  (Ideas @RRozak?)

RRozak, 2023-10-09:
I see two ways.
One is to clear the cache after the conversion, like here:
https://github.com/verilator/verilator/blob/622f0c047c736278d7f06d0d885def72aaca703a/src/V3Task.cpp#L1472
Another one is to add computing the purity of AstExprStmt.

wsnyder, 2023-10-09:
The issue isn't just that the ExprStmt purity changes, but that this propagates upwards to parent Biops etc.  I ended up both computing ExprStmt purity, and also clearing the cache entirely as needed.

closed 2023-10-09 by wsnyder