verilator/verilator #4546: readmemh transform to VL_READMEM_N ,set depth and array_lsb

shibin2018, 2023-10-08:
when I use $readmemh(file, mem)，in the file, here has address of 0x20000000, but the transformed VL_READMEM_N depth is1048576（0x100000）and array_lsb is 0， which will cause error about out of bound of array, how can i modify the depth and array_lsb

wsnyder, 2023-10-08:
Please post the declared "mem" array. Is it bigger than 0x100000?

shibin2018, 2023-10-09:
> Please post the declared "mem" array. Is it bigger than 0x100000?
yes， its index is from 0 to 2^32-1

wsnyder, 2023-10-09:
Indexes in Verilog are integral, so an index of 2^32-1 is actually negative, so Verilator's warning is correct that there is no index 0x100000 (there's a bug pending to have lint warn when this is done).  Also, allocating an array of 4GB is probably a bad idea anyways for memory usage reasons.  Try using an associative array.

Closing as solved but feel free to follow up.

closed 2023-10-09 by wsnyder