verilator/verilator #4547: Simulation performance differs with different Verilog styles

poemonsense, 2023-10-09:
I'm simulating the single-core rocket-chip (https://github.com/OpenXiangShan/rocket-chip/tree/dev-difftest). I observe quite different simulation performance on this design with the latest version of Verilator (Verilator 5.017 devel rev v5.014-161-g622f0c047)

The design is originally in Chisel, but is quite simple. We compare the simulation performance on two Chisel versions. V3 and V6 have identical Chisel (hardware) source code. The only difference is the Chisel compiler version.

1) V6: Chisel 6.0.0-M3 with CIRCT firtool 1.56.1 generates 38K lines of Verilog as attached. [build_v6.tar.gz](https://github.com/verilator/verilator/files/12843948/build_v6.tar.gz)

2) V3: Chisel 3.6.0 generates 59K lines of Verilog as attached. [build_v3.tar.gz](https://github.com/verilator/verilator/files/12843944/build_v3.tar.gz)

We can see that V6 has fewer lines of code with multiple operations in one line. By contrast, V3 typically has only one operator (&, |, +, ...) in each line. Since they are created from the same Chisel code and looks quite similar in the Verilog, I wonder why V6 and V3 simulates with vastly different simulation speed with Verilator.

1) V6: 18.5 seconds for 5 million cycles. 

2) V3: 14.7 seconds for 5 million cycles.

Do you have any idea or recommendations on how we could investigate into the performance deviation between the two versions? Should it be related to the coding style of Verilog? I have not found large structures in the code that differs between the two versions.

At larger designs with multithreading, the simulation performance gap further increases. We observe 50% simulation performance degradation on XiangShan when switching from Chisel 3 to Chisel 6. Some Chisel developers doubt the performance is due to Verilator instead of the Chisel compiler.

Any advice or comments are greatly appreciated. Thank you.

gezalore, 2023-10-09:
The peephole patterns of the DFG optimizer were heavily tuned for that design, and I suspect the older version. The new version probably has some different patterns that are not covered. If you run with --stats, it reports the number of times each DFG optimization was performed, so you might be able to notice something useful, but I expect it will be all over the place. 

Probably would be helpful to do another round of analysis to see if more patterns could be advantageous. There are no guarantees, but I suspect you might be able to find some things. If this is something you would like to attempt yourself I am happy to provide some pointers.

poemonsense, 2023-10-09:
I'll run the `--stats` to see the results. Do you have any ideas on the potential suspect styles?

Yesterday I try force the compiler to create only one binary operator at each line, which should be quite like V3. But the simulation speed does not seem to improve. If there's any suspect sources of performance degradation, then we can try modifying the compiler first, instead of looking into the details in Verilator stats, which, as you said, may report differences all over the place.

gezalore, 2023-10-09:
> I'll run the --stats to see the results. Do you have any ideas on the potential suspect styles?

Not really, the DFG peephole pass is just a collection of rules like "if expression is 'x + 0', drop the '+'". Obviously there are higher complexity ones, but they are quite specific.

> Yesterday I try force the compiler to create only one binary operator at each line, which should be quite like V3.

This optimizer slurps all combinational logic into a big graph, expressions per line should not matter at all.

Keep in mind it might be something completely different as well. I just suspect the DFG peephole as it was tuned for your design originally.

poemonsense, 2023-10-09:
Probably I should just try disabling specific optimizers and see the results. Will come back later with the results

poemonsense, 2023-10-09:
Previously we are using O3, where **V6 simulates 30% slower**.

Now I switch to O0, the performance differs this time. V3: 45 seconds for 5 million cycles. V6: 30 seconds for 5 million cycles. That is, **V6 simulates 30% faster**.

So, now I believe what the Chisel developers said. The performance issue should be of Verilator, instead of the Chisel compiler. 

Luckily we now what to do next. I should try the optimization passes one by one in Verilator and see which passes cause the performance deviation.

wsnyder, 2023-10-09:
That O0 is better with V6 likely means that the code out of Chisel has gotten faster without optimization.  This likely isn't evidence for or against anything in Verilator.  It does suggest though that Chisel did change how the code looks, which might supports @gezalore 's hypothesis

poemonsense, 2023-10-09:
Sorry, I might not express my opinion correctly.

It indicates that Verilator may be further optmized for the new code. I didn't mean to criticize or express any negative opinions on Verilator. Verilator does really good for us. Thank you and I'll try to locate the deviations.

wsnyder, 2023-10-09:
Makes sense.  BTW I have no objection to criticism, just making sure the evidence interpretation was understood ;)

poemonsense, 2023-10-13:
I see from the stats.txt that V6 version has more `Instruction count` than the V3 version at the `Final`/`Final_Fast` stage (193706 vs 151912), though fewer at the `Link` stage (190400 vs 314233). It seems these numbers correlate to the performance deviation between the two versions with `O0` and `O3`. They have similar counts at the `PreOrder` and `Scoped` stages.

Is there any doc describing these stages?

wsnyder, 2023-10-13:
docs/internals.rst describes some of the transformations.  I would suggest use -prof-c-funcs - see the "Profiling" section of the manual. Then using cprof you should be able to see what Verilog code blocks are now slower (or if it's across the board which tells you something else ;)

gezalore, 2023-10-13:
Note the Final_Fast stats are somewhat broken (not all entry points are traced correctly) so I would not draw conclusions based on just those numbers.

poemonsense, 2023-10-13:
After I enable `--prof-c-funcs`, the simulation speed of V6 becomes faster than V3 at `O3`, which is different from the case prof is disabled.

poemonsense, 2023-10-13:
However, I observe the fraction of `VLib` on V6 is larger on V3, increasing from 3.8% to 9.2%. Seems like Verilator is spending different fractions of times on VSimTop and VLib

poemonsense, 2023-10-13:
It looks profiling logic is slower than the hardware verilated logic (dut to the simplicity of the hardware, which runs at hundreds of KHz originally), and thus the `prof-c-funcs` greatly affects the simulation performance.

poemonsense, 2023-10-13:
I was only looking at the SimTop lines.

I just realized that I missed an important function. Is it probably due to the `VL_SHIFTRS_WWI`?

This is the V6:
```
  %   cumulative   self
 time   seconds   seconds      calls   design   type      filename and line number
 16.79      0.73     0.73  182002183   -        VLib      VL_SHIFTRS_WWI(int, int, int, unsigned int*, unsigned int const*, unsigned int)
 10.34      1.18     0.45    1000011   VSimTop  VCommon   VSimTop___024root___eval_nba(VSimTop___024root*)
```

This is the V3:
```
  %   cumulative   self
 time   seconds   seconds      calls   design   type      filename and line number
 15.62      0.72     0.72    1000011   VSimTop  VCommon   VSimTop___024root___eval_nba(VSimTop___024root*)
  6.08      1.00     0.28   72000864   -        VLib      VL_SHIFTRS_WWI(int, int, int, unsigned int*, unsigned int const*, unsigned int)
```

Look like V6 is spending much more time and calls in this SHIFTRS function

poemonsense, 2023-10-13:
I was able to locate (one of) the sources of this performance gap on the number of calls of `VL_SHIFTRS_WWI`. The generated Verilog is quite difficult to identify their original logic. I'm still trying to find why. They have been attached. [case_VL_SHIFTRS_WWI.tar.gz](https://github.com/verilator/verilator/files/12893811/case_VL_SHIFTRS_WWI.tar.gz) I will investigate more tomorrow. 

For the V3 version, it `took 0.739054 seconds to execute`. For the V6 version, it `took 2.167168 seconds to execute`. Note that V6 has one more input `io_detectTininess`, which is tied off with 1'h1. I tried tied it off in the module, but it does not affect the simulation time.

In V3 or V6:
```
verilator --cc --exe --main ./RoundAnyRawFNToRecFN_4.v
cp VRoundAnyRawFNToRecFN_4__main.cpp obj_dir
make -C ./obj_dir -f VRoundAnyRawFNToRecFN_4.mk -j16 -s
./obj_dir/VRoundAnyRawFNToRecFN_4
```

gezalore, 2023-10-13:
If you just run verilator on those 2 modules (separately), and add `--dumpi-dfg 6`, you can see some graphs of the combinational logic involved (e.g.: obj_v6/VRoundAnyRawFNToRecFN_4_033_RoundAnyRawFNToRecFN_4-acyclic-component-0-preinline-pass-5-removeVars.dot, which is after optimizing the graph). The way they do the shift does not seem too different, so I don't think the increased cost of the shift is explained by this, at least not obviously.

poemonsense, 2023-10-13:
Actually I was not able to see the logic differences manually. Given that I'm not familiar with the original Chisel design, these generated Verilog files are too hard to read for me. Probably with your tools it would be more obvious.

If their logics are similar, why did Verilator generate much more function calls for the V6 version?

I used `--prof-cfuncs` for V3 and V6. The V6 indeed calls the `VL_SHIFTRS_WWI` 3X than the V3 (510,000,031 vs 140,000,004).

V3: 
```
  %   cumulative   self
 time   seconds   seconds      calls   design                   type      filename and line number
 50.01      0.29     0.29  140000004   -                        VLib      VL_SHIFTRS_WWI(int, int, int, unsigned int*, unsigned int const*, unsigned int)
 15.52      0.38     0.09   10000000   VRoundAnyRawFNToRecFN_4  VBlock    RoundAnyRawFNToRecFN_4:69
```

V6:
```
  %   cumulative   self
 time   seconds   seconds      calls   design                   type      filename and line number
 82.13      1.47     1.47  510000031   -                        VLib      VL_SHIFTRS_WWI(int, int, int, unsigned int*, unsigned int const*, unsigned int)
  5.59      1.57     0.10   10000001   VRoundAnyRawFNToRecFN_4  VBlock    RoundAnyRawFNToRecFN_4:37
```

poemonsense, 2023-10-13:
Did you mean that the performance deviation between both sources is not caused by the RSSHIFT? I didn't know really but this module is believed to have different simulation speed (about 3X slower with V6), as shown in the previous data. The C++ wrapper simulates for 1M cycles.

gezalore, 2023-10-13:
Looks like the extra shifts are introduces by V3Gate, probably clones some expression (that involves the shift) that it should not. You can try `-fno-gate`, but might hurt performance elsewhere. (You can ignore the warning about that option causing scheduling problems. In Verilator V5 it will still be correct.)

gezalore, 2023-10-13:
FWIW: 
```
verilator --stats -cc V3/RoundAnyRawFNToRecFN_4.v -Mdir obj_v3 --dumpi-tree 9
grep -c -R SHIFTRS obj_v3 | sort
```

gezalore, 2023-10-13:
> Did you mean that the performance deviation between both sources is not caused by the RSSHIFT? I didn't know really but this module is believed to have different simulation speed (about 3X slower with V6), as shown in the previous data. The C++ wrapper simulates for 10M cycles.

That's what I meant, but I spoke too early. What I should have said is that the change in logic complexity does not warrant the increase in time. However, see the misoptimization I mentioned above.

poemonsense, 2023-10-13:
> Looks like the extra shifts are introduced by V3Gate, probably clones some expression (that involves the shift) that it should not. You can try `-fno-gate`, but might hurt performance elsewhere. (You can ignore the warning about that option causing scheduling problems. In Verilator V5 it will still be correct.)

I tried this option, and it surprisingly increases the simulation speed of this module in both versions.
V3:  0.727443 -> 0.234865
V6: 2.089827 -> 0.248282

I also tried this option with the original rocket-chip design.
V3: 3,290ms -> 3,602ms
V6: 3,817ms -> 3,727ms

As you said, it does hurt performance elsewhere. The simulation speed increases for V6 but decreases for V3.

gezalore, 2023-10-13:
Would you be willing to track down the issue?

V3Gate.cpp, and anywhere it does `cloneTree` on a compound expression is a suspect.

poemonsense, 2023-10-13:
> Would you be willing to track down the issue?
> 
> V3Gate.cpp, and anywhere it does `cloneTree` on a compound expression is a suspect.

I'll have a try. Just not sure on the timing because I'll be frequently disrupted during the next three weeks. If you or someone else who is more familiar with the code, this issue could be addresses faster.

> Note that this does not necessarily explain all your performance problems, but is the only clue so far, and there is definitely a performance bug lurking in there.

Sure. Tests are just tests. The system is much more complicated than the simple test cases.

poemonsense, 2023-10-14:
> V3Gate.cpp, and anywhere it does `cloneTree` on a compound expression is a suspect.

Commenting out `optimizeElimVar` resolves this issue. 

```
    void optimizeElimVar(AstVarScope* varscp, AstNode* substp, AstNode* consumerp) {
        if (debug() >= 5) consumerp->dumpTree("-    elimUsePre: ");
        if (!m_substitutions.tryGet(consumerp)) m_optimized.push_back(consumerp);
        m_substitutions(consumerp).emplace(varscp, substp->cloneTreePure(false));
    }
```

poemonsense, 2023-10-14:
Simply return from the `const std::function<void(AstNodeVarRef*)> visit` function without doing anything also addresses this issue.

poemonsense, 2023-10-14:
It's creating a lot of substitution/replacement with the SHIFTRS.

```
Suspect:  SHIFTRS 0x2ed2210 {d21cm} @dt=0x2e4e780@(G/sw65)
Suspect: 1: CONST 0x2e25ea0 {d21bp} @dt=0x2e4e780@(G/sw65)  65'sh10000000000000000
Suspect: 2: NOT 0x2ebb130 {d20bg} @dt=0x2e4eb40@(G/w6)
Suspect: 2:1: SEL 0x2ef3740 {d21de} @dt=0x2e4eb40@(G/w6)
Suspect: 2:1:1: VARREF 0x2edf860 {d8ar} @dt=0x2e29c80@(G/w13)  io_in_sExp [RV] <- VARSCOPE 0x2ec80c0 {d8ar} u1=0x2f06200 @dt=0x2e29c80@(G/w13)  TOP->io_in_sExp [T] [scopep=0x2e24e00] -> VAR 0x2ebdce0 {d8ar} @dt=0x2e29c80@(G/w13)  io_in_sExp [PI] INPUT [P] PORT
Suspect: 2:1:2: CONST 0x2e25880 {d21de} @dt=0x2efa780@(G/w32)  32'h0
Suspect: 2:1:3: CONST 0x2e256c0 {d21de} @dt=0x2efa780@(G/w32)  32'h6
Suspect:  SHIFTRS 0x2ed2210 {d21cm} @dt=0x2e4e780@(G/sw65)
Suspect: 1: CONST 0x2e25ea0 {d21bp} @dt=0x2e4e780@(G/sw65)  65'sh10000000000000000
Suspect: 2: NOT 0x2ebb130 {d20bg} @dt=0x2e4eb40@(G/w6)
Suspect: 2:1: SEL 0x2ef3740 {d21de} @dt=0x2e4eb40@(G/w6)
Suspect: 2:1:1: VARREF 0x2edf860 {d8ar} @dt=0x2e29c80@(G/w13)  io_in_sExp [RV] <- VARSCOPE 0x2ec80c0 {d8ar} u1=0x2f06200 @dt=0x2e29c80@(G/w13)  TOP->io_in_sExp [T] [scopep=0x2e24e00] -> VAR 0x2ebdce0 {d8ar} @dt=0x2e29c80@(G/w13)  io_in_sExp [PI] INPUT [P] PORT
Suspect: 2:1:2: CONST 0x2e25880 {d21de} @dt=0x2efa780@(G/w32)  32'h0
Suspect: 2:1:3: CONST 0x2e256c0 {d21de} @dt=0x2efa780@(G/w32)  32'h6
Suspect:  SHIFTRS 0x2ed2210 {d21cm} @dt=0x2e4e780@(G/sw65)
Suspect: 1: CONST 0x2e25ea0 {d21bp} @dt=0x2e4e780@(G/sw65)  65'sh10000000000000000
Suspect: 2: NOT 0x2ebb130 {d20bg} @dt=0x2e4eb40@(G/w6)
Suspect: 2:1: SEL 0x2ef3740 {d21de} @dt=0x2e4eb40@(G/w6)
Suspect: 2:1:1: VARREF 0x2edf860 {d8ar} @dt=0x2e29c80@(G/w13)  io_in_sExp [RV] <- VARSCOPE 0x2ec80c0 {d8ar} u1=0x2f06200 @dt=0x2e29c80@(G/w13)  TOP->io_in_sExp [T] [scopep=0x2e24e00] -> VAR 0x2ebdce0 {d8ar} @dt=0x2e29c80@(G/w13)  io_in_sExp [PI] INPUT [P] PORT
Suspect: 2:1:2: CONST 0x2e25880 {d21de} @dt=0x2efa780@(G/w32)  32'h0
Suspect: 2:1:3: CONST 0x2e256c0 {d21de} @dt=0x2efa780@(G/w32)  32'h6
Suspect:  SHIFTRS 0x2ed2210 {d21cm} @dt=0x2e4e780@(G/sw65)
Suspect: 1: CONST 0x2e25ea0 {d21bp} @dt=0x2e4e780@(G/sw65)  65'sh10000000000000000
Suspect: 2: NOT 0x2ebb130 {d20bg} @dt=0x2e4eb40@(G/w6)
Suspect: 2:1: SEL 0x2ef3740 {d21de} @dt=0x2e4eb40@(G/w6)
Suspect: 2:1:1: VARREF 0x2edf860 {d8ar} @dt=0x2e29c80@(G/w13)  io_in_sExp [RV] <- VARSCOPE 0x2ec80c0 {d8ar} u1=0x2f06200 @dt=0x2e29c80@(G/w13)  TOP->io_in_sExp [T] [scopep=0x2e24e00] -> VAR 0x2ebdce0 {d8ar} @dt=0x2e29c80@(G/w13)  io_in_sExp [PI] INPUT [P] PORT
Suspect: 2:1:2: CONST 0x2e25880 {d21de} @dt=0x2efa780@(G/w32)  32'h0
Suspect: 2:1:3: CONST 0x2e256c0 {d21de} @dt=0x2efa780@(G/w32)  32'h6
```

poemonsense, 2023-10-14:
BTW, I see the commit https://github.com/verilator/verilator/commit/cb60663d49f1f7c77d136a7c5df1dfc95a8a6c4e optimizing the Verilator run-time. Verilator v5.016 on the large design XiangShan has at least 2X speedup of runtime (generating the verilated model) compared with v4.218. It benefits a lot for us. Thank you for the efforts on these optimizations.

poemonsense, 2023-10-14:
> It's creating a lot of substitution/replacement with the SHIFTRS.

On V3, Verilator creates 12 subsitution of SHIFTRS. On V6, Verilator creates 49.

poemonsense, 2023-10-14:
Possibly we should disable the subsitution with some logic if it has a higher overhead, such as the SHIFTRS. For example, optimize it based on the estimated scheduling instruction count for both sides. But I don't really know what the optimization C++ code would look like. @gezalore  Would you mind help me on this? For now, I'm just adding the dirty code to `optimizeElimVar` to avoid subsituting SHIFTRS in. I believe the optimization should be done on the traverse function instead of this one.

```diff
diff --git a/src/V3Gate.cpp b/src/V3Gate.cpp
index a1594e185..698658bbc 100644
--- a/src/V3Gate.cpp
+++ b/src/V3Gate.cpp
@@ -396,6 +396,7 @@ private:

     void optimizeElimVar(AstVarScope* varscp, AstNode* substp, AstNode* consumerp) {
         if (debug() >= 5) consumerp->dumpTree("-    elimUsePre: ");
+        if (!strcmp("SHIFTRS", substp->typeName())) return;
         if (!m_substitutions.tryGet(consumerp)) m_optimized.push_back(consumerp);
         m_substitutions(consumerp).emplace(varscp, substp->cloneTreePure(false));
     }
```

After avoiding the substitution with SHIFTRS, we got significant performance optimizations on the following designs:

- The smallest case `RoundAnyRawFNToRecFN_4` (V3): 0.712716 seconds -> 0.204842 seconds (3.5X speedup)
- The smallest case `RoundAnyRawFNToRecFN_4` (V6): 2.085753 seconds -> 0.217192 seconds (9.6X speedup)
- rocket-chip (V3): 30,712ms -> 27,525ms (1.1X speedup)
- rocket-chip (V6): 35,826ms -> 27,426ms (1.3X speedup)
- XiangShan (V6): 23,551ms -> 23,469ms (no speedup? possibly becasue of other dominant factors)

gezalore, 2023-10-18:
Thanks, looks like a good find, but unfortunately it might be a while before I can look into this. I can try to guide you if you would like to make improvements yourself.

poemonsense, 2023-11-21:
Hopefully I got some time to look into this. Now I'm having the commit here at https://github.com/poemonsense/verilator/commit/6481d7e324b81efcd834fedd5b92e683daf3006d

Here comes the performance results (for inline threshold at 0, 1, 2, respectively):
- `RoundAnyRawFNToRecFN_4` (V6): 2.12, 0.25, 0.25

poemonsense, 2023-11-21:
Besides, I believe the instrCount for some ops are not accurate.

As seen in the previous XiangShan example, avoid inlining all  instr > 1 brings the most significant performance improvement. So there are some operations with instrCount of 2 but have high overhead that should not be inline.

I'm not sure about the details. But is the instrCount of SHIFTRS evaluated by round_up(data_width / 32), given that Verilator is implementing it with a 32bit array? I assume its overhead is much higher than 2 (e.g, a + b + c?).

poemonsense, 2023-11-21:
I used to hear from others that modern compiler (or language? platform? library?) offers wider integer operations like int128. Will Verilator consider optimizing the code for wider signals?

Probably I'm indeed asking for the roadmap or priorities of Verilator. There is plenty of research work on RTL simulators and I think we can seek ways for colloration with the broader community.

wsnyder, 2023-11-21:
Some very rough experiments showed the instrCounts don't make much difference, since the absolute number doesn't control anything critical, so effort hasn't been applied to turning them.

Currently bits wider than 64 bits use a sequence of 32b integers, rather than 64b or even 128b.  I did very detailed experiments on using 64b, and there was a loss for ~~65-95 bits wide due to needing two 64b's, and a small gain for ~~128+ maybe 5% on average.  128b would have even worse tradeoffs as most x86/arm cores have long latencies to move between SIMD and normal register files.  So, the gains are not worth the code complexity, at least for average designs.   The compiler can still choose to promote if it can figure it out, which it might for some easier AND/OR etc operations.

gezalore, 2023-11-22:
instrCounts does have issues, and I think there are cases where even the monotonity implied by Wilson in the previous reply does not hold.

I have also run some experiments of not inlining expressions of various complexity (though I used different metrics, not instrCount - in hindsight I should have used that), but so far I have failed to find a heuristic that does not harm performance significantly on some designs (well, the heuristic of not inlining SHIFTRS works, but is a bit ugly), so more analysis is required indeed. instrCount is probably the best available metric (with some fixes), but you also need to consider the cost of not inlining an expression (potentially storing it to memory and loading it out again) + the serialization non-inlining imposes on the parallel code (if you inline an expression, only its inputs need to be ready before a use can be executed. If you don't inline it, all uses must be after the expression itself is computed.) I suspect this serialization is what harms the MT performance most, but I have no evidence.

poemonsense, 2023-11-22:
Generally I agree with your opinions. I was also talking about the instruction footprint (code size), data footprint (load/store as you mentioned), instruction dependencies (parallel code as you mentioned as well), which are the major impact factors for not inlining.

Inlining seems to be a challenging issue in the general compiler community as well. So Verilator is yet another example of compilers but it generates C++ instead of ELF. I'm not sure about how other compilers deal with inlining. Or do they simply let the user to specify the inline functions?

Like you said, it's a complicated tradeoff thing. This is also what I have found evidence on the previous examples. But I think we can discuss more on this, probably including more compiler experts.

poemonsense, 2023-11-22:
A temp fix I would propose is, to fix the issues on SHIFTRS (or other shifts) first.

1. As they are confirmed to have higher overheads, we can increase their instrCount first.
2. Use instrCount to filter high really high instrCount AST (expected to filter out SHIFTs only).

This is not a general approach but I assume this is one tiny step we could make. It would not be perfect but we can just leave the general and challenging problems for future research.

gezalore, 2023-11-22:
Personally I would be happy for a patch that fixes your case without notably degrading others (e.g.: OpenTitan), but I think you will find using only instrcount it will be hard to achieve that (not degrading others).

One more thing: expressions used only once should always be inlined, no matter how expensive they are.

poemonsense, 2023-11-23:
> One more thing: expressions used only once should always be inlined, no matter how expensive they are.

Isn't it always inlined naturally? I thought assignments should always have zero overhead (in terms of the execution graph)

gezalore, 2023-11-23:
if you have
```
assign b = a + 1;
always @(posedge clk) begin
   c <= b + 2;
end
```
The generated code will first execute the always block, then the continuous assignment if anything changed `a` (this is if `a` is not a top level input, nor does it depend combinationally on a top level input). If you inline `b` into the always block, the assignment is eliminated entirely.

poemonsense, 2023-11-24:
I'll temporarily skip rocket, given the root cause is determined, and look at the xiangshan case (only 50% performance at V6 vs V3).

poemonsense, 2023-12-23:
I just got some time to look into the performance issues on XiangShan with the latest master branch of Verilator (`Verilator 5.019 devel rev v5.018-99-g621de301c`).

I splited the XiangShan RTL and tested every submodule's simulation time. Here's two of the modules with notable performance changes between two RTL versions (I've filtered out some modules whose RTLs seem to notably differ).

One example is the `SelectPolicy_2`. While V3 takes 0.616307 seconds, V6 take 1.846795 seconds (~3X slowdown). Here is a tar ball: [SelectPolicy_2.tar.gz](https://github.com/verilator/verilator/files/13758117/SelectPolicy_2.tar.gz)

```bash
$ ./build.sh v3.v
Building SelectPolicy_2...
Archive ar -rcs VSelectPolicy_2__ALL.a VSelectPolicy_2__ALL.o
$ ./build.sh v6.v
Building SelectPolicy_2...
Archive ar -rcs VSelectPolicy_2__ALL.a VSelectPolicy_2__ALL.o
$ ./obj_dir/v3.v/VSelectPolicy_2 10000000
fun() took 0.620347 seconds to execute 10000000 cycles
$ ./obj_dir/v6.v/VSelectPolicy_2 10000000
fun() took 1.834867 seconds to execute 10000000 cycles
```

In the RTL code, the most notable code style change is how to determine whether some bits are all zeros. In V3, the RTL is generated as an OR-reduction with a NOT, such as `~(|_matrix_3_0_T)`. In V6, the RTL is a concatenation of bits and a comparison with 0, such as `{io_request[0], io_request[1], io_request[2], io_request[3], io_request[4], io_request[5]} == 6'h0;`. That's also why I notice in the `--stats` results, the V6 version has more (5-bit to 31-bit) variables than V3.

For this case, I'm not sure how Verilator interpretes these expressions. Is this performance degradation expected? It's a little bit difficult for me to identify the slowdown sources from the `--prof-cfuncs` results because the report only shows some Verilog line numbers without telling me which Verilator operators are spending the most time.

The other example is the `StatusArray`, where V3 takes 0.726821 seconds and V6 takes 1.138434 seconds. Here a tar ball: [StatusArray.tar.gz](https://github.com/verilator/verilator/files/13758118/StatusArray.tar.gz)

```bash
$ ./build.sh v3.v
$ ./build.sh v6.v
$ ./obj_dir/v3.v/VStatusArray 1000000
fun() took 0.729048 seconds to execute 1000000 cycles
$ ./obj_dir/v6.v/VStatusArray 1000000
fun() took 1.141129 seconds to execute 1000000 cycles
```
In this case, I could only say that I did not find any notable code style differences. If I enable `--prof-cfuncs`, V6 simulates even faster than V3.

@gezalore Would you have time to take a look at these cases? I've uploaded the minimal testing environment. Much appreciate your help on these cases and I'm gald to do further investigations with your guidance.

poemonsense, 2023-12-23:
> Personally I would be happy for a patch that fixes your case without notably degrading others (e.g.: OpenTitan), but I think you will find using only instrcount it will be hard to achieve that (not degrading others). Not saying it's not worth trying, if you are able to do so!

BTW, I propose to integrate some publicly available RTL benchmarks for Verilator to determine its performance, e.g., XiangShan, OpenTitan, Rocket-chip, BOOM, OpenCores, OpenPiton, and more CPU or non-CPU RTL designs, and even UVM projects in the future.

If we have these benchmarks, then the open-source community could set-up an open acceptance criteria for performance-related PRs. These benchmarks would also benefit the community on evaluating different RTL simulators. I believe it would be easier for Verilator to demonstrate its performance as well as for others to improve the performance of Verilator with some specific goals. Also, the benchmarks could always been updated as we understand them much better.

Probably this vision requires significant human resources, and we can start with discussions now with few benchmarks to test the workflow, e.g., XiangShan that we could work together to adapt its code and make it an official benchmark for Verilator.

wsnyder, 2023-12-23:
>BTW, I propose to integrate some publicly available RTL benchmarks for Verilator to determine its performance

This would be great, note there's already some in [verilator_ext_tests](https://github.com/verilator/verilator_ext_tests)

The hard part is making them self checking in a virtualized CI environment that does not have run-to-run stability.  We'd probably need to setup something that does A/B testing by running two jobs simultaneously on the same machine, on a new and old version of verilator.  We could also run some standard benchmark in another thread but the more it deviates from the same model, the more inaccuracy.

wsnyder, 2023-12-24:
Sounds like a good plan, perhaps run nightly unless really slow. We can also setup automerges from the various subrepos into verilator_ext_tests.

poemonsense, 2023-12-26:
> > Yesterday I try force the compiler to create only one binary operator at each line, which should be quite like V3.
>
> This optimizer slurps all combinational logic into a big graph, expressions per line should not matter at all.
>

I doubt this assumption (and/or the Chisel generator) after I split different modules to test their simulation performance. Some modules seem to benefit a lot from Chisel's style control (`maximumNumberOfTermsPerExpression=2`).

Before and after I enable this option, I observe different simulation performance. Here are some representative results:

| Module | V3 | V6 | V6 with `maximumNumberOfTermsPerExpression=2` | Speedup |
| ------ | -- | -- | --- | --- |
| DataArray_6 | 0.670372 | 1.35592 | 0.67354 | 2.013X |
| AsyncRawDataModuleTemplate_10 | 0.71812 | 1.398789 | 0.747556 | 1.871X |
| StatusArray_6 | 5.394065 | 7.964582 | 6.745437 | 1.180X |
| ...... | ...... | ...... | ...... | ...... |
| DebugModule | 0.396174 | 0.876719 | 4.012099 | 0.218X |
| TLPLIC | 1.039263 | 0.259741 | 1.621718 | 0.160X |
| AgeDetector_4 | 0.734709 | 0.313116 | 6.944876 | 0.045X |

The results could be issues of either Chisel/CIRCT or Verilator. Either CIRCT is generating the wrong code with different logic graph 
(it changes the code style as well as the expression trees), or Verilator is not interpreting the syntax tree correctly.

@sequencer Do you know anything about this? Does CIRCT have issues on codegen?

@gezalore Does Verilator have any optimization passes that depend on the code structure instead of pure logic graph? E.g., is `x = a + b; y = x + c` same as `y = a + b + c` for any Verilator passes?

I'll look into the further details once I have enough time for these.

poemonsense, 2023-12-26:
> if you have
>
> ```
> assign b = a + 1;
> always @(posedge clk) begin
>    c <= b + 2;
> end
> ```
>
> The generated code will first execute the always block, then the continuous assignment if anything changed `a` [...]

From the `AgeDetector_4` example, I was able to conclude with one interesting example, probably related to the inline example you provided.

Here comes the original example:

```verilog
module AgeDetector(
  input clock,
  input reset,
  input io_in1,
  input io_in2,
  output io_out
);

  reg value;
  wire next_value = io_in1 & value;
  always @(posedge clock or posedge reset) begin
    if (reset) begin
      value <= 1'h0;
    end
    else begin
      value <= next_value;
    end
  end

  assign io_out = value;
endmodule
```

We expect next_value to be inlined, and the design should be the same as the following one:

```verilog
module AgeDetector(
  input clock,
  input reset,
  input io_in1,
  input io_in2,
  output io_out
);

  reg value;
  always @(posedge clock or posedge reset) begin
    if (reset) begin
      value <= 1'h0;
    end
    else begin
      value <= io_in1 & value;
    end
  end

  assign io_out = value;
endmodule
```

However, these two designs have different simulation performance. The first one (with an explicit `next_value`) takes 15.664788 seconds to simulate 1000000000 cycles, and the latter takes 14.134015 seconds.

Furthermore, if we replace `wire next_value = io_in1 & value;` with `wire next_value = io_in1 & io_in2;`, then the performance gap no longer exists.

I don't really understand why this happens.

gezalore, 2023-12-26:
> I doubt this assumption (and/or the Chisel generator) after I split different modules to test their simulation performance. Some modules seem to benefit a lot from Chisel's style control (maximumNumberOfTermsPerExpression=2).

What I described, about order independence, and treating the logistic as a graph, only applies to the DFG optimizer (V3Dfg*). Other passes might care about different things. In your case, the slowdown is introduced by V3Gate, which does consider the complexity of the right hand sides when making decisions about whether to do a transformation or not.

> Does Verilator have any optimization passes that depend on the code structure instead of pure logic graph?

See above, DFG doesn't care, and DFG was developed based on XiangShan V3 code, hence my earlier (incorrect) suspicion, that DFG might be less able to optimise your V6 code (the real reason is V3Gate, at least for some modules). Many other optimisations do care about the order of statements (e.g. V3Const), and the complexity of expressions (e.g. V3Gate)

gezalore, 2023-12-26:
> From the `AgeDetector_4` example, I was able to conclude with one interesting example, probably related to the inline example you provided.
>
> Here comes the original example:
>
> ```verilog
> module [...]

You generally can't safely inline into a block an expression that reads a variable that is also written by the same block (`value` in your example). V3Gate doesn't distinguish BAs from NBAs, so it won't inline your original. This could be improved by considering NBAs separately.

gezalore, 2025-09-10:
RTLMeter now exists and it confirms Chisel 3 code is still [faster](https://verilator.github.io/verilator-rtlmeter-results/#N4IgbgpgTgzglgewHYgFwGYA0IAmBDAFwgBkE0QAmABgoFYBaKgFnupG3yIAk5zq7GATnoBGKuxAIADgURIAyhAA2EAMazkaANoBdbKrwwIilernaQADTh4kAc3kALW6gC2cJHHqrHcI0vRUVVc8KABrCWtbB2ckNw8vHz9lQJxHKABPSJt7Jxd3T29ff0D3VSgEACMIJB9s6Ly4gsTi5QA2IJDw+tzY+MKk/w60zJ6Y/ISi5KUOsorq2scQPRAoAFcFZTUNFFQtEDtVVWXsGFcTbfM9kAgADzU1ogACAHonmCkICBwT3AgCPBwJQwACCaFABiMADk8K4IGgkGslEpsOskDC4QikSibvdVABJH6oRHIgC+HH+gOBACFwSBIRAMfDidjURsmVjkdg7mpCZylOT6YYIDB8bUlGgCOsIPphTAAKK3VQS1AAMzwwJlQqMMAAYkCiFByFF7CBSUA=) than Chisel 6 code. We will consider this, and the information in this issue during long term performance improvement work, but closing this for how as inactive, but please feel free to follow up.