verilator/verilator #4549: Verilator Makefile doesn't read `CXX` from environment

gussmith23, 2023-10-09:
`Verilator 5.013 devel rev v5.012-74-g446f21d2a`

This may be intended behavior; I'm not sure!

It would be nice (in my opinion) if Verilator allowed for the overriding of the CXX compiler by setting the CXX environment variable. Currently, instead, you can only override CXX in the traditional Makefile way, i.e. `make CXX=...`, which requires you to be calling Verilator from a Makefile, among other things. For me, I call Verilator deep within a larger data generation script, so plumbing through the update to CXX is somewhat ugly.

I suspect this would be an easy change -- where CXX is set at the top of the Verilator Makefile, perhaps it could instead check whether CXX is already set in the environment, and default to "g++" if not.

What do people think?

wsnyder, 2023-10-09:
I thought this worked, but we seem to have = instead of ?= in the include makefile.in.  Perhaps you can try fixing that, ./configure to rebuild, see if it works for you, then submit it a pull if so?

wsnyder, 2023-10-10:
Thanks for the pull!

gussmith23, 2023-10-10:
Thanks for encouraging it!

wsnyder, 2023-10-12:
This was reverted, see discusssion in #4554.

gussmith23, 2023-10-12:
I see, the issue with GNU make is frustrating but understandable. I will keep looking for other solutions.

The -e flag to make may also enable this. If I call make with -e and my Makefile calls Verilator, I'm unsure whether the -e applies to Verilator's invocation of make. If it does, that will be good enough for me, I think.

However, by my understanding after working through this, there's currently no way to change the Verilator CXX compiler when directly calling Verilator on the command line. The only way to change the compiler is to call Verilator from a Makefile, and to call that Makefile in turn with `make CXX=...`. Am I wrong about that? Because that seems odd!

gussmith23, 2023-10-12:
Seems like -e works for my cases, though it might cause other unexpected overrides -- we'll see! Hopefully not.

My point in the above comment still stands though. The ability to override CXX when calling Verilator from a Makefile combined with the inability to override CXX from anywhere else feels odd to me!

wsnyder, 2023-10-12:
It is a bit odd, but given the compiler flags are all discovered at configure time and so the compiler needs to match, probably not an issue most people will hit.