verilator/verilator #4555: Modport Clocking

hassanilyas000, 2023-10-11:
While verilating modport with clocking block I got unsupported error as attached in screen shot. Here's the command that I ran:
`verilator --cc --timing -j 0 clocking.sv`

clocking.sv contains following code:

```
module sample; 
logic clk;
  intf vif(clk);
  
  initial begin
    $display("----------------------");
  $display("Interface Instantiated");
       $display("----------------------");
  end
  
endmodule

interface intf(input bit clk);

logic en,ack;

clocking cb @(posedge clk);
  default input #1 output #2;
    output en;
    input ack;    
  endclocking
  
   modport tb(clocking cb, input clk);
  modport RTL(output en,input ack);
  
  endinterface
```
Here is the output of command:
[image]

wsnyder, 2023-10-11:
As the message says modport clocking is not currently supported.  This is unlikely to soon be supported without a pull request, perhaps you could work on fixing this?

hassanilyas000, 2023-10-12:
Can you give me a starting point how should I dig into this ?

wsnyder, 2023-10-13:
There's information on adding new features in docs/internals.rst.  First step is to make some test_regress tests that pass on other simulators and fail on verilator showing the problem.  Then see where the unsupported error is firing, plump the feature into the AstNode structures in the parser, and decide where to implement them - likely V3Width needs to elaborate them, and I'm not sure if anything else is needed.

A good way to rationalize about this is to see what verilog code not using clocking would be equivalent. Another thing to do is review where AstClocking is currently used in the code base.

hassanilyas000, 2023-10-30:
Before Modport Clocking I think, Clocking block is not fully supported so I tried to add skew in clocking block whether it is with default keyword or defined explicitly I was getting an internal error so I tried to debug code. The code fails while dumping tree at 
V3ast file:
[image]
It is due to doDump. 
Function dumpTreeFile is getting this argument from V3Global.cpp as:
[image]
And this function is used in multiple files.  
So is it possible that someone might help me through any means in understanding the flow and then help me pinpoint the exact point where this feature should be added.

wsnyder, 2023-11-01:
If that is dumping most likely you have a null pointer or something likewise broken (note a pointer of 0x1 means it is something that was deleted).  Run with "--debug --gdbbt" and usually the debugger will point at exactly what is being dereferenced. If that doesn't help see if the backtrace's nodep suggests where the problem is, and/or use the interactive debugger as described in docs/internals.rst.

solomatnikov, 2024-06-26:
@wsnyder since many people are asking about this, it makes sense for Verilator developers to implement this. Otherwise, with other limitations (e.g. #5203) `clocking` is useless.

Yes, I know the standard line - send a PR. The problem is this is complicated to implement even for experienced Verilator developers, and the chances that somebody inexperienced can do it are close to zero.

PR #6436 (2025-09-16): Support modports referencing clocking blocks (#4555)
It fixes #4555.
First it links clocking block references similar to how variables in modports are linked. Then it replaces them with lists of variables in V3AssertPre, when clocking blocks are replaced with variables and other constructs.