verilator/verilator #4560: Segmentation fault on the Verilated model

poemonsense, 2023-10-13:
Sorry but I have not got enough time to minimize the example. I prefer to upload the overall example here first and minimize it later if this issue cannot be identified with this large example. The build does not take a long time (3 minutes to generate the C++ and 8 minutes to compile them using 16 threads).

Verilator 5.017 devel rev v5.014-172-g66cd08960
Ubuntu 20.04

*May we assist you in trying to fix this in Verilator yourself?*
I tried to identify the issue but obviously I didn't succeed. I'm glad to fix it with some help.

This is a single-core XiangShan and it causes segmentation fault during the initial/reset cycles. I built the model with `-g` enabled today (it takes hours to build). Here is the output:
```C++
Thread 1 "emu" received signal SIGSEGV, Segmentation fault.
0x00000000038aeaca in VSimTop___024root___stl_sequent__TOP__17 (vlSelf=0x7ffff6365100) at VSimTop___024root__DepSet_hd5918264__90__Slow.cpp:4386
4386                = vlSelf->SimTop__DOT__l_soc__DOT__core_with_l2__DOT__core__DOT__frontend__DOT__bpu__DOT__predictors__DOT__ftb__DOT__ftbBank__DOT__ftb__DOT__array_0__DOT__array_0_3_ext__DOT__RW0_random[0U];
(gdb) where
#0  0x00000000038aeaca in VSimTop___024root___stl_sequent__TOP__17 (vlSelf=0x7ffff6365100) at VSimTop___024root__DepSet_hd5918264__90__Slow.cpp:4386
#1  0x0000000000509a77 in VSimTop___024root___eval_stl (vlSelf=0x7ffff6365100) at VSimTop___024root__DepSet_hd5918264__219__Slow.cpp:2775
#2  0x00000000004bf1da in VSimTop___024root___eval_settle (vlSelf=0x7ffff6365100) at VSimTop___024root__DepSet_hd5918264__74__Slow.cpp:1026
#3  0x0000000000433f7c in VSimTop::eval_step (this=0xb823480) at VSimTop.cpp:69
#4  0x000000000041a50b in VSimTop::eval (this=0x7ffff63cc5e0) at ./VSimTop.h:67
#5  Emulator::reset_ncycles (this=0xb823360, cycles=10) at /home/xuyinan/xs/XiangShan/difftest/src/test/csrc/verilator/emu.cpp:550
#6  0x0000000000418d9d in Emulator::Emulator (this=0xb823360, argc=3, argv=0x7fffffffe018) at /home/xuyinan/xs/XiangShan/difftest/src/test/csrc/verilator/emu.cpp:348
#7  0x00000000004118fb in main (argc=0, argv=0x7fffffffe018) at /home/xuyinan/xs/XiangShan/difftest/src/test/csrc/common/main.cpp:41
(gdb)
```

Here is the corresponding C++ code:
```
// build/emu-compile/VSimTop___024root__DepSet_hd5918264__90__Slow.cpp

 4384     } else {
 4385         vlSelf->SimTop__DOT__l_soc__DOT__core_with_l2__DOT__core__DOT__frontend__DOT__bpu__DOT__predictors__DOT__ftb__DOT__ftbBank__DOT__ftb__DOT___array_0_RW0_rdata[0U]
 4386             = vlSelf->SimTop__DOT__l_soc__DOT__core_with_l2__DOT__core__DOT__frontend__DOT__bpu__DOT__predictors__DOT__ftb__DOT__ftbBank__DOT__ftb__DOT__array_0__DOT__array_0_3_ext__DOT__RW0_random[0U];
 4387         vlSelf->SimTop__DOT__l_soc__DOT__core_with_l2__DOT__core__DOT__frontend__DOT__bpu__DOT__predictors__DOT__ftb__DOT__ftbBank__DOT__ftb__DOT___array_0_RW0_rdata[1U]
 4388             = vlSelf->SimTop__DOT__l_soc__DOT__core_with_l2__DOT__core__DOT__frontend__DOT__bpu__DOT__predictors__DOT__ftb__DOT__ftbBank__DOT__ftb__DOT__array_0__DOT__array_0_3_ext__DOT__RW0_random[1U];
 4389         vlSelf->SimTop__DOT__l_soc__DOT__core_with_l2__DOT__core__DOT__frontend__DOT__bpu__DOT__predictors__DOT__ftb__DOT__ftbBank__DOT__ftb__DOT___array_0_RW0_rdata[2U]
 4390             = (0x3ffffU & vlSelf->SimTop__DOT__l_soc__DOT__core_with_l2__DOT__core__DOT__frontend__DOT__bpu__DOT__predictors__DOT__ftb__DOT__ftbBank__DOT__ftb__DOT__array_0__DOT__array_0_3_ext__DOT__RW0_random[2U]);
 4391     }
```

Here I attach the build environment without compiled files because they are too large to upload. It should have the same C++ sources as generated by the following commands.

```
export NOOP_HOME=/path/to/case
make emu -jN EMU_TRACE=1
./build/emu -i ./Makefile --no-diff
```

Output looks like
```
➜  case_segmentation_fault ./build/emu -i ./Makefile --no-diff
emu compiled at Oct 13 2023, 22:48:27
Using simulated 32768B flash
isWriteFetchToIBufferTable0 = 0
isWriteIfuWbToFtqTable0 = 0
isWriteFTQTable0 = 0
isWriteBankConflictTable0 = 0
depth0 = 32
enableDynamicPrefetcher0 = 1
StoreWaitThreshold_0 = 0
nMaxPrefetchEntry0 = 14
isWriteL1MissQMissTable0 = 0
isWriteLoadMissTable0 = 0
isFirstHitWrite0 = 0
isWriteLoadAccessTable0 = 0
isWriteL2TlbPrefetchTable0 = 0
isWriteL1TlbTable0 = 0
isWritePageCacheTable0 = 0
isWritePTWTable0 = 0
isWriteL2TlbMissQueueTable0 = 0
always_update0 = 1
l1_stride_ratio0 = 2
l2_stride_ratio0 = 5
l2DepthRatio0 = 2
l3DepthRatio0 = 3
enableL3StreamPrefetch0 = 0
enableL1StreamPrefetcher0 = 1
ColdDownThreshold_0 = 12
ForceWriteUpper_0 = 60
ForceWriteLower_0 = 55
StoreBufferThreshold_0 = 7
StoreBufferBase_0 = 4
CorrectMissTrain0 = 0
isWriteInstInfoTable0 = 0
[1]    1496177 segmentation fault (core dumped)  ./build/emu -i ./Makefile --no-diff
```

With gdb will see the abortion is due to the previous lines in the verilated model.

poemonsense, 2023-10-13:
Sorry. Forgot to upload the case.

[case_segmentation_fault.tar.gz](https://github.com/verilator/verilator/files/12894862/case_segmentation_fault.tar.gz)

wsnyder, 2023-10-13:
If GDB is correct about the line (Use -O0), it's dereferencing a variable. See where that variable is declared to make sure it makes sense.

Also it shouldn't make a segfault but make sure you didn't run out of memory.

poemonsense, 2023-10-14:
We just found that it was due to the stack overflow. Using `ulimit -s 16384` (default is 8192) solves this issue.

I assume this issue is also related to the #4547 to some extent. Using the V3 style, we never encountered this issue before. Thus, there should be something that changes how the verilated model is built when we change from V3 to V6.

closed 2023-10-14 by poemonsense