verilator/verilator #4565: Fix VlProcess argument inconsistent across virtual member functions

wsnyder, 2023-10-15:
When Verilator inserts a VlProcess argument to a member function, it doesn't add it to all virtual base classes' similar functions.

This causes a clang (not gcc) compile error.  (Plus the code won't work of course).  e.g. in UVM:

```
In file included from ./Vt_uvm_todo_uvm_pkg__03a__03auvm_object_string_pool__Tz2__Vclpkg.h:47:
./Vt_uvm_todo_uvm_pkg__03a__03auvm_pool__Tz5_TBz2__Vclpkg.h:64:18: error: 'Vt_uvm_todo_uvm_pkg__03a__03auvm_pool__Tz5_TBz2::__VnoInFunc_get_type_name' hides over\
loaded virtual function [-Werror,-Woverloaded-virtual]
    virtual void __VnoInFunc_get_type_name(Vt_uvm_todo__Syms* __restrict vlSymsp, std::string &get_type_name__Vfuncrtn);
                 ^
./Vt_uvm_todo_uvm_pkg__03a__03auvm_object__Vclpkg.h:77:18: note: hidden overloaded virtual function 'Vt_uvm_todo_uvm_pkg__03a__03auvm_object::__VnoInFunc_get_typ\
e_name' declared here: different number of parameters (3 vs 2)
    virtual void __VnoInFunc_get_type_name(VlProcessRef vlProcess, Vt_uvm_todo__Syms* __restrict vlSymsp, std::string &get_type_name__Vfuncrtn);
                 ^
```

Blocks #1538 runtime.

kiryk, 2023-10-23:
I've been failing to reproduce it. This issue comes from a day before I opened a PR with a rework of VlProcess propagation (#4569).

It seems you were runing `t_uvm_todo.pl`. I've pulled current `master`, then allowed the test to compile and run ([diff](https://github.com/verilator/verilator/files/13067546/t_uvm_todo.txt)), it finished without failure.

Does the bug still occur to you, @wsnyder?

wsnyder, 2023-10-23:
Indeed!  Yea!

closed 2023-10-23 by wsnyder