verilator/verilator #4566: Fix VlProcess argument breaks DPI export

wsnyder, 2023-10-15:
Verilator's insertion of the VlProcess argument breaks any DPI export of that function.

E.g. with UVM:

```
Vt_uvm_todo__Dpi_Export__0.cpp: In static member function 'static void Vt_uvm_todo::m___05Fuvm_report_dpi(int, const char*, const char*, int, const char*, int)':
Vt_uvm_todo__Dpi_Export__0.cpp:32:14: error: could not convert '(Vt_uvm_todo__Syms*)__Vscopep->VerilatedScope::symsp()' from 'Vt_uvm_todo__Syms*' to 'VlProcessRe\
f' {aka 'std::shared_ptr<VlProcess>'}
   32 |     (*__Vcb)((Vt_uvm_todo__Syms*)(__Vscopep->symsp()), severity__Vcvt, id__Vcvt, message__Vcvt, verbosity__Vcvt, filename__Vcvt, line__Vcvt);
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |              |
      |              Vt_uvm_todo__Syms*
```

Blocks #1538 runtime.

See also #4565

I'm not sure what VlProcess should be used by the call into the export; perhaps we need to save the current process as part of the DPI context.

kbieganski, 2024-12-04:
Also relevant: https://github.com/verilator/verilator/issues/4225#issuecomment-1562738548
(since `VlProcess` is entwined with `--timing`)

wsnyder, 2025-08-31:
I'm unable to recreate this in master, changing t_uvm_todo to use the import/exports works fine (not committed - though did commit some other cleanups). Maybe #5623 #5650 fixed it.  I'll reopen if hit it again.

closed 2025-08-31 by wsnyder