verilator/verilator #4574: Should waiver output include lints that are already ignored using comments in the source file?

Barabas5532, 2023-10-16:
I tried generating a waiver file using the `--waiver-output` flag to quickly ignore some warnings while I plan to fix them later. It looks like the waiver file has entries for warnings that are already ignored inline in the source files using `// verilator lint_off` comments. For example, x0 to x31 from this file: https://github.com/Barabas5532/crush/blob/6e297860a34fa724f010ce467c3b4c5dc2b4814d/rtl/cpu/registers.v#L16 are included in the waiver file:

```
// lint_off -rule UNUSEDSIGNAL -file "*src/crush_0.0.1/rtl/cpu/registers.v" -match "Signal is not used: 'x0'"

// lint_off -rule UNUSEDSIGNAL -file "*src/crush_0.0.1/rtl/cpu/registers.v" -match "Signal is not used: 'x1'"

// lint_off -rule UNUSEDSIGNAL -file "*src/crush_0.0.1/rtl/cpu/registers.v" -match "Signal is not used: 'x2'"

// etc
```

Shouldn't the waiver file only include warnings that are not already ignored using comments? The [docs](https://verilator.org/guide/latest/exe_verilator.html?highlight=waiver#cmdoption-waiver-output) say it will ignore warnings emitted in this session, and no warnings about x0, etc. were printed because of the `lint_off` comments.

I want to use inline ignores to mark lints that I do not plan to fix, and the waiver file to mark temporarily ignored lints that I will come back to fix later. With the current behaviour even the permanently ignored lints are showing up in the waiver file, and I have to do some manual cleanup to remove these. This is not really going to scale as I do more work and add new temporary ignores to the waiver file.

wallento, 2023-10-17:
Yes, you are right, and I agree that the described behavior is probably what everyone expects.

Can you have a look at the code on how to fix that? I can guide you, but probably cannot fix it myself quickly.

joschroeter, 2023-12-06:
Hi, 
I am interested on working on this issue.

wallento, 2023-12-06:
Great, assigned!

wsnyder, 2024-01-24:
Thanks for your work on this.

closed 2024-01-24 by wsnyder