verilator/verilator #4575: Design's inputs are blocked during simulation when using an array as input

youfares, 2023-10-17:
verilator version : 5.012

I created a SV design to do a bitwise AND between 3 separated inputs of 8-bits and another bitwise AND between 3 inputs, 2 of them are packed in an array as illustrated bellow. Then, I created a SV testbench to test this module with randomized inputs in every iteration.  The results at the end show that the inputs packed in the array are blocked after the first iteration unlike the other inputs and the fact that the out_1 changes when C changes mean that design is simulated only the array inputs are blocked. Is it that verilator can't handle array inputs or am I missing something like an option that I should use. Thank in advance for you help.

//design
module And ( output wire [7:0] out_0,
             output wire [7:0] out_1,
        	 input wire [7:0] In [1:0],
             input wire [7:0] In0,
             input wire [7:0] In1,
             input wire [7:0] C);

    assign out_1 = In[0] & In[1] & C ;
    assign out_0 = In0 & In1 & C ;

    always @ (In[0], In[1], In0, In1) begin
        $display("design_level : in_array = {%b, %b, %b} | in_sing = {%b, %b, %b}", In[0], In[1], C, In0, In1, C);
    end
    endmodule

// Testbench
`include "And.sv"
module Test ;
          //cell's parameters and variables 
    parameter w=8 ; //w=width of data
    parameter stimu_num = 20; //number of stimulus
    //inputs
    reg [w-1:0] In0;//an array of N inputs with w width
    reg [w-1:0] In1;//an array of N inputs with w width
    reg [w-1:0] In [1:0];//an array of N inputs with w width
    reg [7:0] C = 8'b00000000;
    int i = 0;
    //outputs
    wire [w-1:0] out_0;
    wire [w-1:0] out_1;
    int tmp_rand;
      //instances
    And and_inst_0 (.out_0(out_0), .out_1(out_1), .In(In), .In0(In0), .In1(In1), .C(C) );
    //And and_inst_1 ( .out(out_1), .In(In));
      initial begin
        repeat (stimu_num) begin
            $display("iteration %0d", i);
            C =~C;
            tmp_rand = $urandom_range(0, (2**w)-1); 
            In0 = tmp_rand[7:0];
            tmp_rand= $urandom_range(0, (2**w)-1); 
            In1 = tmp_rand[7:0];
            In ={In1, In0};
            #5; //signal propagation delay 
                //chack and operator first instance with no array
            $display("Test ");
            if (out_0 !== (In0 & In1 & C))
                $display("AND gate test failed for In0=%b, In1=%b and result is out_sing=%b", In0, In1, out_0);//
            else 
                $display("AND gate test passed for In0=%b, In1=%b and result is out_sing=%b", In0, In1, out_0);

            //chack and operator second instance with ana array 
            if (out_1 !== (In[0] & In[1] & C))
                $display("AND gate test failed for In0=%b, In1=%b and result is out_arr=%b", In[0], In[1], out_1);
            else 
               $display("AND gate test passed for In0=%b, In1=%b and result is out_arr=%b", In[0], In[1], out_1);
            i++;
            end
        $finish;
      end
endmodule

//results
iteration 0
design_level : in_array = {11000100, 10011100, 11111111} | in_sing = {11000100, 10011100, 11111111}
Test 
AND gate test passed for In0=11000100, In1=10011100 and result is out_sing=10000100
AND gate test passed for In0=11000100, In1=10011100 and result is out_arr=10000100
iteration 1
design_level : in_array = {11000100, 10011100, 00000000} | in_sing = {00000010, 11100100, 00000000}
Test 
AND gate test passed for In0=00000010, In1=11100100 and result is out_sing=00000000
AND gate test passed for In0=00000010, In1=11100100 and result is out_arr=00000000
iteration 2
design_level : in_array = {11000100, 10011100, 11111111} | in_sing = {01111000, 10111100, 11111111}
Test 
AND gate test passed for In0=01111000, In1=10111100 and result is out_sing=00111000
AND gate test failed for In0=01111000, In1=10111100 and result is out_arr=10000100
iteration 3
design_level : in_array = {11000100, 10011100, 00000000} | in_sing = {10110110, 11100100, 00000000}
Test 
AND gate test passed for In0=10110110, In1=11100100 and result is out_sing=00000000
AND gate test passed for In0=10110110, In1=11100100 and result is out_arr=00000000

wsnyder, 2023-10-17:
Thanks for making a good self contained test. Your test passes for me.  As you aren't using the most recent version, perhaps this was already fixed, please try with the most recent version.

youfares, 2023-10-17:
Thank you for your answer, okay I will try with the most recent version. Can you please give me the options you used as it passes for you.

wsnyder, 2023-10-17:
Just `verilator --binary`

youfares, 2023-10-18:
Okay, Thank you so much

wsnyder, 2023-11-26:
Assuming your issue got resolved in latest version, otherwise feel free to follow up here.

closed 2023-11-26 by wsnyder

youfares, 2023-11-27:
The problem has been resolved in version 5.016. I tested it, and it functioned perfectly. I appreciate your diligent follow-up.