verilator/verilator #4577: Can __VForceRd be declared as public when "forceable" is used ?

fredrequin, 2023-10-17:
Hello,
this is a side issue of issue #4514.
Sometimes, the __VForceRd signal does not show as public in the generated C++ code.
We do need it to correctly implement the "signal_deposit" keyword.
I have noticed that one line is missing in V3Force.cpp:
```
    void visit(AstVarScope* nodep) override {
        // If this signal is marked externally forceable, create the public force signals
        if (nodep->varp()->isForceable()) {
            const ForceComponentsVarScope& fc = getForceComponents(nodep);
            fc.m_rdVscp->varp()->sigPublic(true); <=== HERE
            fc.m_enVscp->varp()->sigPublic(true);
            fc.m_valVscp->varp()->sigPublic(true);
        }
    }

```
Can it be added ?
Regards,
Frédéric REQUIN

gezalore, 2023-10-18:
> Hello, the "signal_deposit" keyword is implemented as follow:
> 
> * Set the __VForceEn and __VForceVal signals
> * Wait for (__VForceRd & __VForceEn) to be equal to (__VForceVal & __VForceEn)
> * Clear __VForceEn

You should not have to "wait", forceRd should have the appropriate forced value after the next call to eval, after you set the force control signals. If it does not that is possibly a bug that you should report.

fredrequin, 2023-10-18:
Well, I am using Verilator v4.218.
I am not sure this is true for this old version.
For example, I can see that if I have two "signal_deposit", they are spaced by one clock cycle.
I still need to investigate this one line change in V3Force.cpp : I was expecting to see just a signal rename (privage -> public) but I see some statements being dupplicated in the generated C++.

gezalore, 2023-11-05:
> You should not have to "wait", forceRd should have the appropriate forced value after the next call to eval, after you set the force control signals. If it does not that is possibly a bug that you should report.

There was a bug that prevented this from working, which will be fixed in #4668

commit c563175f (2023-11-05): Fix scheduling of external force signals (#4668)

Used to set the wrong public flag on forceEn/forceVal, which means they
were not included in ICO as necessary, but V3Gate tended to inline them,
so this was hard to hit.

Fixes #4577
closed 2023-11-05 by gezalore