verilator/verilator #4578: deep loop cause C++ code redeclaration error

kranerup, 2023-10-17:
This compilation error seems to be related to initial blocks and loop unrolling. I have attached a verilog file that cause the compilation error.

[t.txt](https://github.com/verilator/verilator/files/12950178/t.txt)

To reproduce with the attached example:
verilator --trace-fst --compiler gcc -O3 -cc --vpi --exe --top-module top t.v -Wno-fatal  --unroll-stmts 600000 --unroll-count 600000 -sv
make -C obj_dir -f Vtop.mk

Causes the following g++ error:
Vtop___024root__DepSet_heccd7ead__0__Slow.cpp:83:19: error: redefinition of ‘void Vtop___024root___eval_initial__TOP__0(Vtop___024root*)’
   83 | VL_ATTR_COLD void Vtop___024root___eval_initial__TOP__0(Vtop___024root* vlSelf) {

Verilator 5.014 2023-08-06 rev v5.014-35-ge6b0bdd4d
Ubuntu 22.04

> May we assist you in trying to fix this in Verilator yourself?
Sure.

wsnyder, 2023-10-17:
This is likely something already fixed, please try the trunk version.

kranerup, 2023-10-18:
Indeed. It works on trunk version. Thanks and sorry, I should have tried trunk version before reporting.

closed 2023-10-18 by kranerup