verilator/verilator #4584: Performance Issue When Simulating A Ring Oscillator

qrqiuren, 2023-10-19:
Hi,

I wrote a mixed-signal SystemVerilog model of a ring oscillator. It uses timescale 1s/1fs. The period of the oscillator is about 1ns, but it takes about 20s wall clock time to simulate the whole testbench of 3ns.

From the print messages, we can see from 0ns~2ns, it is very fast. But in 2ns~3ns, it takes most of the wall clock time.

The detailed information is as the following.

> Thanks for taking the time to report this.

> Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

Verilation command: `verilator --language 1800-2017 --top tb_osc --binary -j 0 -O3 --trace-fst --trace-threads 2 --threads `nproc` --assert -DSIMULATION --autoflush --prof-cfuncs`

Run command: `./obj_dir/Vtb_osc`

`tb_osc.sv`:

```systemverilog
module tb_osc;

timeunit 1s;
timeprecision 1fs;

// Keep this string identical to the module name
localparam string module_name = "tb_osc";

// ******************
// Signal Definitions
// ******************

logic dco_out;

// *****************
// Module Connection
// *****************

bhv_dco dco (
    // Inputs
    .coarse_cw(8.0),
    .medium_cw(8.0),
    .fine_cw(32.0),

    // Outputs
    .rf_out(dco_out)
);

// ******************
// Behavioral Stimuli
// ******************

initial fork
    $display("Start simulation and dump waveform to ", {module_name, ".fst"});
    for (;;) begin
        $display("Current time: %g", $realtime);
        #1ns;
    end
join

// Record waveform
initial begin
    $dumpfile({module_name, ".fst"});
    $dumpvars;
    #3ns $finish;
end

endmodule : tb_osc
```

`bhv_dco.sv`:

```systemverilog
module bhv_dco #(
    parameter string jitter_file_name = "dco_pn.txt",

    parameter realtime coarse_ofst = 600ps,
    parameter realtime coarse_res  = 60ps,
    parameter realtime medium_ofst = 130ps,
    parameter realtime medium_res  = 6ps,
    parameter realtime fine_ofst   = 70ps,
    parameter realtime fine_res    = 0.2ps
) (
    input real coarse_cw,
    input real medium_cw,
    input real fine_cw,
    output logic rf_out
);

timeunit 1s;
timeprecision 1fs;

// Construct jitter arrays

int fjitter, jitter_arr_size, jitter_pos;
realtime jitter_arr[];

initial begin
    fjitter = $fopen(jitter_file_name, "r");
    $fscanf(fjitter, "%d", jitter_arr_size);
    jitter_arr = new[jitter_arr_size];
    jitter_pos = 0;

    for (int i=0; i<jitter_arr_size; i++) begin
        $fscanf(fjitter, "%g", jitter_arr[i]);
    end
end

// Delay calculation

realtime coarse_delay, medium_delay, fine_delay, jitter;

assign coarse_delay = 0.5 * (coarse_cw * coarse_res + coarse_ofst         );
assign medium_delay = 0.5 * (medium_cw * medium_res + medium_ofst         );
assign   fine_delay = 0.5 * (  fine_cw *   fine_res +   fine_ofst + jitter);

assign jitter = jitter_arr[jitter_pos];

// Circuit connection

logic coarse_out, medium_out, fine_out;

// Use transport delay to model the delay of the coarse bank, which is
// constructed by logic gate delays.
initial coarse_out = 0;
always @ (fine_out) coarse_out <= #coarse_delay ~fine_out;

// Use inertial delay to model medium and fine banks, which are RC delays.
assign #medium_delay medium_out = ~coarse_out;
assign #fine_delay   fine_out   = ~medium_out;

// Trigger jitter array
always @ (posedge coarse_out)
    jitter_pos <= jitter_pos >= jitter_arr_size - 1 ? 0 : jitter_pos + 1;

// Final output
assign #50ps rf_out = fine_out;

endmodule : bhv_dco

```

> What 'verilator --version' are you using?  Did you try it with the git master version?

Yes, it is the latest git master.

```
Verilator 5.017 devel rev v5.014-206-g0c2bab1f6
```

> What OS and distribution are you using?

```
  Operating System: Rocky Linux 8.8 (Green Obsidian)
       CPE OS Name: cpe:/o:rocky:rocky:8:GA
            Kernel: Linux 6.4.10-1.el8.elrepo.x86_64
      Architecture: x86-64
```

> May we assist you in trying to fix this in Verilator yourself?

Although I'm looking into the C++ code, now I still have no idea...

qrqiuren, 2023-10-19:
I modified the time recording block to show the wall clock time. And the simulation time is modified to 5ns.

```systemverilog
initial fork
    $display("Start simulation and dump waveform to ", {module_name, ".fst"});
    for (;;) begin
        $system("echo \"[\" `date +%H:%M:%S.%N` \"]\"");
        $display("Current time: %g", $realtime);
        #1ns;
    end
join
```

Output:

```
[ 12:36:59.730399142 ]
Current time: 0
[ 12:36:59.805627217 ]
Current time: 1e-09
[ 12:37:01.675904127 ]
Current time: 2e-09
[ 12:37:13.439497779 ]
Current time: 3e-09
[ 12:37:27.959531514 ]
Current time: 4e-09
[ 12:37:42.273408397 ]
Current time: 5e-09
```

You can see the time taken by each nanosecond of simulation time

| Sim time | Wall clock time |
| ------------ | ------------ |
| 0ns ~ 1ns | 0.03s |
| 1ns ~ 2ns | 1.9s |
| 2ns ~ 3ns | 11.8s |
| 3ns ~ 4ns | 14.5s |
| 4ns ~ 5ns | 14.3s |

wsnyder, 2023-10-19:
Can you please enable g++ profiling as described https://veripool.org/guide/latest/simulating.html#code-profiling , compile with "-g -O0" and post the report showing where the time is spent?

qrqiuren, 2023-10-19:
Thanks. The `gprof obj_dir/Vtb_pll gmon.out` result is like this: (I just changed the name of `tb_osc` to `tb_pll`)

[analysis.txt](https://github.com/verilator/verilator/files/13037469/analysis.txt)

qrqiuren, 2023-10-19:
I changed `-O3` to `-O0` in verilator command args. It took 10 mins and still not complete the verilation at this moment...

qrqiuren, 2023-10-19:
Oh, sorry. I just realized the above `-O3` and `-O0` are for verilated models. Now I deleted the arguments.

I added `-CFLAGS -O0 -CFLAGS -g` to the `verilator` command, and modified simulation time to 6ns. The following is the `gprof` analysis.

[gprof.txt](https://github.com/verilator/verilator/files/13038824/gprof.txt)

Of course the wall clock time spent is about 6--8 times more than the reported data in the above replies.

qrqiuren, 2023-10-19:
I used a constant-delay ring oscillator model to simulate and found no such problem. Maybe the problem is variable delay.

Constant delay RO model:

```systemverilog
module bhv_const_ro (
    output logic rf_out
);

timeunit 1s;
timeprecision 1fs;

// Circuit connection

logic coarse_out, medium_out, fine_out;

// Use transport delay to model the delay of the coarse bank, which is
// constructed by logic gate delays.
initial coarse_out = 0;
always @ (fine_out) coarse_out <= #650ps ~fine_out;

// Use inertial delay to model medium and fine banks, which are RC delays.
assign #250ps medium_out = ~coarse_out;
assign #100ps   fine_out = ~medium_out;

// Final output
assign #50ps rf_out = fine_out;

endmodule : bhv_const_ro
```

qrqiuren, 2023-10-20:
I think I almost located (maybe one of) the problem. It is in the inertial delay assignments, like (in the above file `bhv_
const_ro.sv` for example):

```
assign #250ps medium_out = ~coarse_out;
assign #100ps   fine_out = ~medium_out;
assign #50ps rf_out = fine_out;
```

Add the following code into `VlDelayScheduler::delay()` in file `verilated_timing.h` to see the scheduler's queue:

```
        VL_PRINTF("[VlDelayScheduler]@%d\n", m_context.time());
        VL_PRINTF("File name: %s\n", filename);
        VL_PRINTF("Line number: %d\n", lineno);
        VL_PRINTF("Delay: %lld time units\n", delay);
        VL_PRINTF("Currently in the queue:\n");
        for (auto& coro : m_queue) {
            VL_PRINTF("    %lld time units\n", coro.m_timestep);
        }
        VL_PRINTF("---------------------------------\n");
```

We can get the following output at stdout:  [test_const_ro.txt](https://github.com/verilator/verilator/files/13050641/test_const_ro.txt)

The simulation only runs for 5ns (=5000ps). In the output, by using find function of our text editor, pushing 650ps delay into the queue only happens 6 times, because `fine_out` exactly generates 6 edges during the 5ns simulation. But pushing 250ps, 100ps, and 50ps happens 102 times, 202 times, and 202 times.

I think wire assignments like `assign #sometime some_signal = some_other_signal` should be processed similarly to `always @(some_other_signal) #sometime some_signal = some_other_signal`, except that the values in the `assign` statement should initialize at the startup. I'm continuing checking the problem.

I also did the same thing on the testbench with `bhv_dco`, but the output file is extremely huge (just about 2 seconds it became 2.2GB). I think it is better to take a look at first on the above file.

mathieu-tp, 2023-10-20:
Hello,
As far as I know, "inertial delay" is not implemented into Verilator. So "assign" statement gives a "transport" delay result. Not sure that it's the explanation of your problems.  See #3730 and #3711 for this issue.
As a workaround you may use fork/join_any statements to emulate "inertial" delay

wsnyder, 2023-10-20:
Aside from the larger algorithm, I think some code tunings may help significantly.  Try changing the existing VlDelayedCoroutingQueue vector, which will require a lot of copying which is inefficient, into a std::set ordered by the existing operator<.

wsnyder, 2023-10-20:
(P.S. a pull to fix the std::set, if it helps, would be appreciated even if you rewrite the code to fix the performance otherwise.)

qrqiuren, 2023-10-21:
@mathieu-tp Thanks for the information. I will look into it.

@wsnyder Thanks. Although it seems that `gprof` gives most of the time on `VlDelayedCorouting::operator<`, I think the real problem is that too many things are pushed into the queue. It might because of wrong implementation of inertial delay.

qrqiuren, 2023-10-21:
After my survey on `test_const_ro.txt`, I found interesting things.

In the testbench of `bhv_const_ro`, the greatest common divisor (GCD) of all delays is 50ps. The delay of 250ps is pushed into the queue *once* every 50ps, and the delay of 100ps and 50ps are pushed into the queue `twice` every 50ps.

I can mostly say, this is the problem of the implementation of inertial delay.

As a result, in the `bhv_dco` model, the frequency of pushing is much more higher because it has small random jitter read from external files, which leads to much smaller GCD of historical time delays. This file is the first 1000000 lines of the stdout of simulating the model of `bhv_dco`: [test_dco_head.txt](https://github.com/verilator/verilator/files/13059583/test_dco_head.txt)

wsnyder, 2023-10-21:
Nearly all the operator< time will go away if you change the container.  It's because it needs to compare Deprh^2/2 times instead of log(depth) times.

wsnyder, 2023-10-21:
Actually I think we want a std::priority_queue, as we only need access to the lowest time entry.

wsnyder, 2023-11-11:
While not the ultimate optimization, I fixed the std::vector discussed here, it gave a 3x speedup.

PR #4669 (2023-11-11): Optimize timing-delayed queue (#4584)
Fixes #4584
closed 2023-11-11 by wsnyder