verilator/verilator #4589: default keyword does not apply recursively to subarrays in an array assignment expression

jamesrbailey, 2023-10-20:
In array assignment patterns, verilator does not apply the default keyword recursively to subarrays.

For example, in this case:
```verilog
module dut;
   int a[4][4] = '{default: 0};
endmodule
```
verilating gives this error:
```
%Error: dut.sv:2:30: CONST '32'sh0' is not an unpacked array, but is in an unpacked array context
    2 |     int a[4][4] = '{default: 0};
      |                              ^
```

The IEEE 1800-2017 SystemVerilog standard says this (emphasis mine),
>The default:value applies to members that are not matched by either member name or type key.
If the member type is a simple bit vector type, matches the self-determined type of the value, or is
not an array or structure type, then the value is evaluated in the context of each assignment to a
member by the default and shall be castable to the member type; otherwise, an error is generated.
**For unmatched structure members, the type and default specifiers are applied recursively according
to the rules in this subclause to each member of the substructure.** For unmatched array members, the
type and default keys are applied to the array according to the rules for arrays.

I am using `Verilator 5.015 devel rev v5.014-82-g567644313`, which is fairly recent. I don't see any release notes indicating that this has subsequently been fixed.

I note also that several commercial simulators support this.

I am interested in contributing to the Verilator project, but it will likely be some time before I can contribute productively. If I can be pointed in the right direction I may try to look at fixing this issue.

Ps. for those looking for a temporary workaround you can nest your default statements and Verilator will not throw an error (although I have not functionally tested this):
```verilog
module dut;
   int a[4][4] = '{default: '{default: 0}};
endmodule
```

wsnyder, 2023-10-20:
The default handling code is fairly isolated in V3Width, so if you get a chance to look at it should be straight forward to fix.

alibillalhammoud, 2025-12-14:
Hey,  
I took a look at V3Width.cpp and found the patternArray function, [here](https://github.com/verilator/verilator/blob/dc00bf248413c0c40c3f144764691e7e0ed38009/src/V3Width.cpp#L5082), would need modifications. This is my first time peeking at this project so it may take me sometime. Will make an attempt over the long break (coming weeks) if I feel motivated 😄.

closed 2026-03-05 by wsnyder