verilator/verilator #4590: Escaped identifier grammar is not the same as the standard

dwRchyngqxs, 2023-10-20:
Empty escaped identifier are not supported but are allowed by the Verilog standard A.9.3.

This file is rejected
```
module xorator(input a, b, output \ );
    assign \ = a ^ b;
endmodule
```

I tried fixing it but I don't know how the name encoding/decoding works so my PR might be incomplete.

wsnyder, 2023-10-20:
I disagree that's legal as there's no printable ASCII character there, and even if it was, I'd refuse to support it ;)

    escaped_identifier ::= '\' { any_printable_ASCII_character_except_white_space } white_space

closed 2023-10-20 by wsnyder

dwRchyngqxs, 2023-10-20:
I understand your point about not wanting to supporting it, that's fair.
As I do not want to waste your time so I will just make my strongest argument once.
To be clear I do not argue about you not wanting to implement it, I tried it myself, it's not a 5 minutes fix.

I disagree with your interpretation of the standard.
Annex A specifies
```
Braces ( { } ) enclose items that can be repeated zero or more times.
```
So the rule you copied from the standard means:
`'\'` followed by `any_printable_ASCII_character_except_white_space` repeated _zero_ or more times, followed by `white_space`.
`\ ` is `'\'` followed by zero times `any_printable_ASCII_character_except_white_space` followed by a `white_space`, so it is valid.
Though I feel like this is not what is intended by the standard and the standard formal syntax should be fixed.