verilator/verilator #4595: PCH break cache in the CI

gezalore, 2023-10-21:
Since the PCH patch (for verilator itself) landed, ccache in the CI 'build' step is reporting high miss rates and is notably slower to compile Verilator on a rebuild. Note locally ccache seems to work fine, but in the CI we fetch the cache archive from the previous run.

Here is the ccache log for an arbitrary file that should have been in the cache. Looks like the hashes are completely off.

```
[2023-10-21T12:01:02.549638 7703 ] === CCACHE 4.5.1 STARTED =========================================
[2023-10-21T12:01:02.549638 7703 ] Config: (default) absolute_paths_in_stderr = false
[2023-10-21T12:01:02.549638 7703 ] Config: (default) base_dir = 
[2023-10-21T12:01:02.549638 7703 ] Config: (environment) cache_dir = /home/runner/work/verilator/verilator/.ccache
[2023-10-21T12:01:02.549638 7703 ] Config: (default) compiler = 
[2023-10-21T12:01:02.549638 7703 ] Config: (default) compiler_check = mtime
[2023-10-21T12:01:02.549638 7703 ] Config: (default) compiler_type = auto
[2023-10-21T12:01:02.549638 7703 ] Config: (environment) compression = true
[2023-10-21T12:01:02.549638 7703 ] Config: (default) compression_level = 0
[2023-10-21T12:01:02.549638 7703 ] Config: (default) cpp_extension = 
[2023-10-21T12:01:02.549638 7703 ] Config: (default) debug = false
[2023-10-21T12:01:02.549638 7703 ] Config: (default) debug_dir = 
[2023-10-21T12:01:02.549638 7703 ] Config: (default) depend_mode = false
[2023-10-21T12:01:02.549638 7703 ] Config: (default) direct_mode = true
[2023-10-21T12:01:02.549638 7703 ] Config: (default) disable = false
[2023-10-21T12:01:02.549638 7703 ] Config: (default) extra_files_to_hash = 
[2023-10-21T12:01:02.549638 7703 ] Config: (default) file_clone = false
[2023-10-21T12:01:02.549638 7703 ] Config: (default) hard_link = false
[2023-10-21T12:01:02.549638 7703 ] Config: (default) hash_dir = true
[2023-10-21T12:01:02.549638 7703 ] Config: (default) ignore_headers_in_manifest = 
[2023-10-21T12:01:02.549638 7703 ] Config: (default) ignore_options = 
[2023-10-21T12:01:02.549638 7703 ] Config: (default) inode_cache = false
[2023-10-21T12:01:02.549638 7703 ] Config: (default) keep_comments_cpp = false
[2023-10-21T12:01:02.549638 7703 ] Config: (environment) limit_multiple = 0.9
[2023-10-21T12:01:02.549638 7703 ] Config: (environment) log_file = /home/runner/work/verilator/verilator/repo/ccache.log
[2023-10-21T12:01:02.549638 7703 ] Config: (default) max_files = 0
[2023-10-21T12:01:02.549638 7703 ] Config: (environment) max_size = 250.0M
[2023-10-21T12:01:02.549638 7703 ] Config: (default) namespace = 
[2023-10-21T12:01:02.549638 7703 ] Config: (default) path = 
[2023-10-21T12:01:02.549638 7703 ] Config: (default) pch_external_checksum = false
[2023-10-21T12:01:02.549638 7703 ] Config: (default) prefix_command = 
[2023-10-21T12:01:02.549638 7703 ] Config: (default) prefix_command_cpp = 
[2023-10-21T12:01:02.549638 7703 ] Config: (default) read_only = false
[2023-10-21T12:01:02.549638 7703 ] Config: (default) read_only_direct = false
[2023-10-21T12:01:02.549638 7703 ] Config: (default) recache = false
[2023-10-21T12:01:02.549638 7703 ] Config: (default) reshare = false
[2023-10-21T12:01:02.549638 7703 ] Config: (default) run_second_cpp = true
[2023-10-21T12:01:02.549638 7703 ] Config: (default) secondary_storage = 
[2023-10-21T12:01:02.549638 7703 ] Config: (environment) sloppiness = time_macros, pch_defines
[2023-10-21T12:01:02.549638 7703 ] Config: (default) stats = true
[2023-10-21T12:01:02.549638 7703 ] Config: (default) stats_log = 
[2023-10-21T12:01:02.549638 7703 ] Config: (default) temporary_dir = /run/user/1001/ccache-tmp
[2023-10-21T12:01:02.549638 7703 ] Config: (default) umask = 
[2023-10-21T12:01:02.549756 7703 ] Command line: ccache g++ -O3 -DVERILATOR_INTERNAL_ -MMD -I. -I.. -I.. -I../../include -I../../include -MP -W -Wall -Wextra -Wfloat-conversion -Wlogical-op -faligned-new -Wno-unused-parameter -Wno-shadow -Werror -DDEFENV_SYSTEMC="" -DDEFENV_SYSTEMC_ARCH="" -DDEFENV_SYSTEMC_INCLUDE="" -DDEFENV_SYSTEMC_LIBDIR="" -DDEFENV_VERILATOR_ROOT="/usr/local/share/verilator" -include V3PchAstNoMT.h -c ../V3EmitXml.cpp -o V3EmitXml.o
[2023-10-21T12:01:02.549769 7703 ] Hostname: fv-az570-335
[2023-10-21T12:01:02.549773 7703 ] Working directory: /home/runner/work/verilator/verilator/repo/src/obj_opt
[2023-10-21T12:01:02.549776 7703 ] Compiler type: gcc
[2023-10-21T12:01:02.549819 7703 ] Detected use of precompiled header: V3PchAstNoMT.h.gch
[2023-10-21T12:01:02.549855 7703 ] Source file: ../V3EmitXml.cpp
[2023-10-21T12:01:02.549859 7703 ] Dependency file: V3EmitXml.d
[2023-10-21T12:01:02.549863 7703 ] Object file: V3EmitXml.o
[2023-10-21T12:01:02.549880 7703 ] Trying direct lookup
[2023-10-21T12:01:02.549956 7703 ] No 56b8ptpl2tuvcgvdrcv8f4f4ek4su6cou in primary storage
[2023-10-21T12:01:02.550011 7703 ] Running preprocessor
[2023-10-21T12:01:02.550018 7703 ] Executing /usr/bin/g++ -O3 -W -Wall -Wextra -Wfloat-conversion -Wlogical-op -faligned-new -Wno-unused-parameter -Wno-shadow -DVERILATOR_INTERNAL_ -I. -I.. -I.. -I../../include -I../../include -DDEFENV_SYSTEMC="" -DDEFENV_SYSTEMC_ARCH="" -DDEFENV_SYSTEMC_INCLUDE="" -DDEFENV_SYSTEMC_LIBDIR="" -DDEFENV_VERILATOR_ROOT="/usr/local/share/verilator" -include V3PchAstNoMT.h -fpch-preprocess -E ../V3EmitXml.cpp
[2023-10-21T12:01:02.622087 7703 ] Got result key from preprocessor
[2023-10-21T12:01:02.622143 7703 ] No 28a6ff7ub1l0h3r8nnvssf0lu225bo7eo in primary storage
[2023-10-21T12:01:02.622152 7703 ] Running real compiler
[2023-10-21T12:01:02.622192 7703 ] Executing /usr/bin/g++ -O3 -W -Wall -Wextra -Wfloat-conversion -Wlogical-op -faligned-new -Wno-unused-parameter -Wno-shadow -Werror -DVERILATOR_INTERNAL_ -I. -I.. -I.. -I../../include -I../../include -DDEFENV_SYSTEMC="" -DDEFENV_SYSTEMC_ARCH="" -DDEFENV_SYSTEMC_INCLUDE="" -DDEFENV_SYSTEMC_LIBDIR="" -DDEFENV_VERILATOR_ROOT="/usr/local/share/verilator" -include V3PchAstNoMT.h -fpch-preprocess -c -MMD -MP -fdiagnostics-color -o V3EmitXml.o ../V3EmitXml.cpp
[2023-10-21T12:01:04.542915 7703 ] Base dir not set, skip using relative paths
[2023-10-21T12:01:04.543032 7703 ] Using default compression level 1
[2023-10-21T12:01:04.543133 7703 ] Storing result file V3EmitXml.o
[2023-10-21T12:01:04.543140 7703 ] Storing embedded file #0 .o (221848 bytes) from V3EmitXml.o
[2023-10-21T12:01:04.544015 7703 ] Storing result file V3EmitXml.d
[2023-10-21T12:01:04.544025 7703 ] Storing embedded file #1 .d (390 bytes) from V3EmitXml.d
[2023-10-21T12:01:04.544363 7703 ] Stored 28a6ff7ub1l0h3r8nnvssf0lu225bo7eo in primary storage (/home/runner/work/verilator/verilator/.ccache/2/8/a6ff7ub1l0h3r8nnvssf0lu225bo7eoR)
[2023-10-21T12:01:04.544398 7703 ] Adding result key to /home/runner/work/verilator/verilator/.ccache/5/6/b8ptpl2tuvcgvdrcv8f4f4ek4su6couM
[2023-10-21T12:01:04.544574 7703 ] Using default compression level 1
[2023-10-21T12:01:04.544764 7703 ] Stored 56b8ptpl2tuvcgvdrcv8f4f4ek4su6cou in primary storage (/home/runner/work/verilator/verilator/.ccache/5/6/b8ptpl2tuvcgvdrcv8f4f4ek4su6couM)
[2023-10-21T12:01:04.544779 7703 ] Result: cache_miss
[2023-10-21T12:01:04.544783 7703 ] Result: direct_cache_miss
[2023-10-21T12:01:04.544786 7703 ] Result: preprocessed_cache_miss
[2023-10-21T12:01:04.544789 7703 ] Result: primary_storage_miss
[2023-10-21T12:01:04.544844 7703 ] Acquired lock /home/runner/work/verilator/verilator/.ccache/5/stats.lock
[2023-10-21T12:01:04.544962 7703 ] Releasing lock /home/runner/work/verilator/verilator/.ccache/5/stats.lock
[2023-10-21T12:01:04.544977 7703 ] Unlink /home/runner/work/verilator/verilator/.ccache/5/stats.lock
[2023-10-21T12:01:04.544998 7703 ] Acquired lock /home/runner/work/verilator/verilator/.ccache/2/stats.lock
[2023-10-21T12:01:04.545081 7703 ] Releasing lock /home/runner/work/verilator/verilator/.ccache/2/stats.lock
[2023-10-21T12:01:04.545094 7703 ] Unlink /home/runner/work/verilator/verilator/.ccache/2/stats.lock
[2023-10-21T12:01:04.545108 7703 ] Unlink /run/user/1001/ccache-tmp/tmp.stderr.mL0C1M
[2023-10-21T12:01:04.545115 7703 ] Unlink /run/user/1001/ccache-tmp/tmp.stdout.M6hRg6
[2023-10-21T12:01:04.545121 7703 ] Unlink /run/user/1001/ccache-tmp/tmp.cpp_stderr.nTn5rN
[2023-10-21T12:01:04.545127 7703 ] Unlink /run/user/1001/ccache-tmp/tmp.cpp_stdout.Gim6V4.ii
[2023-10-21T12:01:04.545146 7703 ] Unlink /run/user/1001/ccache-tmp/tmp.cpp_stdout.Gim6V4
```

wsnyder, 2023-10-21:
I'm assuming @gezalore you aren't debugging, if you are let me know.

Useful resource: https://interrupt.memfault.com/blog/ccache-debugging

gezalore, 2023-10-21:
I'm not debugging further at the moment, couldn't find anything obvious unfortunately.

wsnyder, 2023-10-21:
It's just due to PCH being large and overflowing cache.  Fortunately GitHub increased storage to 10G last year, so I bumped up.  It might not be large enough still for everything but it'll be better.

closed 2023-10-21 by wsnyder

gezalore, 2023-10-21:
Yep, that seems to have helped.