verilator/verilator #4601: Optimize compile time when many classes

wsnyder, 2023-10-21:
Compiling UVM without even using it takes very long. (UVM had many things commented out - it's not ready for use - see #1538).
 
This affects usability experience of UVM but doesn't directly block #1538.

On my system:

```
    1146user 452system 2:51elapsed  (13 minutes if was running on a Github Actions 2 core!)

     237 classes before specialization
     592 classes after specialization
    3919 .cpp files
    1184 .cpp _Slow files
    1760 .cpp files if combine DepSet
     592 .cpp files if no DepSet at all
```

Compiling all files as a single .cpp:

```
    all w/-O0 0m53s  (gcc only no linking)
    all w/-Os 2m44s  (gcc only no linking)
```

So we should be able to get under 1 minute wall time assuming 4+ CPUs.

Actions/ideas:

- [ ] Improve V3Dead. I think most of the generated classes really aren't used, and could be not be generated.
- [ ] When a class is specalized through parameters, merge all of the specializations into a single .cpp file (which might then get split if too large).  This will also likely help runtime performance as there's likely a lot of shared code and opportunity for inlining.
- [ ] Avoid making multiple __DepSet's that are small (e.g. one class has two each under 2KB)
- [ ] Combine small .cpp files (measured similar to --output-split) into larger ones.  Would need to be careful that a small change doesn't require a lot of recompiles.  I likely won't get to this.
- [ ] Through some fancy algorithm, disassociate .cpp files from modules, choosing the number of .cpp files by some algorithm based on a CPU count and output-split.  I likely won't get to this.

wsnyder, 2023-10-21:
See #2227

PeterMonsson, 2023-10-22:
Here are some unsolicited ideas that would help usability without actually helping compile times:
1) Ship verilator with a pre compiled UVM which is linked at the end
2) compile UVM in a ccache friendly way during the installation process

I am just putting them out here for inspiration.

kbieganski, 2023-10-23:
>  Improve V3Dead. I think most of the generated classes really aren't used, and could be not be generated

Yes, though so far this defect has proven to be useful in finding missing features, bugs, and regressions :)

> Ship verilator with a pre compiled UVM which is linked at the end

Some form of this is probably a good long time goal, but there are some complications:
* Verilator would have to know which parameterized classes are already precompiled and which ones should be emitted as C++,
* IIRC currently parameterized class names are suffixed with an incrementing integer, so the name of a class depends on the order of elaboration; a stable naming scheme would have to be implemented (possibly based on hashing),
* there are some compile time flags that the user can set which wouldn't affect UVM if it were precompiled, so we'd have to figure out what to do about this (maybe it doesn't matter?),
* managing compatibility with different compiler and stdlib versions would likely be a nightmare.

> compile UVM in a ccache friendly way during the installation process

This is very much doable in the short term. For this we need parameterized class name stability (and emit fewer files for bigger gains).

wsnyder, 2026-03-29:
Most of the ideas here have been implemented, the key remaining one has a separate issue #2227.

closed 2026-03-29 by wsnyder