verilator/verilator #4605: 51266898ec495196eb8187314bd3d96fdfc62dd8 breaks variable resolution in some packages

gezalore, 2023-10-21:
https://github.com/verilator/verilator/commit/51266898ec495196eb8187314bd3d96fdfc62dd8 introduces a failure on an old version of OpenTitan I'm using.

Unfortunately I don't have time to make a smaller case, so I tarred up my sources. To reproduce, run `verilator -f options.f` in the archive directory
[titan.tar.gz](https://github.com/verilator/verilator/files/13061894/titan.tar.gz)

Reverting the change to `AstNode::sameTreeIter` here is a fix:

https://github.com/verilator/verilator/commit/51266898ec495196eb8187314bd3d96fdfc62dd8#diff-e906e55f643fad56cc33abbacd2ffe7aa795a2fce460389336d160a1d7a4c5eaL1085

@RRozak would you mind attempting a more proper fix?

gezalore, 2023-10-21:
Note: GDB suggested an AstNodeVarRef to a variable inside a package does not have `classOrPackagep` set, but I don't see how that is related to the patch that introduced this...

RRozak, 2023-10-23:
I will look into it, but unfortunately not in the next few days.

gezalore, 2023-10-23:
Appreciate it regardless

RRozak, 2023-11-29:
I fixed it: https://github.com/verilator/verilator/pull/4723.
As you said, the problem was with `classOrPackagep` field.
The revert helped, because due to it, an additional variable was added, which was passed to a port instead of the original value defined in a package.

gezalore, 2023-11-29:
Thanks again for tracking this down

closed 2023-11-29 by gezalore