verilator/verilator #4607: Internal error and invalid lifetime errors with clocking blocks & repeat loops

amykyta3, 2023-10-23:
Thanks for the super quick fix for #4593!
Expanding the scope of Verilator on my end and found some more edge cases related to clocking blocks that Verilator disagrees with. This is a pruned down version of a pretty common pattern I have in a few of my regression tests.

There are a few internal errors in here, but all appear to be related to how loops in forked processes are verilated, so including them together in this testcase.

# Example file `test.sv`
```systemverilog
module tb();
    logic clk = 0;
    logic x;
    logic y;

    always #1ns clk = ~clk;

    assign y = x;

    default clocking cb @(posedge clk);
        output #1ns x;
        input #1step y;
    endclocking

    initial begin
        cb.x <= '0;
        ##2;
        fork
            begin
                repeat(5) begin
                    cb.x <= ~cb.y;
                    @cb;
                end
            end
            begin
                repeat(4) begin
                    ##1;
                end
            end
        join_any
        disable fork;
        $finish();
    end
endmodule
```

# Command-line & error
```bash
$ ./verilator/bin/verilator --sv --binary  test.sv
%Error-LIFETIME: test.sv:20:17: Invalid reference: Process might outlive variable `__Vrepeat0`.
                                    : ... note: In instance 'tb'
                                    : ... Suggest use it as read-only to initialize a local copy at the beginning of the process, or declare it as static. It is also possible to refer by reference to objects and their members.
   20 |                 repeat(5) begin
      |                 ^~~~~~
                 ... For error description see https://verilator.org/warn/LIFETIME?v=5.017
%Error: Internal Error: test.sv:22:22: ../V3Fork.cpp:599: Variable's lifetime is unknown. Can't determine if a capture is necessary.
                                           : ... note: In instance 'tb'
   22 |                     @cb;
      |                      ^~
```

# Version & OS
* Verilator 5.017 devel rev v5.014-234-g33c5b5feb
* Ubuntu 22.04

wsnyder, 2023-10-23:
Perhaps you can look into a pull to fix this?

The internal error should be improved to print the signal name.  Then the problem seems to be the repeat temporary that was inserted is in the wrong place to keep the fork block happy - you can experiment by making a for loop instead and making the loop variable in the right place to avoid the error - then change the Verilator Repeat loop to insert its variable in a similar spot.

kbieganski, 2023-10-23:
I looked into this, and prepared two patches for Verilator that fix your test case. However, they also break regression tests (at least the first one) so more work is needed on this. Unfortunately, right now I cannot spare the time, so I'll just paste the diffs here.

The first one fixes `repeat` loops in forks by making the repeat counter a func-local var. Sadly, it breaks some regression tests. If someone picks this up, remember that this var cannot be static in forks.

```diff
diff --git a/src/V3LinkJump.cpp b/src/V3LinkJump.cpp
index e777ac164..5687fa626 100644
--- a/src/V3LinkJump.cpp
+++ b/src/V3LinkJump.cpp

         AstVar* const varp
             = new AstVar{nodep->fileline(), VVarType::BLOCKTEMP, name, nodep->findSigned32DType()};
         varp->lifetime(VLifetime::AUTOMATIC);
+        varp->funcLocal(true);
         varp->usedLoopIdx(true);
-        m_modp->addStmtsp(varp);
+        nodep->addHereThisAsNext(varp);
         AstNode* initsp = new AstAssign{
             nodep->fileline(), new AstVarRef{nodep->fileline(), varp, VAccess::WRITE}, countp};
         AstNode* const decp = new AstAssign{
```

But even with that working, there are still failures related to lifetimes of vars generated for clocking blocks. That's an omission from the original PR, and possibly a simple fix:

```diff
diff --git a/src/V3AssertPre.cpp b/src/V3AssertPre.cpp
index c06f3b93f..92eda85e5 100644
--- a/src/V3AssertPre.cpp
+++ b/src/V3AssertPre.cpp

                 AstVar* const queueVarp = new AstVar{
                     flp, VVarType::MODULETEMP,
                     "__Vqueue__" + m_clockingp->name() + "__DOT__" + varp->name(), queueDtp};
+                queueVarp->lifetime(VLifetime::STATIC);
                 m_clockingp->addNextHere(queueVarp);
                 // Create a process like this:
                 //     always queue.push(<sampled var>);

         const std::string delayName = m_cycleDlyNames.get(nodep);
         AstVar* const cntVarp = new AstVar{flp, VVarType::BLOCKTEMP, delayName + "__counter",
                                            nodep->findBasicDType(VBasicDTypeKwd::UINT32)};
+        cntVarp->lifetime(VLifetime::AUTOMATIC);
+        cntVarp->funcLocal(true);
         AstBegin* const beginp = new AstBegin{flp, delayName + "__block", cntVarp, false, true};
         beginp->addStmtsp(new AstAssign{flp, new AstVarRef{flp, cntVarp, VAccess::WRITE}, valuep});
         beginp->addStmtsp(new AstWhile{
diff --git a/src/V3LinkDot.cpp b/src/V3LinkDot.cpp
index df0028782..ecf4de66c 100644
--- a/src/V3LinkDot.cpp
+++ b/src/V3LinkDot.cpp

             AstVar* const eventp = new AstVar{
                 clockingp->fileline(), VVarType::MODULETEMP, clockingp->name(), VFlagChildDType{},
                 new AstBasicDType{clockingp->fileline(), VBasicDTypeKwd::EVENT}};
+            eventp->lifetime(VLifetime::STATIC);
             clockingp->eventp(eventp);
             // Trigger the clocking event in Observed (IEEE 1800-2017 14.13)
             clockingp->addNextHere(new AstAlwaysObserved{
```

As an aside, I think the whole var lifetime/locality/type system needs to be reworked, because right now it's not entirely clear what are the rules for those.

kbieganski, 2023-10-23:
Just noticed the `repeat` part of this is a duplicate of https://github.com/verilator/verilator/issues/4451

wsnyder, 2025-09-08:
Example passes on master, presume above merged pulls fixed it, otherwise please open a new issue thanks.

closed 2025-09-08 by wsnyder