verilator/verilator #4614: Parsing Bug: syntax error, unexpected TYPE-IDENTIFIER

IAMAl, 2023-10-24:
- coomand

verilator -cc -trace -exe top_module.cpp -F hdl_file_list_copy.txt --top-module DataPath_SL

- error message

%Error: /media/takano/disk1_8TB/workspace/ElectronNest-Verilator/src_hdl/RALU/DataPath_SL.sv:210:13: syntax error, unexpected TYPE-IDENTIFIER
  210 |  ShiftLogic ShiftLogic
      |             ^~~~~~~~~~
%Error: Exiting due to 1 error(s)

-DataPath_SL.sv

module DataPath_SL
	import	pkg_en::*;
	import	pkg_alu::*;
	import	pkg_extend_index::*;
#(
	parameter UNIT_TYPE			= 0,
	parameter WIDTH_OPCODE			= 8,
	parameter WIDTH_CONSTANT		= 8,
	parameter WIDTH_DATA			= 32,
	parameter ExtdConfig			= 0
)(...);
...
	ShiftLogic ShiftLogic
	(/*brank*/
	);
...
endmodule

- ShiftLogicUnit.sv

module ShiftLogic
	import	pkg_en::*;
	import	pkg_alu::*;
(...);
...
endmodule

- hdl_file_list_copy.txt

/media/hogege/disk1_8TB/workspace/ElectronNest-Verilator/src_hdl/Packages/pkg_alu.sv
/media/hogege/disk1_8TB/workspace/ElectronNest-Verilator/src_hdl/Packages/pkg_extend_index.sv
/media/hogege/disk1_8TB/workspace/ElectronNest-Verilator/src_hdl/Packages/pkg_en.sv
/media/hogege/disk1_8TB/workspace/ElectronNest-Verilator/src_hdl/Packages/pkg_bram_if.sv
/media/hogege/disk1_8TB/workspace/ElectronNest-Verilator/src_hdl/Packages/pkg_link.sv
/media/hogege/disk1_8TB/workspace/ElectronNest-Verilator/src_hdl/Packages/pkg_mem.sv
/media/hogege/disk1_8TB/workspace/ElectronNest-Verilator/src_hdl/IComp/SkipInit.sv
/media/hogege/disk1_8TB/workspace/ElectronNest-Verilator/src_hdl/IComp/SyncUnit.sv
/media/hogege/disk1_8TB/workspace/ElectronNest-Verilator/src_hdl/RALU/DataPath_SL.sv
/media/hogege/disk1_8TB/workspace/ElectronNest-Verilator/src_hdl/RALU/ShiftLogicUnit.sv

- verilator --version

Verilator 4.038 2020-07-11 rev v4.036-114-g0cd4a57ad

- OS and distribution

Linux, Ubuntu 22.04.2

- May we assist you in trying to fix this in Verilator yourself?
Please

donlon, 2023-10-24:
Probably `ShiftLogic` has also been declared as a type (class/struct/typedef/...) before? If so it's similar to #2412.

IAMAl, 2023-10-24:
The ShiftLogic is not declared anywhere, just only in ShiftLogicUnit.sv, and xilinx vivado can elaborate.

IAMAl, 2023-10-24:
DataPath_SL module, when removing ShiftLogic instance does not make the error, and when renaming instance name as ShiftLogicUnit or others makes another syntax error for “(” on that the instance declaration. I also checked whether hidden not appropriate text code is there or not, no such code in the file.

donlon, 2023-10-24:
Could you confirm that the name is not appeared in the packages imported? If this were not the case it shouldn't become a TYPE-IDENTIFIER.

IAMAl, 2023-10-24:
package files do only declare parameters, and not includes module name.

wsnyder, 2023-10-24:
Run with --debug and look at the obj_dir/*__inputs.vpp file, and look for everywhere ShiftLogic occurs.

I agree with @dolon this is almost certainly #2412 and you'll see it on some lines other than the instance.

IAMAl, 2023-10-24:
ah,

in pkg_alu.sv;
//	 ShiftLogicUnit Opcode
	typedef enum logic {
		SHT					= 1'b0,
		LGC					= 1'b1
	} ShiftLogic;
This may make the error, thank you for pointing out.

wsnyder, 2023-10-27:
Duplicate of #2414.  As workaround use two different names.

closed 2023-10-27 by wsnyder