verilator/verilator #4621: "the coroutine header requires -fcoroutines"

sifferman, 2023-10-25:
## Error

I just updated to `Verilator 5.017 devel rev v5.014-242-gcf6e36297` from `Verilator 5.017 devel rev v5.014-161-g622f0c047` and started receiving an error that "the coroutine header requires -fcoroutines".

I didn't see any issues/PRs that address this, but I may have missed something.

I am getting my binary from OSS CAD Suite, so there is a chance their build scripts simply became outdated if there was some recent dependency change.

## System Info

```plain
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.3 LTS
Release:        22.04
Codename:       jammy
```

## Test

```systemverilog
module tb;
    initial begin
        #1;
        $finish();
    end
endmodule
```
```bash
verilator --binary tb.sv
```
### Full Error
```plain
make: Entering directory './obj_dir'
g++ -Os  -I.  -MMD -I./share/verilator/include -I./share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-overloaded-virtual -Wno-shadow -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -DVL_TIME_CONTEXT    -c -o verilated.o ./share/verilator/include/verilated.cpp
g++ -Os  -I.  -MMD -I./share/verilator/include -I./share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-overloaded-virtual -Wno-shadow -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -DVL_TIME_CONTEXT    -c -o verilated_timing.o ./share/verilator/include/verilated_timing.cpp
In file included from ./share/verilator/include/verilated_timing.h:45,
                 from ./share/verilator/include/verilated_timing.cpp:23:
/usr/include/c++/11/coroutine:334:2: error: #error "the coroutine header requires -fcoroutines"
  334 | #error "the coroutine header requires -fcoroutines"
      |  ^~~~~
In file included from ./share/verilator/include/verilated_timing.cpp:23:
./share/verilator/include/verilated_timing.h:98:10: error: ‘coroutine_handle’ in namespace ‘std’ does not name a template type
   98 |     std::coroutine_handle<> m_coro;  // The wrapped coroutine handle
      |          ^~~~~~~~~~~~~~~~
./share/verilator/include/verilated_timing.h:112:44: error: expected ‘)’ before ‘<’ token
  112 |     VlCoroutineHandle(std::coroutine_handle<> coro, VlProcessRef process, VlFileLineDebug fileline)
      |                      ~                     ^
      |                                            )
./share/verilator/include/verilated_timing.h:114:9: error: expected unqualified-id before ‘,’ token
  114 |         , m_process{process}
      |         ^
./share/verilator/include/verilated_timing.h:115:9: error: expected unqualified-id before ‘,’ token
  115 |         , m_fileline{fileline} {
      |         ^
./share/verilator/include/verilated_timing.h:115:32: error: expected unqualified-id before ‘{’ token
  115 |         , m_fileline{fileline} {
      |                                ^
./share/verilator/include/verilated_timing.h: In constructor ‘VlCoroutineHandle::VlCoroutineHandle(VlProcessRef)’:
./share/verilator/include/verilated_timing.h:108:11: error: class ‘VlCoroutineHandle’ does not have any field named ‘m_coro’
  108 |         : m_coro{nullptr}
      |           ^~~~~~
./share/verilator/include/verilated_timing.h: In constructor ‘VlCoroutineHandle::VlCoroutineHandle(VlCoroutineHandle&&)’:
./share/verilator/include/verilated_timing.h:122:11: error: class ‘VlCoroutineHandle’ does not have any field named ‘m_coro’
  122 |         : m_coro{std::exchange(moved.m_coro, nullptr)}
      |           ^~~~~~
./share/verilator/include/verilated_timing.h:122:38: error: ‘class VlCoroutineHandle’ has no member named ‘m_coro’
  122 |         : m_coro{std::exchange(moved.m_coro, nullptr)}
      |                                      ^~~~~~
In file included from ./share/verilator/include/verilated.h:42,
                 from ./share/verilator/include/verilated_timing.h:28,
[... 140 lines cut ...]
      |                                                            ^~~~

In file included from ./share/verilator/include/verilated.h:42,
                 from ./share/verilator/include/verilated_timing.h:28,
                 from ./share/verilator/include/verilated_timing.cpp:23:
./share/verilator/include/verilated_timing.cpp: In member function ‘void VlCoroutineHandle::resume()’:
./share/verilator/include/verilated_timing.cpp:31:19: error: ‘m_coro’ was not declared in this scope
   31 |     if (VL_LIKELY(m_coro)) {
      |                   ^~~~~~
./share/verilator/include/verilatedos.h:65:43: note: in definition of macro ‘VL_LIKELY’
   65 | # define VL_LIKELY(x) __builtin_expect(!!(x), 1)  // Prefer over C++20 [[likely]]
      |                                           ^
./share/verilator/include/verilated_timing.cpp: In destructor ‘VlCoroutine::VlPromise::~VlPromise()’:
./share/verilator/include/verilated_timing.cpp:217:9: error: ‘m_continuation’ was not declared in this scope
  217 |     if (m_continuation) m_continuation.destroy();
      |         ^~~~~~~~~~~~~~
./share/verilator/include/verilated_timing.cpp: At global scope:
./share/verilator/include/verilated_timing.cpp:220:6: error: ‘suspend_never’ in namespace ‘std’ does not name a type
  220 | std::suspend_never VlCoroutine::VlPromise::final_suspend() noexcept {
      |      ^~~~~~~~~~~~~
make: *** [./share/verilator/include/verilated.mk:278: verilated_timing.o] Error 1
make: Leaving directory './obj_dir'
%Error: make -C obj_dir -f Vtb.mk -j 1 exited with 2
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec verilator_bin --binary tb.sv
```

wsnyder, 2023-10-26:
Please try the github master to see if you get the same error.  As you suspect this is likely a OOSS CAD issue.

sifferman, 2023-10-30:
@wsnyder Sorry for not responding, had a busy week.

Building from master fixed the error, but I would like to figure out why OSS CAD Suite is not working: YosysHQ/oss-cad-suite-build#84.

The first release of a broken Verilator binary was on Oct 21th, 2023, 1:48AM UTC. That means that OSS CAD Suite likely broke due to a commit between 5af271cf3af2a2276530e347dfb1a03493fd8632 and 26e378596382e654f5e39e17e6bc61942feb55c2. My guess is the commit "Support Clang 16 #4592".

Do you have any guesses? I can look into it later this week.

sifferman, 2023-11-06:
Alright, this is the commit that broke things:
> <https://github.com/verilator/verilator/commit/c7a0613c57827f0f9267fbfb0a064b5133a76678#diff-49473dca262eeab3b4a43002adb08b4db31020d190caaad1594b47f1d5daa810L374-R376>
> ```diff
> < _MY_CXX_CHECK_SET(CFG_CXXFLAGS_COROUTINES,-fcoroutines-ts)
> < _MY_CXX_CHECK_SET(CFG_CXXFLAGS_COROUTINES,-fcoroutines)
> < _MY_CXX_CHECK_SET(CFG_CXXFLAGS_COROUTINES,-fcoroutines-ts -Wno-deprecated-experimental-coroutine)
> ---
> > _MY_CXX_CHECK_IFELSE(
> >   -fcoroutines-ts,
> >   [CFG_CXXFLAGS_COROUTINES="-fcoroutines-ts"],
> >   [CFG_CXXFLAGS_COROUTINES="-fcoroutines"])
> ```

Verilator used to always include `-fcoroutines` by default, but now it only includes `-fcoroutines` if it is supported. I believe that the error is because OSS CAD Suite builds Verilator with a C++ compiler that does not support coroutines.

I think a good addition to Verilator would be a AC warning that a compiler is used that doesn't support coroutines.

wsnyder, 2023-11-07:
If Verilator was configured and built without coroutines it will tell you.  Run "verilator -V" and see COROUTINES.  If that doesn't match your compiler's flags, you are using a different compiler setup then Verilator was built for, which I don't think is something adding a warning to configure can solve.

sifferman, 2023-11-07:
Thanks for the response, and yes I agree. This issue is now officially resolved. Thanks!