verilator/verilator #4625: Rand fields of dynamic array types

RRozak, 2023-10-26:
Rand fields of dynamic array and queue types are currently unsupported by Verilator.
I prepared a test:
```systemverilog
`define check_rand(cl, field) \
begin \
   longint prev_result; \
   int ok = 0; \
   for (int i = 0; i < 10; i++) begin \
      longint result; \
      void'(cl.randomize()); \
      result = longint'(field); \
      if (i > 0 && result != prev_result) ok = 1; \
      prev_result = result; \
   end \
   if (ok != 1) $stop; \
end

int N = 10;

class Cls;
   rand int dyn[];
   function new;
      dyn = new[N];
   endfunction
endclass

module t;
   initial begin
      Cls c = new;
      for (int i = 0; i < N; i++) begin
         `check_rand(c, c.dyn[i]);
      end

      $write("*-* All Finished *-*\n");
      $finish;
   end
endmodule
```

There are a few occurrences of them in UVM.

wsnyder, 2023-11-29:
@kiryk curious if there might be a pull coming for this. No pressure - just have a use case.

kiryk, 2023-11-30:
@wsnyder yes, this is in preparation, and IMO close to being finished

wsnyder, 2024-10-04:
See also pull #4999  ( @YilouWang note that please)

YilouWang, 2024-10-04:
Thanks! I will.

YilouWang, 2024-10-08:
Hello, :)
After PR #5515 and #5415 , all kinds of arrays' basic(unconstrained) randomization can be supported in Verilator, except for some minor situations like multiple dimension associative arrays(I will fix soon) and some complex structures like issue #5455 .

And Here’s a brief summary of the current status regarding **constrained randomization** for arrays, to better align with everyone:

1. 1-dimensional queues and dynamic arrays are now supported, thanks to the recent PR #5413 .
2. 1-dimensional and multi-dimensional unpacked arrays are also supported, thanks to the PR #5489 .

What remains unsupported are: (I am currently working on these two)

1. 1-dimensional and multi-dimensional associative arrays
2. Multi-dimensional dynamic arrays and queues (also the reason why the two remaining comments in uvm_pkg_todo.vh cannot be commented-in yet)

Thanks for your attention to this!

YilouWang, 2024-10-21:
After PR #5529 , #5530 , #5537 , #5547 , all found and reported issues have been solved and all array types' basic (unconstrained) randomization is fully supported.

Next step is supporting constrained randomization for multi-d queue and dyn-arr which can be expected to be done soon.

RRozak, 2024-10-24:
Thanks for your contribution.
I noticed that `rand` dynamic arrays don't work if their elements are of class objects. I created a separate issue for it: https://github.com/verilator/verilator/issues/5557 and I already started working on it.

YilouWang, 2024-12-03:
Hello, guys

I’d like to provide an update on the progress in November to help us align:

After PR #5591, Verilator now supports multi-dimensional dynamic arrays and queues. The remaining task is to add support for constraints on functions like `size()`. I noticed that you have already created and started addressing related issues (#5582 #5647 ), as well as resolving the earlier ones.

On my side, I’ve started working on adding support for constrained randomization for associative arrays and unpacked structs. This task is already underway.

Let me know if there’s anything else that needs to be coordinated.