verilator/verilator #4630: generate is very slow

muzafferkal, 2023-10-26:
I'm having significant speed penalty in using a (large?) generate loop both for verilator and also c compile pass after.
I have a module which implements a multi-input adder tree and I need to instantiate this more than 100 times which I do in a generate loop. The problem is running it at the max number takes verilator more than 10 minutes (not actually timed but I thought it hanged so noticed it much later that it actually worked) as opposed to 20-30 seconds when the generate loop count is 1. The c compiler pass after verilator is also very slow.
Any idea how to accelerate this? compiler settings etc? I'm running -O3 for verilator and it's doing -Os on the compiler flags.

I'm afraid I don't have a test case. I can't share this code because it's proprietary. I'll try VCS next just to see what it would do.

Verilator 5.017 devel rev v5.014-145-g44e7d2ebe

wsnyder, 2023-10-26:
Please run with --stats, and you can look at the breakdown of the time taken.  Verilator flatens the design, so if the time is about the same for every stage there's not much we can do, but if a single stage is massively more time this might indicate a bug.

Note #1314 which seems similar.

muzafferkal, 2023-10-26:
thanks for the suggestion. Here is the stats file. Actually it turns out the c++ compile is the really long pole. Are there are compile options I can pass to make it faster? 

[Vcim_tb__stats.txt](https://github.com/verilator/verilator/files/13183533/Vcim_tb__stats.txt)

wsnyder, 2023-10-26:
Compile with -O0, be sure to use ccache, and make with -j. There are tips in the manual too.

wsnyder, 2023-10-27:
I didn't see anything surprising in the stats.  You may compile faster without using --trace, or at least trace as little as possible.

wsnyder, 2023-10-30:
Assuming matches #1314 unless hear otherwise (and even if so likely #1314 should extend to cover it).

closed 2023-10-30 by wsnyder